open-power/skiboot #145: OPAL checkstops on P8 habanero platform.

pridhiviraj, 2018-01-09:
```
 48.68328|ISTEP 21. 3
[   49.035784849,5] OPAL v5.9-219-g4e23b42 starting...
[   49.035788992,7] initial console log level: memory 7, driver 5
[   49.035792104,6] CPU: P8 generation processor (max 8 threads/core)
[   49.035795045,7] CPU: Boot CPU PIR is 0x0058 PVR is 0x004d0200
[   49.035798168,7] CPU: Initial max PIR set to 0x1fff
[   49.036136756,7] OPAL table: 0x300f5430 .. 0x300f5980, branch table: 0x30002000
[   49.036142286,7] Assigning physical memory map table for unused
[   49.036146239,7] FDT: Parsing fdt @0xff00000
[   49.039950384,6] CHIP: Initialised chip 0 from xscom@3fc0000000000
[   49.040030377,5] CHIP: Chip ID 0000 type: P8 DD2.0
[   49.040033568,7] XSCOM: Base address: 0x3fc0000000000
[   49.040041392,7] XSTOP: XSCOM addr = 0x2010c82, FIR bit = 31
[   49.040044784,6] MFSI 0:0: Initialized
[   49.040047144,6] MFSI 0:2: Initialized
[   49.040049388,6] MFSI 0:1: Initialized
[   49.040366176,5] LPC: LPC[000]: Initialized, access via XSCOM @0xb0020
[   49.040373548,7] LPC: Default bus on chip 0x0
[   49.040476630,6] MEM: parsing reserved memory from node /ibm,hostboot/reserved-memory
[   49.040492030,7] HOMER: Init chip 0
[   49.040494872,7]   PBA BAR0 : 0x0000001ffd800000
[   49.040497602,7]   PBA MASK0: 0x0000000000300000
[   49.040500341,7]   HOMER Image at 0x1ffd800000 size 4MB
[   49.040504161,7]   PBA BAR2 : 0x4000001ffda00000
[   49.040506880,7]   PBA MASK2: 0x0000000000000000
[   49.040509558,7]   SLW Image at 0x1ffda00000 size 1MB
[   49.040513051,7]   PBA BAR3 : 0x0000001fff800000
[   49.040515787,7]   PBA MASK3: 0x0000000000700000
[   49.040518516,7]   OCC Common Area at 0x1fff800000 size 8MB
[   49.040542899,7] CPU idle state device tree init
[   49.040546366,5] SLW: HB-provided idle states property found
[   49.040550476,5] SLW: Enabling: nap
[   49.040552703,5] SLW: Enabling: fastsleep_
[   49.040555453,5] SLW: Enabling: winkle
[   49.040747560,5] PLAT: Fixing up PSI BAR on chip 0 BAR=3fffe80000001
[   49.040780433,7] AST: PNOR LPC offset: 0x0c000000
[   49.040852432,5] PLAT: Using virtual UART
[   49.041152498,7] UART: Using LPC IRQ 4
[   51.306177289,5] PLAT: Detected Habanero platform
[   51.306243838,5] PLAT: Detected BMC platform AMI
[   51.333314543,5] CENTAUR: Found centaur for chip 0x0 channel 0
[   51.374166000,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 3
[   51.419562617,5] CENTAUR: Found centaur for chip 0x0 channel 1
[   51.461119577,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 2
[   51.497942372,5] CENTAUR: Found centaur for chip 0x0 channel 4
[   52.017239379,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 7
[   52.042892897,5] CENTAUR: Found centaur for chip 0x0 channel 5
[   52.109878266,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 6
[   52.249537318,5] CPU: All 80 processors called in...
[   54.489323463,5] FLASH: Found system flash: Macronix MXxxL51235F id:0
[   54.489493132,5] BT: Interface initialized, IO 0x00e4
[   55.438472236,5] NVRAM: Size is 576 KB
[   56.104933074,5] STB: Found ibm,secureboot-v1
[   56.105002972,5] STB: secure mode off
[   56.374904516,5] STB: trusted mode off
[   56.374988335,5] OPAL: Using OPAL UART console
[   56.424931509,5] Chip 0 Found PBCQ0 at /xscom@3fc0000000000/pbcq@2012000
[   56.427732347,7] FLASH: CAPP partition has ECC
[   56.427733551,7]   3.62305|Ignoring boot flags, incorrect version 0x0
  3.68837|ISTEP  6. 3
  4.04964|ISTEP  6. 4
  4.05021|ISTEP  6. 5
 20.51594|HWAS|PRESENT> DIMM[03]=AAAA0000AAAA0000
 20.51595|HWAS|PRESENT> Membuf[04]=CC00000000000000
 20.51595|HWAS|PRESENT> Proc[05]=8000000000000000
 20.65881|ISTEP  6. 6
 22.41183|================================================
 22.41183|Error reported by unknown (0xE500)
 22.41184|  <none>
 22.41184|  ModuleId   0x0b unknown
 22.41184|  ReasonCode 0xe540 unknown
 22.41184|  UserData1  unknown : 0x0005000000000101
 22.41184|  UserData2  unknown : 0x4241000500000000
 22.41185|User Data Section 0, type UD
 22.41185|  Subsection type 0x06
 22.41185|  ComponentId errl (0x0100)
 22.41185|  CALLOUT
 22.41185|  PROCEDURE ERROR
 22.41186|  Procedure: 16
 22.41186|User Data Section 1, type UD
 22.41186|  Subsection type 0x04
 22.41186|  ComponentId errl (0x0100)
 22.41186|User Data Section 2, type UD
 22.41187|  Subsection type 0x06
 22.41187|  ComponentId errl (0x0100)
 22.41187|  CALLOUT
 22.41187|  HW CALLOUT
 22.41187|  Reporting CPU ID: 16
 22.41188|  Called out entity:
 22.41188|User Data Section 3, type UD
 22.41188|  Subsection type 0x33
 22.41188|  ComponentId unknown (0xe500)
 22.41189|User Data Section 4, type UD
 22.41189|  Subsection type 0x01
 22.41189|  ComponentId unknown (0xe500)
 22.41189|  STRING
 22.41189|  
 22.41190|User Data Section 5, type UD
 22.41190|  Subsection type 0x15
 22.41190|  ComponentId hb-trace (0x3100)
 22.41190|User Data Section 6, type UD
 22.41191|  Subsection type 0x01
 22.41191|  ComponentId unknown (0xe500)
 22.41191|  STRING
 22.41191|  ^B
 22.41191|User Data Section 7, type UD
 22.41192|  Subsection type 0x03
 22.41192|  ComponentId errl (0x0100)
 22.41192|User Data Section 8, type UD
 22.41192|  Subsection type 0x01
 22.41192|  ComponentId errl (0x0100)
 22.41193|  STRING
 22.41193|  Hostboot Build ID: hostboot-695bd89/hbicore.bin
 22.41193|User Data Section 9, type UD
 22.41193|  Subsection type 0x04
 22.41194|  ComponentId errl (0x0100)
 22.41194|================================================
 26.65630|ISTEP  6. 7
 28.43339|ISTEP  6. 8
 28.48022|ISTEP  6. 9
 32.79379|ISTEP  6.10
 32.79499|ISTEP  6.11
 32.91425|ISTEP  6.12
```
Reproducible with every run.

pridhiviraj, 2018-01-09:
OPAL commit is in 4e23b42d2ad76da21422a1d2de471df29f76b8df which latest upstream commit.

pridhiviraj, 2018-01-09:
We cannot test with upstream op-build PNOR as we have HB failure dependency https://github.com/open-power/op-build/issues/1711 . So to test latest OPAL code, we need to WA that 1711 first or flash slightly older PNOR and flash the latest skiboot alone.

oohal, 2018-01-10:
What PNOR version did you get this with? I tested the latest upstream skiboot flashed on top of whatever PNOR our habenero had loaded and it booted fine.

pridhiviraj, 2018-01-10:
@oohal Below is the PNOR version flashed.
```
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-41c9602
 Product Extra         : 	buildroot-2017.08.1-9-g6ddc3b3
 Product Extra         : 	skiboot-v5.9-21-g88bd3bc5e6f8
 Product Extra         : 	hostboot-695bd89
 Product Extra         : 	linux-4.13.11-openpower1-p09eaad5
 Product Extra         : 	petitboot-v1.6.3-pba431a5
 Product Extra         : 	machine-xml-6a78496-p97b731b
 Product Extra         : 	occ-28f2cec
 Product Extra         : 	hos
```
I think from the failure what i guessed is while it probing for chip 8 PBCQ, it hitting the checkstop.

pridhiviraj, 2018-01-10:
Sorry, It's a habanero system, only one chip. so it's not during PBCQ probe, may be some thing else.

stewart-ibm, 2018-01-15:
I've reproduced on my habanero. Interesting.... bisecting to find out when we broke it at least.

pridhiviraj, 2018-01-15:
@stewart-ibm Good to know that it is reproducible. is it some thing only for habanero or across P8 platforms?

stewart-ibm, 2018-01-16:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> @stewart-ibm Good to know that it is reproducible. is it some thing
> only for habanero or across P8 platforms?

I haven't tried other p8 platforms.

IT appears to have started with this patch:

commit 8c762d5359d6d00ae2ca2d9e5a4e2ae87a402cf0
Author: Stewart Smith <stewart@linux.vnet.ibm.com>
Date:   Wed Dec 13 20:33:22 2017 +1100

    allow secure boot if not enforcing it
    
    We check the secure boot containers no matter what, only *enforcing*
    secure boot if we're booting in secure mode. This gives us an extra
    layer of checking firmware is legit even when secure mode isn't enabled,
    as well as being really useful for testing.
    
    Signed-off-by: Stewart Smith <stewart@linux.vnet.ibm.com>

and as a result of that something is going wrong. Not sure what yet,
only bisected it down at the end of yesterday.

-- 
Stewart Smith
OPAL Architect, IBM.

stewart-ibm, 2018-01-30:
Fixed in https://patchwork.ozlabs.org/patch/863285/ i.e. 43eb8bbcbaa5b4c01305b51b061ddd98e75efcae

closed 2018-01-30 by stewart-ibm

pridhiviraj, 2018-01-31:
Thanks @stewart-ibm .