open-power/skiboot #148: STB: primary CPU thread not aborting immedaitely after a secondary CPU job got abort.

pridhiviraj, 2018-02-08:
```
 66.01168|====================[   66.011887081,5] OPAL skiboot-v5.9-240-g081882690163-pcbedce4 starting...
[   66.011894300,7] initial console log level: memory 7, driver 5
[   66.011896673,6] CPU: P9 generation processor (max 4 threads/core)
[   66.011898763,7] CPU: Boot CPU PIR is 0x0004 PVR is 0x004e1200
[   66.011901029,7] CPU: Initial max PIR set to 0x7fff
[   66.013275671,7] OPAL table: 0x300f3430 .. 0x300f3980, branch table: 0x30002000
[   66.013279680,7] Assigning physical memory map table for nimbus
[   66.013282901,7] Parsing HDAT...
[   66.013284388,5] SPIRA-S found.
[   66.013287548,6] BMC #0: HW version 3, SW version 2, chip DD1.0
[   66.013594020,6] SP Family is ibm,ast2500,supermicro
[   66.013601470,7] LPC: IOPATH chip id = 0
[   66.013602989,7] LPC: FW BAR       = f0000000
[   66.013604654,7] LPC: MEM BAR      = e0000000
[   66.013606323,7] LPC: IO BAR       = d0010000
[   66.013607996,7] LPC: Internal BAR = c0012000
[   66.013621312,7] LPC UART: base addr = 3f8 (3f8) size = 1 clk = 1843200, baud = 115200
[   66.013624456,7] LPC: BT [0, 0] sms_int: 0, bmc_int: 0
[   66.069159648,5] UART: Using UART at 0x60300d00103f8
[   66.074079250,3] I2C: v2 found, but not supported. Parsing as v1
[   66.077302945,3] I2C: v2 found, but not supported. Parsing as v1
[   66.082093710,5] P9 DD2.00 detected
[   66.083712465,5] CHIP: Chip ID 0000 type: P9N DD2.0
[   66.086219502,5] P9 DD2.00 detected
[   66.088692685,5] CHIP: Chip ID 0008 type: P9N DD2.0
[   66.091625140,5] PLAT: Using���m���r�ʚ������b���
                                                     �('H�outing irq 9, policy: 0 (r=1)
[   66.093776222,7] LPC: SerIRQ 9 using route 0 targetted at OPAL
[   66.099718681,5] PLAT: Detected p9dsu2u platform
[   66.102200394,5] PLAT: Detected BMC platform SMC
[   66.133107624,5] CPU: All 128 processors called in...
[   68.004878257,3] MBOX-FLASH: Bad response code from BMC 2
[   68.007064573,3] MBOX-FLASH: Error waiting for BMC
[   68.420329767,5] FLASH: Found system flash:  id:0
[   68.422313593,5] BT: Interface initialized, IO 0x00e4
[   68.425131211,7] LPC: Routing irq 10, policy: 0 (r=1)
[   68.425132341,7] LPC: SerIRQ 10 using route 1 targetted at OPAL
[   69.483549221,0] STB: VERSION NOT VERIFIED, invalid param. buf=0x305ed9e0, len=4096 key-hash=0x0 hash-size=0
[   69.489470246,5] NVRAM: Size is 576 KB
[   70.079277952,5] console: Setting driver log level to 7
[   70.079287356,7] LPC-MBOX: Sending BMC interrupt
[   70.079383270,6] BT: seq 0x05 netfn 0x06 cmd 0x06: IPMI MSG done
[   70.079437430,6] BT: seq 0x06 netfn 0x06 cmd 0x2e: Message sent to host
[   70.090028497,7] NVRAM: 'log-level-memory' not found
[   70.092866731,5] STB: Found ibm,secureboot-v2
[   70.094995784,7] NVRAM: Searched for 'force-secure-mode' found 'always'
[   70.098544526,5] STB: secure mode on (FORCED by nvram)
[   70.100682149,6] STB: Found CVC @ 203ffd260000-203ffd26ffff
[   70.103520390,6] STB: Found CVC-sha512 @ 203ffd260040, version=1
[   70.107065944,6] STB: Found CVC-verify @ 203ffd260050, version=1
[   70.109907450,7] NVRAM: 'force-trusted-mode' not found
[   70.112745453,5] STB: trusted mode off
[   70.114170054,5] OPAL: Using OPAL UART console
[   70.117004278,7] NVRAM: 'uart-con-policy' not found
[   70.119141261,7] LPC: Routing irq 4, policy: 0 (r=1)
[   70.119142308,7] LPC: SerIRQ 4 using route 2 targetted at OPAL
[   70.124816744,7] SLW: Init chip 0x0
[   70.126240442,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.129789010,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.133335923,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.136179875,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.139729120,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.143276085,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.146825633,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.149668887,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.153217803,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.156764344,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.159610267,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[   70.163156883,7] SLW: EC_PPM_SPECIAL_WKUP_HYP read  0x0000000000000000
[... 211 lines cut ...]
[   74.404962901,6] BT: seq 0x19 netfn 0x3a cmd 0xf0: IPMI MSG done
[   74.407830502,6] BT: seq 0x1a netfn 0x3a cmd 0xf0: Message sent to host
[   75.001403963,6] BT: seq 0x1a netfn 0x3a cmd 0xf0: IPMI MSG done
[   75.004224409,6] BT: seq 0x1b netfn 0x3a cmd 0xf0: Message sent to host
[   75.109799146,6] BT: seq 0x1b netfn 0x3a cmd 0xf0: IPMI MSG done
[   75.112575361,6] BT: seq 0x1c netfn 0x3a cmd 0xf0: Message sent to host
[   75.218853641,6] BT: seq 0x1c netfn 0x3a cmd 0xf0: IPMI MSG done
[   75.221583085,6] BT: seq 0x1d netfn 0x3a cmd 0xf0: Message sent to host
[   75.327871991,6] BT: seq 0x1d netfn 0x3a cmd 0xf0: IPMI MSG done
[   75.330549114,6] BT: seq 0x1e netfn 0x0a cmd 0x44: Message sent to host
[   75.436837860,6] BT: seq 0x1e netfn 0x0a cmd 0x44: IPMI MSG done
[   75.439466772,6] IPMI: SEL: New event logged [ID : 02d]
[   75.442307465,6] BT: seq 0x0a netfn 0x00 cmd 0x02: Message sent to host
[   76.035894135,6] BT: seq 0x0a netfn 0x00 cmd 0x02: IPMI MSG done

--== Welcome to Hostboot  ==--

  2.73076|secure|SecureROM valid - enabling functionality
 13.83996|secure|Booting in non-secure mode.
 14.56856|Ignoring boot flags, incorrect version 0x0
 14.57840|Booting from SBE side 0 on master proc=00050000
 14.73354|ISTEP  6. 5 - host_init_fsi
 14.81271|ISTEP  6. 6 - host_set_ipl_parms
```
If there are multiple flash resources pre-load requests came, before one CPU job gets abort the other CPU can download the PNOR resource. @cclaudio @stewart-ibm Please have a look at it.

pridhiviraj, 2018-02-08:
Primary main thread took additional 6 seconds to abort the boot process, after secondary job gets abort.

stewart-ibm, 2018-02-08:
Ouch. We should probably think about a better way to deal with that.

debmc, 2019-02-07:
Adding cross reference -> https://bugzilla.linux.ibm.com/show_bug.cgi?id=165960