open-power/skiboot #151: pflash failed to do block level smart erase.

pridhiviraj, 2018-02-26:
```
[root@ltc-boston125 pflash]# pflash  --version
Open-Power Flash tool v5.10
[root@ltc-boston125 pflash]# pflash  -e -f -p fake.gard  -P GUARD
Failed to get file size: No such file or directory
[root@ltc-boston125 pflash]# cd
[root@ltc-boston125 ~]# pflash  -e -f -p fake.gard  -P GUARD
About to erase 0x0002c000..0x0002c03f !
Erasing...
[                                                  ] 0%Failed to blocklevel_smart_erase(): 8
[root@ltc-boston125 ~]# 
```

cyrilbur-ibm, 2018-02-26:
Interesting, can you please rerun with the -g flag to pflash. So:
`pflash  -g -e -f -p fake.gard  -P GUARD`
and also please
`ls fake.gard`

Thanks,
Cyril

pridhiviraj, 2018-02-27:
```
[root@xxxx ~]# pflash -g -e -f -p fake.gard -P GUARD
blocklevel_read: 0x0	0x7ffffe309950	0x30
blocklevel_raw_read: 0x0	0x7ffffe309950	0x30
FFS: Partition map size: 0x1000
blocklevel_read: 0x0	0x1a3c01c0	0x1000
blocklevel_raw_read: 0x0	0x1a3c01c0	0x1000
About to erase 0x0002c000..0x0002c03f !
Erasing...
[                                                  ] 0%blocklevel_smart_erase: 0x2c000	0x3f
blocklevel_smart_erase: preserving 0x20000..0x2c000
mtd_erase: dst: 0x20000, len: 0x10000
blocklevel_smart_erase: Early exit, everything was in one erase block
Failed to blocklevel_smart_erase(): 8
[root@xxxx ~]# ls fake.gard 
fake.gard
[root@xxxx ~]# ls fake.gard  -l
-rw-r--r--. 1 root root 63 Feb 26 06:18 fake.gard
[root@xxxx ~]# 
```

cyrilbur-ibm, 2018-02-27:
Thanks! I've posted the fix: https://lists.ozlabs.org/pipermail/skiboot/2018-February/010536.html

closed 2018-03-01 by stewart-ibm