open-power/skiboot #152: ECC: uncorrectable error: ffffffffffffffff ff when flashing fake GUARD without erasing.

pridhiviraj, 2018-02-26:
```
[root@ltc-xxxx ~]# pflash  -p fake.gard  -P GUARD
About to program "fake.gard" at 0x0002c000..0x0002c03f !
WARNING ! This will modify your HOST flash chip content !
Enter "yes" to confirm:yes
Programming & Verifying...
[=                                                 ] 0%
Updating actual size in partition header...
[root@ltc-xxx ~]# opal-gard list -all
ECC: uncorrectable error: ffffffffffffffff ff
 ID       | Error    | Type       | Path
ECC: uncorrectable error: ffffffffffffffff ff
-----------------------------------------------------------
 00000001 | 90000002 | Fatal      | /Sys0/Node0/Membuf5
ECC: uncorrectable error: ffffffffffffffff ff
===========================================================
libflash ecc invalid
[root@ltc-xxxx ~]# 

```

cyrilbur-ibm, 2018-02-26:
Yes, programming flash without erasing first will do these kinds of things. What is your expected behaviour here?

pridhiviraj, 2018-02-27:
@cyrilbur-ibm As we have #151 causing the problems to erase and flash, i just tried flashing without erasing to see whether that works or not :)

oohal, 2019-07-15:
not a bug

closed 2019-07-15 by oohal