open-power/skiboot #154: Ubuntu 17.10 boot fail

artemsen, 2018-02-27:
I'm trying to test PNOR 1.21-rc3 on our P8 servers.
I have found the commit "dt: add /cpus/ibm, powerpc-cpu-features device tree bindings" (https://github.com/open-power/skiboot/commit/7f4c8e8ce0b78ca046643d7f4f63d81f4fd11746) breaks Ubuntu 17.10 booting (kernel 4.13). I can't even boot to the net installer. The system hangs and reboots by the watchdog timeout from OpenBMC.
Meanwhile Ubuntu 16.04, Debian 8 and Fedora 27 work fine.

Here is parts of skiboot debug log:
```
...
[  101.359218047,6] CPU: P8 generation processor (max 8 threads/core)
[  101.359224110,7] CPU: Boot CPU PIR is 0x0858 PVR is 0x004d0200
[  101.359230412,7] CPU: Initial max PIR set to 0x1fff
[  101.368211291,7] OPAL table: 0x30103830 .. 0x30103d80, branch table: 0x30002000
[  101.368234487,7] Assigning physical memory map table for unused
[  101.417607342,7] FDT: Parsing fdt @0xff00000
[  101.442663745,7] CPUFT: creating cpufeatures for cpu:2 isa:2070
[  101.442700242,7] CPUFT:   'big-endian'
[  101.442714682,7] CPUFT:   'little-endian'
[  101.442730016,7] CPUFT:   'hypervisor'
[  101.442743005,7] CPUFT:   'interrupt-facilities'
[  101.442760912,7] CPUFT:   'smt'
[  101.442776985,7] CPUFT:   'program-priority-register'
[  101.442793866,7] CPUFT:   'virtual-page-class-key-protection'
[  101.442817985,7] CPUFT:   'strong-access-ordering'
[  101.442838232,7] CPUFT:   'no-execute'
[  101.442854381,7] CPUFT:   'cache-inhibited-large-page'
[  101.442873905,7] CPUFT:   'debug-facilities'
[  101.442890866,7] CPUFT:   'come-from-address-register'
[  101.442912995,7] CPUFT:   'branch-tracing'
[  101.442931130,7] CPUFT:   'floating-point'
[  101.442952737,7] CPUFT:   'vector'
[  101.442977324,7] CPUFT:   'vector-scalar'
[  101.443002205,7] CPUFT:   'vector-crypto'
[  101.443023149,7] CPUFT:   'quadword-load-store'
[  101.443041253,7] CPUFT:   'decimal-integer'
[  101.443055675,7] CPUFT:   'decimal-floating-point'
[  101.443076312,7] CPUFT:   'data-stream-control-register'
[  101.443104355,7] CPUFT:   'branch-history-rolling-buffer'
[  101.443127318,7] CPUFT:   'transactional-memory'
[  101.443156141,7] CPUFT:   'event-based-branch'
[  101.443182604,7] CPUFT:   'target-address-register'
[  101.443215231,7] CPUFT:   'control-register'
[  101.443233703,7] CPUFT:   'processor-control-facility'
[  101.443260470,7] CPUFT:   'processor-utilization-of-resources-register'
[  101.443296620,7] CPUFT:   'coprocessor-icswx'
[  101.443315974,7] CPUFT:   'mmu-hash'
[  101.443336744,7] CPUFT:   'machine-check-power8'
[  101.443361088,7] CPUFT:   'performance-monitor-power8'
[  101.443389190,7] CPUFT:   'idle-nap'
[  101.443407146,7] CPUFT:   'wait'
[  101.443428470,7] CPUFT:   'subcore'
[  101.443454860,7] CPUFT:   'timer-facilities'
[  101.443487209,7] CPUFT: feature branch-tracing has 1 dependencies (debug-facilities)
[  101.443501404,7] CPUFT:  branch-tracing found dep (debug-facilities)
[  101.443517112,7] CPUFT: feature come-from-address-register has 1 dependencies (debug-facilities)
[  101.443534504,7] CPUFT:  come-from-address-register found dep (debug-facilities)
[  101.443555526,7] CPUFT: feature decimal-floating-point has 1 dependencies (floating-point)
[  101.443572011,7] CPUFT:  decimal-floating-point found dep (floating-point)
[  101.443597870,7] CPUFT: feature subcore has 1 dependencies (smt)
[  101.443611189,7] CPUFT:  subcore found dep (smt)
[  101.443623841,7] CPUFT: feature vector has 1 dependencies (floating-point)
[  101.443636413,7] CPUFT:  vector found dep (floating-point)
[  101.443648946,7] CPUFT: feature vector-crypto has 1 dependencies (vector)
[  101.443665436,7] CPUFT:  vector-crypto found dep (vector)
[  101.443677869,7] CPUFT: feature vector-scalar has 1 dependencies (vector)
[  101.443694191,7] CPUFT:  vector-scalar found dep (vector)
...
[  225.195534366,7] OPAL: Returning CPU 0x0028
[  225.290254304,7] OPAL: CPU re-init with flags: 0x1
[  225.290688084,5] OPAL: Switch to big-endian OS
[  228.061870249,7] CPU: [00000008] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.062269232,7] CPU: [00000010] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.067394021,7] CPU: [00000020] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.072518399,7] CPU: [00000028] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.077642824,7] CPU: [00000030] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.082768047,7] CPU: [00000050] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.087893070,7] CPU: [00000068] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.093017598,7] CPU: [00000070] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.098157495,7] CPU: [00000408] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.103282159,7] CPU: [00000410] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.108406996,7] CPU: [00000418] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.113532405,7] CPU: [00000428] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.118657633,7] CPU: [00000448] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.123782703,7] CPU: [00000450] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.128907835,7] CPU: [00000460] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.134032539,7] CPU: [00000468] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.139172793,7] CPU: [00000810] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.144297691,7] CPU: [00000820] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.149422721,7] CPU: [00000828] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.154547777,7] CPU: [00000830] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.159673097,7] CPU: [00000848] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.164798102,7] CPU: [00000858] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.169922908,7] CPU: [00000860] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.175047709,7] CPU: [00000868] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.180187167,7] CPU: [00000c08] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.185312219,7] CPU: [00000c18] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.190437084,7] CPU: [00000c20] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.195561615,7] CPU: [00000c28] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.200686697,7] CPU: [00000c30] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.205811743,7] CPU: [00000c48] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.210937402,7] CPU: [00000c60] HID0 change 0x0000100000000000 -> 0x0000000000000000
[  228.216062369,7] CPU: [00000c68] HID0 change 0x0000100000000000 -> 0x0000000000000000
<END>
```
Note that the message about switching to the little-endian OS is not displayed.

npiggin, 2018-02-28:
Do you have kernel patch 3b7af5c0fd9631762d1c4d7b4cee76f571dd3c2c ?

You could try booting with dt_cpu_ftrs=off kernel command line.

I'm not sure what we can do to minimise breakage here unfortunately.

artemsen, 2018-02-28:
I can boot old Ubuntu 17.10 kernel with dt_cpu_ftrs=off option.
Newly installed Ubuntu (kernel 4.13.0-36-generic by Canonical) works fine, so the problem is resolved.
Thank you!

closed 2018-02-28 by artemsen

stewart-ibm, 2018-02-28:
@npiggin suggests that you may need linux 3b7af5c0fd9631762d1c4d7b4cee76f571dd3c2c

Now... seeing as that kernel already exists and we broke booting, we might want to reconsider something that we're doing so we don't break existing p8 kernels.

stewart-ibm, 2018-02-28:
(this Is what I get for half writing a response one day and sending it the next. I'll go get more :coffee: :)