open-power/skiboot #155: nx_compress_powernv: NX842 FIFO nodes are missing with latest skiboot

pridhiviraj, 2018-02-27:
```[Tue Feb 27 01:33:42 2018] nx_compress_powernv: NX842 FIFO nodes are missing
```

mikey, 2018-02-27:
That's a linux message, right?

pridhiviraj, 2018-02-27:
Yes

mpe, 2018-02-27:
Presumably related to 31e5e988632f `vas: Disable VAS/NX-842 on some P9 revisions` ?

mpe, 2018-02-27:
Yes, I think so. Which NX related device nodes appear in the device tree?

pridhiviraj, 2018-02-27:
@mpe Currently with latest skiboot. These are the nodes it appeared.
```
[root@ltc-xxxx device-tree]# find . -name *nx*
./xscom@623fc00000000/nx@2010000
./imc-counters/nest-nx-events
./imc-counters/nx
./xscom@603fc00000000/nx@2010000
```

pridhiviraj, 2018-02-27:
@mpe 
```
With old skiboot:
=======================
[root@ltc-boston125 device-tree]# ls ./xscom@623fc00000000/nx@2010000
compatible  ibm,842-high-fifo  ibm,842-normal-fifo  ibm,gzip-high-fifo  ibm,gzip-normal-fifo  name  phandle  reg
[root@ltc-boston125 device-tree]# ls ./xscom@623fc00000000/nx@2010000/ibm,842-high-fifo
compatible  lpid  name  phandle  pid  priority  rx-fifo-address  rx-fifo-size  tid
[root@ltc-boston125 device-tree]# ls ./xscom@623fc00000000/nx@2010000/ibm\,842-normal-fifo/
compatible  lpid  name  phandle  pid  priority  rx-fifo-address  rx-fifo-size  tid
[root@ltc-boston125 device-tree]# ls ./xscom@623fc00000000/nx@2010000/ibm\,gzip-normal-fifo/
compatible  lpid  name  phandle  pid  priority  rx-fifo-address  rx-fifo-size  tid

With new v5.10 skiboot:
====================
[root@ltc-boston125 ~]# cd /proc/device-tree/
[root@ltc-boston125 device-tree]# ls ./xscom@623fc00000000/nx@2010000
compatible  name  phandle  reg
```
So we are missing those ibm,842-* and ibm,gzip-* DT nodes with new skiboot.

sukadev, 2018-02-27:
@pridhiviraj @mpe 
Yes, we removed the nodes because they would not be useful on some versions of P9.
Should we fix the kernel to suppress the message? Or are you seeing some other behaviour besides the message/missing nodes?

pridhiviraj, 2018-02-28:
@sukadev I am just seeing the only that message in linux, other than that no issues.

mpe, 2018-02-28:
It's the xscom nodes that are causing the confusion I think.

```
# lsprop /proc/device-tree/xscom@603fc00000000/nx@2010000
compatible       "ibm,power9-nx"
```
```
static __init int nx842_powernv_init(void)
{
...
	for_each_compatible_node(dn, NULL, "ibm,power9-nx") {
		ret = nx842_powernv_probe_vas(dn);
```

So why did we leave the xscom nodes in the tree?
Or possibly the better question is why is Linux looking for the xscom nodes in the first place? That seems wrong.

hmyneni, 2018-02-28:
@mpe I followed the same way as on P8 - created nodes under xscom@*/nx@* and Linux looks these entries. But we want to have some error message in dmesg so that users know why nx842 crupto is not available. Otherwise, we can suppress this message and return  -ENODEV.

hegdevasant, 2020-11-18:
Not able to reproduce on fairly recent skiboot/kernel.
Device tree from my P9 system:
nx@2010000]# ls
compatible  ibm,842-high-fifo  ibm,842-normal-fifo  ibm,gzip-high-fifo  ibm,gzip-normal-fifo  name  phandle  reg

 Hence closing this issue.  Feel free to REOPEN if this issue is still exists.

-Vasant

closed 2020-11-18 by hegdevasant