open-power/skiboot #156: P8 Habanero system checkstops when booting ubuntu kernel with latest skiboot

pridhiviraj, 2018-02-27:
```
Petitboot (v1.6.6-p451cb3f)                          8348-21C 6856D5A
 ──────────────────────────────────────────────────────────────────────────────
  [Disk: sda2 / d90c198f-c07d-44dd-8cd6-04f37206dbf0]
    Red Hat Enterprise Linux Server (0-rescue-688d22361d1640eba1262d8897638a7b)
    Red Hat Enterprise Linux Server (3.10.0-693.el7.ppc64le) 7.4 (Maipo)
  [Disk: sdb2 / 32c2e9d6-7cb6-49ce-8379-d263f53fc9f7]
    Ubuntu, with Linux 4.4.0-101-generic (recovery mode)
    Ubuntu, with Linux 4.4.0-101-generic
    Ubuntu, with Linux 4.4.0-103-generic (recovery mode)
    Ubuntu, with Linux 4.4.0-103-generic
    Ubuntu, with Linux 4.10.0-35-generic (recovery mode)
    Ubuntu, with Linux 4.10.0-35-generic
    Ubuntu, with Linux 4.13.0-rc2 (recovery mode)
    Ubuntu, with Linux 4.13.0-rc2
 *  Ubuntu                                                                     

  System information
  System configuration
  System status log
  Language
  Rescan devices
 ──────────────────────────────────────────────────────────────────────────────
 Enter=accept, e=edit, n=new, x=exit, l=language, g=log, h=help
The system is going down NOW!
Sent SIGTERM to all processes
Sent SIGKILL to all processes
[   22.950507] kexec_core: Starting new kernel
[   87.450673294,5] OPAL: Switch to big-endian OS
  3.63773|Ignoring boot flags, incorrect version 0x0
  3.70636|ISTEP  6. 3 - host_init_fsi
  4.06821|ISTEP  6. 4 - host_set_ipl_parms
  4.06922|ISTEP  6. 5 - host_discover_targets
  8.85327|HWAS|PRESENT> DIMM[03]=AAAA0000AAAA0000
  8.85328|HWAS|PRESENT> Membuf[04]=CC00000000000000
  8.85328|HWAS|PRESENT> Proc[05]=8000000000000000
  8.99816|ISTEP  6. 6 - host_gard
 10.08523|================================================
 10.08524|Error reported by unknown (0xE500)
 10.08524|  <none>
 10.08524|  ModuleId   0x0b unknown
 10.08525|  ReasonCode 0xe540 unknown
 10.08525|  UserData1  unknown : 0x0005000000000101
 10.08525|  UserData2  unknown : 0x4241000500000000
 10.08526|User Data Section 0, type UD
 10.08526|  Subsection type 0x06
 10.08526|  ComponentId errl (0x0100)
 10.08526|  CALLOUT
 10.08527|  PROCEDURE ERROR
 10.08527|  Procedure: 16
 10.08527|User Data Section 1, type UD
 10.08527|  Subsection type 0x04
 10.08528|  ComponentId errl (0x0100)
 10.08528|User Data Section 2, type UD
 10.08528|  Subsection type 0x06
 10.08529|  ComponentId errl (0x0100)
 10.08529|  CALLOUT
 10.08529|  HW CALLOUT
 10.08529|  Reporting CPU ID: 16
 10.08529|  Called out entity:
 10.08530|User Data Section 3, type UD
 10.08530|  Subsection type 0x33
 10.08530|  ComponentId unknown (0xe500)
 10.08531|User Data Section 4, type UD
 10.08531|  Subsection type 0x01
 10.08531|  ComponentId unknown (0xe500)
 10.08531|  STRING
 10.08532|  
 10.08532|User Data Section 5, type UD
 10.08532|  Subsection type 0x15
 10.08532|  ComponentId hb-trace (0x3100)
 10.08533|User Data Section 6, type UD
 10.08533|  Subsection type 0x01
 10.08533|  ComponentId unknown (0xe500)
 10.08533|  STRING
 10.08534|  
 10.08534|User Data Section 7, type UD
 10.08534|  Subsection type 0x03
 10.08534|  ComponentId errl (0x0100)
 10.08535|User Data Section 8, type UD
 10.08535|  Subsection type 0x01
 10.08535|  ComponentId errl (0x0100)
 10.08535|  STRING
 10.08536|  Hostboot Build ID: hostboot-d3025f5/hbicore.bin
 10.08536|User Data Section 9, type UD
 10.08536|  Subsection type 0x04
 10.08537|  ComponentId errl (0x0100)
 10.08537|================================================
 14.92499|ISTEP  6. 7 - host_cancontinue_clear
 16.73371|ISTEP  6. 8 - proc_revert_sbe_mcs_setup
 16.78219|ISTEP  6. 9 - host_slave_sbe_config
 17.80684|ISTEP  6.10 - host_sbe_start
 17.80835|ISTEP  6.11 - proc_check_slave_sbe_seeprom_complete
 17.92845|ISTEP  6.12 - proc_cen_ref_clk_enable
 17.93904|ISTEP  6.13 - proc_xmit_sbe
 17.94026|ISTEP  7. 1 - proc_attr_update
 18.01698|ISTEP  7. 2 - proc_a_x_pci_dmi_pll_initf
 18.12264|ISTEP  7. 3 - proc_a_x_pci_dmi_pll_setup
 18.14369|ISTEP  7. 4 - proc_startclock_chiplets
 18.18272|ISTEP  7. 5 - proc_chiplet_scominit
 18.37856|ISTEP  7. 6 - proc_xbus_scominit
 18.41705|ISTEP  7. 7 - proc_abus_scominit
 18.42730|ISTEP  7. 8 - proc_pcie_scominit
 18.50089|ISTEP  7. 9 - proc_scomoverride_chiplets
 18.50166|ISTEP  8. 1 - fabric_erepair
 18.61959|ISTEP  8. 2 - fabric_io_dccal
 18.62085|ISTEP  8. 3 - fabric_pre_trainadv
 18.63115|ISTEP  8. 4 - fabric_io_run_training
 18.63188|ISTEP  8. 5 - fabric_post_trainadv
 18.63487|ISTEP  8. 6 - host_startprd_pbus
 18.65995|ISTEP  8. 7 - host_attnlisten_proc
 18.66093|ISTEP  8. 8 - proc_fab_iovalid
 18.67155|ISTEP  9. 1 - proc_build_smp
 19.01334|ISTEP  9. 2 - host_slave_sbe_update
 19.64092|ISTEP 10. 1 - host_prd_hwreconfig
 19.87117|ISTEP 10. 2 - cen_sbe_tp_chiplet_init1
 20.70842|ISTEP 10. 3 - cen_sbe_pll_initf
 20.70934|ISTEP 10. 4 - cen_sbe_pll_setup
 20.71019|ISTEP 10. 5 - cen_sbe_tp_chiplet_init2
 20.71293|ISTEP 10. 6 - cen_sbe_tp_arrayinit
 20.71404|ISTEP 10. 7 - cen_sbe_tp_chiplet_init3
 20.71640|ISTEP 10. 8 - cen_sbe_chiplet_init
 20.71740|ISTEP 10. 9 - cen_sbe_arrayinit
 20.71895|ISTEP 10.10 - cen_sbe_dts_init
 20.71970|ISTEP 10.11 - cen_sbe_initf
 20.72058|ISTEP 10.12 - cen_sbe_do_manual_inits
 20.72140|ISTEP 10.13 - cen_sbe_nest_startclocks
```

artemsen, 2018-02-27:
I think we have the same issue https://github.com/open-power/skiboot/issues/154

hegdevasant, 2020-11-18:
Based on previous comment I guess this is fixed. Hence closing . Feel free to REOPEN if you hit issue again.

-Vasant

closed 2020-11-18 by hegdevasant