open-power/skiboot #158: Failure to read flash

shenki, 2018-02-28:
This system has a problem where it fails to load anything but ff's from flash. While there's not much that skiboot could do, perhaps we could detect this and print a helpful message before aborting?

```
[   37.488023096,5] OPAL v5.10-3-g90e34127e356 starting...
[   37.488029257,7] initial console log level: memory 7, driver 5
[   37.488032834,6] CPU: P8 generation processor (max 8 threads/core)
[   37.488036104,7] CPU: Boot CPU PIR is 0x0028 PVR is 0x004d0200
[   37.488039621,7] CPU: Initial max PIR set to 0x1fff
[   37.488457398,7] OPAL table: 0x300f9630 .. 0x300f9b80, branch table: 0x30002000
[   37.488463353,7] Assigning physical memory map table for unused
[   37.488467797,7] FDT: Parsing fdt @0xff00000
[   37.489211974,3] DT: dt_attach_root failed, duplicate sensor@33
[   37.489216814,3] DT: Found duplicate node: sensor@33
[   37.490234335,6] CHIP: Initialised chip 0 from xscom@3fc0000000000
[   37.490283983,5] CHIP: Chip ID 0000 type: P8 DD2.0
[   37.490287533,7] XSCOM: Base address: 0x3fc0000000000
[   37.490296146,7] XSTOP: XSCOM addr = 0x2010c82, FIR bit = 31
[   37.490299974,6] MFSI 0:0: Initialized
[   37.490302530,6] MFSI 0:2: Initialized
[   37.490305064,6] MFSI 0:1: Initialized
[   37.490497865,5] LPC: LPC[000]: Initialized, access via XSCOM @0xb0020
[   37.490505900,7] LPC: Default bus on chip 0x0
[   37.490572317,6] MEM: parsing reserved memory from node /ibm,hostboot/reserved-memory
[   37.490589713,7] HOMER: Init chip 0
[   37.490592803,7]   PBA BAR0 : 0x0000000ffd800000
[   37.490596008,7]   PBA MASK0: 0x0000000000300000
[   37.490599084,7]   HOMER Image at 0xffd800000 size 4MB
[   37.490603194,7]   PBA BAR2 : 0x4000000ffda00000
[   37.490606386,7]   PBA MASK2: 0x0000000000000000
[   37.490609386,7]   SLW Image at 0xffda00000 size 1MB
[   37.490613225,7]   PBA BAR3 : 0x0000000fff800000
[   37.490616315,7]   PBA MASK3: 0x0000000000700000
[   37.490619297,7]   OCC Common Area at 0xfff800000 size 8MB
[   37.490761831,5] PLAT: Fixing up PSI BAR on chip 0 BAR=3fffe80000001
[   37.490792125,7] AST: PNOR LPC offset: 0x0e000000
[   37.490864687,5] PLAT: Using virtual UART
[   37.491135392,7] UART: Using LPC IRQ 4
[   37.493632053,5] PLAT: Detected Palmetto platform
[   37.493709655,5] PLAT: Detected BMC platform AMI
[   37.494248008,5] CENTAUR: Found centaur for chip 0x0 channel 1
[   37.494337047,5] CENTAUR:   FSI host: 0x0 cMFSI0 port 2
[   37.494604771,5] CPU: All 32 processors called in...
[   39.164168133,3] FFS: Reading the flash has returned all 0xFF.
[   39.164307097,3]      Are you reading erased flash?
[   39.164409433,3]      Is something else using the flash controller?
[   39.164512657,4] FLASH: No ffs info; using raw device only
[   39.164609471,5] BT: Interface initialized, IO 0x00e4
[   40.113582618,4] NVRAM: Query before is done loading
[   40.113663743,4] NVRAM: Waiting for load
[   40.113736824,2] NVRAM: Failed to load
[   40.113792399,4] NVRAM: Query before is done loading
[   40.113889356,4] NVRAM: Waiting for load
[   40.113953471,2] NVRAM: Failed to load
[   40.114010015,5] STB: Found ibm,secureboot-v1
[   40.114073771,4] NVRAM: Query before is done loading
[   40.114194206,4] NVRAM: Waiting for load
[   40.114251631,2] NVRAM: Failed to load
[   40.114314655,5] STB: secure mode off
[   40.115223874,4] NVRAM: Query before is done loading
[   40.115319247,4] NVRAM: Waiting for load
[   40.115368959,2] NVRAM: Failed to load
[   40.115432158,5] STB: trusted mode off
[   40.115487970,5] OPAL: Using OPAL UART console
[   40.115545914,4] NVRAM: Query before is done loading
[   40.115609106,4] NVRAM: Waiting for load
[   40.115668828,2] NVRAM: Failed to load
[   40.115800364,5] SLW: HB-provided idle states property found
[   40.115891540,5] SLW: Enabling: nap
[   40.115943668,5] SLW: Enabling: fastsleep_
[   40.116004269,5] SLW: Enabling: winkle
[... 121 lines cut ...]
[   43.130969202,3] GPR01: 0000000031ca3b80 GPR17: 0000000000000000
[   43.131108784,3] GPR02: 0000000030101600 GPR18: 0000000000000000
[   43.131200969,3] GPR03: 00000000305b6950 GPR19: 0000000000000000
[   43.131294257,3] GPR04: 0000000000000000 GPR20: 0000000000000000
[   43.131433394,3] GPR05: 0000000000000000 GPR21: 0000000000000000
[   43.131549178,3] GPR06: 0000000065504150 GPR22: 0000000000000000
[   43.131664841,3] GPR07: 0000000040000001 GPR23: 00000000300cb591
[   43.131804182,3] GPR08: 0000000030000000 GPR24: 00000000300b8362
[   43.131897015,3] GPR09: 00000000300050b0 GPR25: 0000000030301440
[   43.131988784,3] GPR10: 00000000000050b0 GPR26: 000000003031cc18
[   43.132105657,3] GPR11: 0000000000000000 GPR27: 0000000000000000
[   43.132220769,3] GPR12: 0000000040002808 GPR28: 00000000300b8201
[   43.132337603,3] GPR13: 0000000031ca0000 GPR29: 0000000040000000
[   43.132475925,3] GPR14: 0000000030002680 GPR30: 0000000030301600
[   43.132594081,3] GPR15: 0000000000000000 GPR31: 000000003031cbd8
[   43.132687590,0] Aborting!
CPU 0028 Backtrace:
 S: 0000000031ca3870 R: 0000000030013640   .backtrace+0x44
 S: 0000000031ca3910 R: 0000000030019e38   ._abort+0x4c
 S: 0000000031ca3990 R: 0000000030027b44   .exception_entry+0x18c
 S: 0000000031ca3b80 R: 0000000000001f10 * 
 S: 0000000031ca3d60 R: 0000000030014494   .load_and_boot_kernel+0xa30
 S: 0000000031ca3e30 R: 0000000030014bbc   .main_cpu_entry+0x614
 S: 0000000031ca3f00 R: 00000000300026e0   boot_entry+0x1b8
```

mikey, 2018-02-28:
In that case we could just check that the 32 bits at 0x20000000 isn't all 0s and bail out.

Something like this could work (untested):
```
diff --git a/core/init.c b/core/init.c
index 5633a9d38f..a84c562c40 100644
--- a/core/init.c
+++ b/core/init.c
@@ -405,6 +405,10 @@ static bool load_kernel(void)
 
 	if (kh->ei_ident != ELF_IDENT) {
 		prerror("INIT: ELF header not found. Assuming raw binary.\n");
+
+		/* Check there is something there before we branch to it */
+		assert(*(uint32_t *)kh != 0);
+
 		return true;
 	}
 
```

mikey, 2018-02-28:
How about this?
```
diff --git a/core/init.c b/core/init.c
index 5633a9d38f..c768dde9d1 100644
--- a/core/init.c
+++ b/core/init.c
@@ -405,6 +405,13 @@ static bool load_kernel(void)
 
 	if (kh->ei_ident != ELF_IDENT) {
 		prerror("INIT: ELF header not found. Assuming raw binary.\n");
+
+		/* Check there is something there before we branch to it */
+		if (*(uint32_t *)kh == 0) {
+			prerror("INIT: raw binary is zeros, can't execute!\n");
+			assert(0);
+		}
+
 		return true;
 	}
 
```
which gives this output:

```[    0.006650393,4] FLASH: Can't load resource id:0. No system flash found
[    0.006667359,4] FLASH: Can't load resource id:1. No system flash found
[    0.007001230,5] PCI: Resetting PHBs and training links...
[    0.007010769,5] PCI: Probing slots...
[    0.007183899,5] INIT: Waiting for kernel...
[    0.007188863,5] INIT: platform wait for kernel load failed
[    0.007194061,5] INIT: Assuming kernel at 0x20000000
[    0.007205006,3] INIT: ELF header not found. Assuming raw binary.
[    0.007210219,3] INIT: raw binary is zeros, can't execute!
[    0.007215088,0] Assert fail: core/init.c:412:0
[    0.007221311,0] Aborting!
CPU 0000 Backtrace:
 S: 0000000031c03ba0 R: 000000003001362c   .backtrace+0x48
 S: 0000000031c03c40 R: 0000000030019e64   ._abort+0x4c
 S: 0000000031c03cc0 R: 0000000030019ee0   .assert_fail+0x34
 S: 0000000031c03d40 R: 0000000030013c80   .load_and_boot_kernel+0x268
 S: 0000000031c03e30 R: 0000000030014bd0   .main_cpu_entry+0x610
 S: 0000000031c03f00 R: 00000000300026e0   boot_entry+0x1b8
```

mikey, 2018-02-28:
Posted patch: http://patchwork.ozlabs.org/patch/879478/

mikey, 2018-03-01:
Fixed in 18d7ee718bef3c95787473e3537be5f0653470c4

closed 2018-03-01 by mikey