open-power/skiboot #159: IPMI Partition Read fails with AMI BMC on Romulus/Zaius platforms.

klema1, 2018-03-09:
Romulus AMI BMC is receiving the following error from from the host when
attempting to perform “Read PNOR Partition” command for the VERSION
partition.
bash-4.1$ ipmitool -H 9.41.164.103 -I lanplus -U ADMIN -P admin raw 0x3a
0xb 0x56 0x45 0x52 0x53 0x49 0x4f 0x4e 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
0x0 0x0 0x0 0x0 0x0 -vvvv
IPMI Request Match found
Closed Session 579be1c0

Host messages:
The Related log in host debug console is shown as below:
[  299.380666738,3] BT: seq 0x3b netfn 0x0a cmd 0x49: Retry sending message
[  416.292496518,5] IPMI: PNOR access requested
[  416.292542395,3] IPMI: PNOR BUG: access requested but BMC doesn't support request
[  431.337019797,5] IPMI: PNOR access released

Jeremy Kerr responded:
"this means that the skiboot firmware running on that machine does not
have the configuration to tell it that the PNOR access commands are
implemented on this machine (and what netfn/cmd to use for that)

I'm assuming that you're using the upstream definitions for Romulus for
skiboot. This doesn't define which BMC implementation in use, so that's
why this would be failing.

You can probably fix this by defining the .bmc member of the platform
definition (see firestone.c for an AMI-BMC platform)."
++++++++++++++++++++++++++++++++++
A test PNOR image was built with the romulus.c file updated with 
Change line 47 from 
.bmc = NULL, /* FIXME: Add openBMC */ 
to
.bmc = &astbmc_ami, 

Read PNOR Partition was successfull!!!!!!

++++++++++++++++++++++++++++++++++++

Stewart Smith responded that Jeremy's solution not sufficient and provided the following as to why we need to differentiate open power BMCs:

I'd hoped we'd be able to stay BMC neutral too, and for the *most* part
we are.

The reasoning goes:
Way back on P8 with Palmetto/Habanero, some of the functionality we
needed to communicate with the BMC was decided to be done over IPMI. We
chose a IPMI command in the OEM space for this. When we came to do
Briggs/Stratton, we still needed this functionality, but the OEM command
we were already using was already used by SuperMicro for some of their
functionality, so we got a different one. With OpenBMC, it doesn't
support one of the commands, and there's no way to query that, so we
code it as non-implemented.

*Currently*, in skiboot, we hard code which BMC type it is per
platform... which was a quick solution to the original problem when we
did Briggs&Stratton. When OpenBMC work started, it wasn't a problem
because the functionality that's BMC specific isn't critical (creating
partial ESELs and the pnor access commands), this was fine as
neither of those bits of functionality are critical for bringup.
The way Hostboot dealt with this problem was that they hardcoded the
values, and SuperMicro carried a patch. Thus, it was not possible to
build a correctly functioning hostboot for habanero and briggs from
the same source tree.

But, seeing as we now have platforms that could have different BMC
implementations, we should probably move to a model where we query the
BMC for what type it is and make assumptions based on that.

My suggestion is that we do this using the 'mc info' IPMI command, and
then we just have a table of the BMCs we know about in skiboot and take
appropriate action. I'd expect the same strategy could be used inside
hostboot.

The BMC specific bits are kept to firmware, and we abstract away any
differences for Linux from within skiboot.

Currently, the actual per-BMC-type code is currently outrageously simple:

const struct bmc_platform astbmc_ami = {
        .name = "AMI",
        .ipmi_oem_partial_add_esel   = IPMI_CODE(0x32, 0xf0),
        .ipmi_oem_pnor_access_status = IPMI_CODE(0x3a, 0x07),
};
const struct bmc_platform astbmc_openbmc = {
        .name = "OpenBMC",
        .ipmi_oem_partial_add_esel   = IPMI_CODE(0x32, 0xf0),
};
static const struct bmc_platform astbmc_smc = {
        .name = "SMC",
        .ipmi_oem_partial_add_esel   = IPMI_CODE(0x3a, 0xf0),
        .ipmi_oem_pnor_access_status = IPMI_CODE(0x3a, 0x07),
};

and then each platform (p9dsu, habanero, romulus etc) just points to
which one they have."

++++++++++++++++++++++++++++++++++++++++++
AMI responded that it should not fill in Manufacturing ID and Name as follows:
"Regarding the solution using ipmi “mc info” to distinguish BMCs, our advice is not to use this solution as “mc info” might be different due to different customers’ platform."
and 
"Read partition command solution plan is currently to use "mc info" so AMI should fill in Manufacturing ID and Name. AMI has concern??
è Due to Manufactory ID will be changed by customer, that can’t be used for fixing the issue. And Read PNOR Partition command is IBM OEM Command. It should be supported in PNOR as default?"

++++++++++++++++++++++++++++++++++++++++++
Stewart Smith responded :

I was talking to Jeremy about this (added to thread) and for 'mc info'
it's meant to be information about the management controller, not the
vendor of the system. So... I'm not quite sure why this wouldn't be
populated.. It could be a configuration option for the vendor of the
particular system, but it seems odd that it's not populated at all.

Any which way, we need a *standard* way to probe for the BMC we're
talking to, and I'm not sure what else we have available other than the
'mc info' command. We can't poke in the OEM space as that could
literally do anything, so we're stuck with standard IPMI...
+++++++++++++++++++++++++++++++++++++

Can AMI and its customers have a version of PNOR with the .bmc line changed for Zaius and Romulus or make the default astbmc_ami similar to what was supported with P8 prior to OpenBMC and only change it if OpenBMC? 

Current romulus.c and zaius.c have 
.bmc = NULL, /* FIXME: Add openBMC */ 

AMI needs 
.bmc = &astbmc_ami,

stewart-ibm, 2018-03-13:
How do we detect that it's OpenBMC though?

klema1, 2018-03-16:
I thought once OpenBMC fills in the response to the "mc info" operation,   https://github.com/openbmc/openbmc/issues/2947,  this info could be used.  Unfortunately,  there has not been any updates to this issue.  

I do not now what values are expected for Manufacturing Name and ID for the different BMCs thus have not been able to recommend one for AMI to use.  

AMI also feels their customer, an ODM, will fill in these values so I do not know what values skiboot can use from the Manufacture Name or ID to identify the BMC as supporting the Read Partition ID command.

klema1, 2018-03-16:
I mistakenly closed.

klema1, 2018-03-16:
My bad again.

joseph-reynolds, 2018-05-16:
Stuart, Maury and myself discussed the 0x32/ 0x3a issue yesterday.
The thinking was that all of the BMCs now support IPMI code 0x3a, so there is no need to use 0x32, so skiboot can change the default to 0x3a.
Stuart sent me a patch to test on its was upstream.

stewart-ibm, 2018-05-24:
https://patchwork.ozlabs.org/patch/913997/ is that patch.

stewart-ibm, 2018-05-24:
(and as such I think it's okay to close this issue - but the BMC request for PNOR isn't really a command that should be being sent on P9 systems with mbox)

closed 2018-05-24 by stewart-ibm