open-power/skiboot #160: hdata/i2c.c fails to build with gcc 8

shenki, 2018-03-14:
```
hdata/i2c.c:184:1: error: alignment 1 of ‘struct host_i2c_hdr’ is less than 4 [-Werror=packed-not-aligned]
 } __packed;
 ^
```

```
$ powerpc64-linux-gnu-gcc-8 --version
powerpc64-linux-gnu-gcc-8 (Ubuntu 8-20180218-1ubuntu1) 8.0.1 20180218 (experimental) [trunk revision 257787]
```

sharkcz, 2018-05-16:
It could be a GCC bug, but following could be a workaround (we see the same problem on Fedora with eg. gcc-8.1.1-1.fc28.ppc64le)
````
diff -up skiboot-5.10.2/hdata/i2c.c.me skiboot-5.10.2/hdata/i2c.c
--- skiboot-5.10.2/hdata/i2c.c.me       2018-05-14 12:23:30.413536519 +0200
+++ skiboot-5.10.2/hdata/i2c.c  2018-05-14 12:26:27.036752163 +0200
@@ -181,7 +181,7 @@ static bool is_zeros(const void *p, size
 struct host_i2c_hdr {
        const struct HDIF_array_hdr hdr;
        __be32 version;
-} __packed;
+} __packed __align(0x4);

 int parse_i2c_devs(const struct HDIF_common_hdr *hdr, int idata_index,
        struct dt_node *xscom)
````

oohal, 2018-05-17:
Tbh we should probably just remove the __align() attributes entirely. If we want to enforce alignment of structures it should probably be done in code so we can print errors to flag alignment problems, etc.

closed 2018-05-29 by stewart-ibm