open-power/skiboot #170: Seeing softlockups in kernel for OPAL RTC API calls in OpenBMC based systems.

pridhiviraj, 2018-04-04:
```
======================================================================
ERROR [87.594s]: runTest (testcases.OpalGard.OpalGard)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jenkins_data/sath/pridhivi/op-test-framework/testcases/OpalGard.py", line 75, in tearDown
    res = self.c.run_command_ignore_fail(cmd, timeout=120)
  File "/home/jenkins_data/sath/pridhivi/op-test-framework/common/OpTestSSH.py", line 305, in run_command_ignore_fail
    output = self.run_command(command, timeout)
  File "/home/jenkins_data/sath/pridhivi/op-test-framework/common/OpTestSSH.py", line 269, in run_command
    rc = console.expect([expect_prompt, r"[Pp]assword", pexpect.TIMEOUT, pexpect.EOF], timeout=timeout)
  File "/home/jenkins_data/sath/pridhivi/op-test-framework/common/OPexpect.py", line 121, in expect
    raise KernelCrashUnknown(state, log)
KernelCrashUnknown: Kernel crash unknown state (machine in state 6): INFO: rcu_sched self-detected stall on CPU
[Wed Apr  4 10:00:09 2018] 	72-....: (5249 ticks this GP) idle=9f2/140000000000001/0 softirq=4838/4838 fqs=2624 
[Wed Apr  4 10:00:09 2018] 	 (t=5250 jiffies g=10079 c=10078 q=792)
[Wed Apr  4 10:00:09 2018] NMI backtrace for cpu 72
[Wed Apr  4 10:00:09 2018] CPU: 72 PID: 931 Comm: kworker/72:1 Not tainted 4.15.0-13-generic #14-Ubuntu
[Wed Apr  4 10:00:09 2018] Workqueue: events rtc_timer_do_work
[Wed Apr  4 10:00:09 2018] Call Trace:
[Wed Apr  4 10:00:09 2018] [c000201caa94f240] [c000000000ce91bc] dump_stack+0xb0/0xf4 (unreliable)
[Wed Apr  4 10:00:09 2018] [c000201caa94f280] [c000000000cf2ba8] nmi_cpu_backtrace+0x1f8/0x200
[Wed Apr  4 10:00:09 2018] [c000201caa94f310] [c000000000cf2d48] nmi_trigger_cpumask_backtrace+0x198/0x1f0
[Wed Apr  4 10:00:09 2018] [c000201caa94f3a0] [c00000000002f2d8] arch_trigger_cpumask_backtrace+0x28/0x40
[Wed Apr  4 10:00:09 2018] [c000201caa94f3c0] [c0000000001a7abc] rcu_dump_cpu_stacks+0xf4/0x158
[Wed Apr  4 10:00:09 2018] [c000201caa94f410] [c0000000001a6b68] rcu_check_callbacks+0x8e8/0xb40
[Wed Apr  4 10:00:09 2018] [c000201caa94f540] [c0000000001b4e28] update_process_times+0x48/0x90
[Wed Apr  4 10:00:09 2018] [c000201caa94f570] [c0000000001ccb54] tick_sched_handle.isra.5+0x34/0xd0
[Wed Apr  4 10:00:09 2018] [c000201caa94f5a0] [c0000000001ccc50] tick_sched_timer+0x60/0xe0
[Wed Apr  4 10:00:09 2018] [c000201caa94f5e0] [c0000000001b59d4] __hrtimer_run_queues+0x144/0x370
[Wed Apr  4 10:00:09 2018] [c000201caa94f660] [c0000000001b692c] hrtimer_interrupt+0xfc/0x350
[Wed Apr  4 10:00:09 2018] [c000201caa94f730] [c0000000000248f0] __timer_interrupt+0x90/0x260
[Wed Apr  4 10:00:09 2018] [c000201caa94f780] [c000000000024d08] timer_interrupt+0x98/0xe0
[Wed Apr  4 10:00:09 2018] [c000201caa94f7b0] [c000000000009014] decrementer_common+0x114/0x120
[Wed Apr  4 10:00:09 2018] --- interrupt: 901 at opal_get_rtc_time+0x98/0x110
                               LR = opal_return+0x14/0x48
[Wed Apr  4 10:00:09 2018] [c000201caa94faa0] [c000000000a506d8] opal_get_rtc_time+0x98/0x110 (unreliable)
[Wed Apr  4 10:00:09 2018] [c000201caa94fae0] [c000000000a4a8ac] __rtc_read_time+0x7c/0x180
[Wed Apr  4 10:00:09 2018] [c000201caa94fb60] [c000000000a4c658] rtc_timer_do_work+0x78/0x250
[Wed Apr  4 10:00:09 2018] [c000201caa94fc90] [c000000000131618] process_one_work+0x298/0x5a0
[Wed Apr  4 10:00:09 2018] [c000201caa94fd20] [c0000000001319b8] worker_thread+0x98/0x630
[Wed Apr  4 10:00:09 2018] [c000201caa94fdc0] [c00000000013a5e8] kthread+0x1a8/0x1b0
[Wed Apr  4 10:00:09 2018] [c000201caa94fe30] [c00000000000b528] ret_from_kernel_thread+0x5c/0xb4
[Wed Apr  4 10:00:15 2018] Modules linked in: ofpart cmdlinepart idt_89hpesx powernv_flash mtd at24 uio_pdrv_genirq uio ipmi_powernv ipmi_devintf ipmi_msghandler ibmpowernv opal_prd vmx_crypto crct10dif_vpmsum sch_fq_codel ip_tables x_tables autofs4 mlx5_ib ib_core nouveau ast i2c_algo_bit mlx5_core ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ahci mlxfw crc32c_vpmsum drm tg3 libahci devlink
[Wed Apr  4 10:00:15 2018] CPU: 72 PID: 931 Comm: kworker/72:1 Not tainted 4.15.0-13-generic #14-Ubuntu
[Wed Apr  4 10:00:15 2018] Workqueue: events rtc_timer_do_work
[Wed Apr  4 10:00:15 2018] NIP:  c000000000a506d8 LR: c00000000009d0dc CTR: 000000003003da24
[Wed Apr  4 10:00:15 2018] REGS: c000201caa94f820 TRAP: 0901   Not tainted  (4.15.0-13-generic)
[Wed Apr  4 10:00:15 2018] MSR:  9000000000009033 <SF,HV,EE,ME,IR,DR,RI,LE>  CR: 44002482  XER: 20040000
[Wed Apr  4 10:00:15 2018] CFAR: c00000000009d0c8 SOFTE: 1 
                           GPR00: 0000000030005164 c000201caa94faa0 c0000000016eaf00 0000000000000000 
                           GPR04: 0000000044002482 c000000000a506d8 9000000000009033 0000000033c00050 
                           GPR08: 0000000000000000 0000000000000000 c00000000009d0f0 9000000000001003 
                           GPR12: c00000000009d0c8 c00000000fab1800 
[Wed Apr  4 10:00:15 2018] NIP [c000000000a506d8] opal_get_rtc_time+0x98/0x110
[Wed Apr  4 10:00:15 2018] LR [c00000000009d0dc] opal_return+0x14/0x48
[Wed Apr  4 10:00:15 2018] Call Trace:
[Wed Apr  4 10:00:15 2018] [c000201caa94faa0] [c000000000a506d8] opal_get_rtc_time+0x98/0x110 (unreliable)
[Wed Apr  4 10:00:15 2018] [c000201caa94fae0] [c000000000a4a8ac] __rtc_read_time+0x7c/0x180
[Wed Apr  4 10:00:15 2018] [c000201caa94fb60] [c000000000a4c658] rtc_timer_do_work+0x78/0x250
[Wed Apr  4 10:00:15 2018] [c000201caa94fc90] [c000000000131618] process_one_work+0x298/0x5a0
[Wed Apr  4 10:00:15 2018] [c000201caa94fd20] [c0000000001319b8] worker_thread+0x98/0x630
[Wed Apr  4 10:00:15 2018] [c000201caa94fdc0] [c00000000013a5e8] kthread+0x1a8/0x1b0
[Wed Apr  4 10:00:15 2018] [c000201caa94fe30] [c00000000000b528] ret_from_kernel_thread+0x5c/0xb4
[Wed Apr  4 10:00:15 2018] Instruction dump:
[Wed Apr  4 10:00:15 2018] 38810020 38610028 7d3f07b4 4b64cd41 60000000 2ebf0000 393fffff 2fa3fff4 
[Wed Apr  4 10:00:15 2018] 2f23fffe 409effc4 38600000 4b64d061 <60000000> 4bffff90 3860000a 4b764075 
[console-pexpect]#[  835.303855094,3] BT: seq 0x72 netfn 0x0a cmd 0x48: Timeout sending message
<class 'pexpect.TIMEOUT'>
```

mikey, 2018-04-05:
This looks like we are taking an NMI caused by RCU while in skiboot which is the same as we've seen elsewhere. I'm guessing linux needs 
> 47712a921b Nicholas Piggin    powerpc/watchdog: remove arch_trigger_cpumask_backtrace

and also this patch
http://patchwork.ozlabs.org/patch/886688/

npiggin, 2018-04-11:
Yeah we need those patches, although they are tackling the symptoms of the problem.

We also have the RTC timing out. The OPAL RTC driver in Linux likes to provoke these lockups when the firmware returns errors. Trying to work on improving that situation as well.

https://patchwork.ozlabs.org/patch/896614/

And finally, is there a reason why the RTC is failing? Is it a stress test or delibrate reboot of the BMC?

stewart-ibm, 2018-09-18:
I think we've ended up in a better state now with Nick's patches merged and the bugs fixed... leaving only the problem of why we went off into the weeds talking to the BMC. I'll close this particular issue for now, and let's re-open it or open a new one if we get a test case to reproduce that behavior.

closed 2018-09-18 by stewart-ibm