open-power/skiboot #172: LPC[000]: Got SYNC no-response error. Error address reg: 0xf004202c

pridhiviraj, 2018-04-05:
After a kernel crash OPAL floods the console with this LPC no-response errors on OpenBMC systems.
```
[console-pexpect]#[17582052845.989499] Oops: Kernel access of bad area, sig: 7 [#1]
[17582052845.993305] LE SMP NR_CPUS=2048 NUMA PowerNV
[17582052845.997756] Modules linked in: vhost_net vhost tap xt_CHECKSUM iptable_mangle ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack libcrc32c ipt_REJECT nf_reject_ipv4 xt_tcpudp bridge stp llc ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter kvm_hv kvm idt_89hpesx vmx_crypto at24 ofpart uio_pdrv_genirq uio cmdlinepart ipmi_powernv ipmi_devintf ipmi_msghandler powernv_flash ibmpowernv opal_prd crct10dif_vpmsum mtd sch_fq_codel jc42 ip_tables x_tables autofs4 mlx5_ib ib_core nouveau ast mlx5_core i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ahci mlxfw crc32c_vpmsum drm tg3 libahci
[17582052845.997756] Modules linked in: vhost_net vhost tap xt_CHECKSUM iptable_mangle ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack libcrc32c ipt_REJECT nf_reject_ipv4 xt_tcpudp bridge stp llc ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter kvm_hv kvm idt_89hpesx vmx_crypto at24 ofpart uio_pdrv_genirq uio cmdlinepart ipmi_powernv ipmi_devintf ipmi_msghandler powernv_flash ibmpowernv opal_prd crct10dif_vpmsum mtd sch_fq_codel jc42 ip_tables x_tables autofs4 mlx5_ib ib_core nouveau ast mlx5_core i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ahci mlxfw crc32c_vpmsum drm tg3 libahci
[52479.130024] KVM: CPU 69 seems to be stuck
[52479.130024] KVM: CPU 69 seems to be stuck
[17582052845.998534]  devlink
[17582052845.998534]  devlink
[17582052845.998559] CPU: 69 PID: 0 Comm: swapper/69 Not tainted 4.15.0-13-generic #14-Ubuntu
[17582052845.998559] CPU: 69 PID: 0 Comm: swapper/69 Not tainted 4.15.0-13-generic #14-Ubuntu
[17582052845.998583] NIP:  c0000000000ec87c LR: c0000000000ec6ac CTR: 000000000000006c
[17582052845.998583] NIP:  c0000000000ec87c LR: c0000000000ec6ac CTR: 000000000000006c
[17582052845.998595] REGS: c00000003fcc3c70 TRAP: 0300   Not tainted  (4.15.0-13-generic)
[17582052845.998595] REGS: c00000003fcc3c70 TRAP: 0300   Not tainted  (4.15.0-13-generic)
[17582052845.998596] MSR:  9000000002803033 <SF,HV,VEC,VSX,FP,ME,IR,DR,RI,LE>  CR: 44400888  XER: 00000000
[17582052845.998596] MSR:  9000000002803033 <SF,HV,VEC,VSX,FP,ME,IR,DR,RI,LE>  CR: 44400888  XER: 00000000
[17582052845.998711] CFAR: c0000000000ec818 DAR: 0006030203190010 DSISR: 02000002 SOFTE: 0 
[17582052845.998711] GPR00: 0000000000000001 c00000003fcc3ef0 c0000000016eaf00 007fffffffffeb25 
[17582052845.998711] GPR04: c0000017208d5540 c0000016d4185900 0000000000000000 ffffffffffffffff 
[17582052845.998711] GPR08: 0040000003d4f41f 0000000000000010 0006030203190000 ffff00ff0000ff00 
[17582052845.998711] GPR12: 0000000000000e80 c00000000faaf700 c00000000000d3b8 0000000000000000 
[17582052845.998711] GPR16: 0000000000000000 0000000000000000 0000000000000000 0000000000000000 
[17582052845.998711] GPR20: 0000000000000000 0000000000000000 0000000000000000 0000000000000000 
[17582052845.998711] GPR24: c000000001134688 c0000000010b3680 c000000001095b70 0000000000000005 
[17582052845.998711] GPR28: c000000001a49df8 c0000000016d8180 c000000001a49df0 c00000000162a300 
[17582052845.998711] CFAR: c0000000000ec818 DAR: 0006030203190010 DSISR: 02000002 SOFTE: 0 
[17582052845.998711] GPR00: 0000000000000001 c00000003fcc3ef0 c0000000016eaf00 007fffffffffeb25 
[17582052845.998711] GPR04: c0000017208d5540 c0000016d4185900 0000000000000000 ffffffffffffffff 
[17582052845.998711] GPR08: 0040000003d4f41f 0000000000000010 0006030203190000 ffff00ff0000ff00 
[17582052845.998711] GPR12: 0000000000000e80 c00000000faaf700 c00000000000d3b8 0000000000000000 
[17582052845.998711] GPR16: 0000000000000000 0000000000000000 0000000000000000 0000000000000000 
[17582052845.998711] GPR20: 0000000000000000 0000000000000000 0000000000000000 0000000000000000 
[17582052845.998711] GPR24: c000000001134688 c0000000010b3680 c000000001095b70 0000000000000005 
[17582052845.998711] GPR28: c000000001a49df8 c0000000016d8180 c000000001a49df0 c00000000162a300 
[17582052845.999059] NIP [c0000000000ec87c] kvmppc_got_guest+0x33c/0x358
[17582052845.999059] NIP [c0000000000ec87c] kvmppc_got_guest+0x33c/0x358
[17582052845.999061] LR [c0000000000ec6ac] kvmppc_got_guest+0x16c/0x358
[17582052845.999061] LR [c0000000000ec6ac] kvmppc_got_guest+0x16c/0x358
[17582052845.999062] Call Trace:
[17582052845.999062] Call Trace:
[17582052845.999063] Instruction dump:
[17582052845.999063] Instruction dump:
[17582052845.999081] 7d1e4ba6 4c00012c 7c764aa6 60000000 2c230200 418001a4 e94d0948 282a0000 
[17582052845.999081] 7d1e4ba6 4c00012c 7c764aa6 60000000 2c230200 418001a4 e94d0948 282a0000 
[17582052845.999098] 4182002c e9641ca0 39200010 7c0006ac <7d6957ea> 81641c98 39200018 7d69572a 
[17582052845.999098] 4182002c e9641ca0 39200010 7c0006ac <7d6957ea> 81641c98 39200018 7d69572a 
[17582052845.999103] ---[ end trace d7bec06b0361fa31 ]---
[17582052845.999103] ---[ end trace d7bec06b0361fa31 ]---
[52599.106899414,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf004202c
[52599.106906294,6] IPMI: dropping non severe PEL event
[52599.108915426,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf0042608
[52599.108919537,6] IPMI: dropping non severe PEL event
[52599.109865866,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf0042870
[52599.109917375,6] IPMI: dropping non severe PEL event
[52599.112134341,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf0042f9c
[52599.112185284,6] IPMI: dropping non severe PEL event
[52599.114931198,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf00434d8
[52599.114988842,6] IPMI: dropping non severe PEL event
[52599.116387867,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf00437ac
[52599.116449408,6] IPMI: dropping non severe PEL event
[52599.116949056,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf0043988
[52599.116953162,6] IPMI: dropping non severe PEL event
[52599.117065314,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf00439bc
[52599.117069302,6] IPMI: dropping non severe PEL event
[52599.117144949,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf00439d0
[52599.117148991,6] IPMI: dropping non severe PEL event
[52599.117239574,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf00439f8
[52599.117243513,6] IPMI: dropping non severe PEL event
[... 159 lines cut ...]
[52600.188541192,6] IPMI: dropping non severe PEL event
[52600.188735332,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ad26c
[52600.188739369,6] IPMI: dropping non severe PEL event
[52600.188909943,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ad608
[52600.188914059,6] IPMI: dropping non severe PEL event
[52600.189038257,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ad84c
[52600.189042314,6] IPMI: dropping non severe PEL event
[52600.189158758,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ada34
[52600.189162823,6] IPMI: dropping non severe PEL event
[52600.189285226,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04adc64
[52600.189289391,6] IPMI: dropping non severe PEL event
[52600.189427834,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04adf10
[52600.189431786,6] IPMI: dropping non severe PEL event
[52600.189609562,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ae2f0
[52600.189613678,6] IPMI: dropping non severe PEL event
[52600.189735779,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ae520
[52600.189739866,6] IPMI: dropping non severe PEL event
[52600.189874036,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ae7ac
[52600.189878105,6] IPMI: dropping non severe PEL event
[52600.190007773,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ae9c0
[52600.190011730,6] IPMI: dropping non severe PEL event
[52600.190093355,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04ae9f4
[52600.190097262,6] IPMI: dropping non severe PEL event
[52600.190177125,3] LPC[000]: Got SYNC no-response error. Error address reg: 0xf04aea24
```

amboar, 2019-07-15:
@pridhiviraj any idea what was happening to the BMC at the time? This looks like the LPC2AHB bridge died somehow.

amboar, 2019-07-15:
Actually, given the addresses mentioned, it might be a skiboot bug as the PNOR (hiomap or not) should be mapped at the top of the LPC FW address space. Was this on P8 or P9?

hegdevasant, 2019-07-15:
I think I have seen few reports related "LPC SYNC no-response error" on P9. But I don't have a way to reproduce this issue.

Between, FYI @pridhiviraj left IBM few months ago. So if we are not able to reproduce then best thing we can do is to close this issue.

-Vasant

amboar, 2019-07-16:
@hegdevasant they can occur for a variety of reasons. A common one recently was boot progress updates on port 80/81/82. Given @pridhiviraj's departure, the lack of information and the age of the issue I suggest we close this for the time being, reopen it if we encounter the same errors, and open separate issues for other no-response failures (based on the failing address).

closed 2019-07-16 by amboar