open-power/skiboot #176: PCIe to PCI bridge will not train

madscientist159, 2018-05-11:
We've run across an interesting device that's failing to train on POWER9.  It's a PCIe to PCI bridge built around the PI7C9X130 chip, it looks solidly constructed and appears to be powered up correctly, but skiboot fails to train the link:

```
[   58.229791464,5] PCI: Resetting PHBs and training links...
[   60.243078664,3] PHB#0001[0:1]: LINK: Timeout waiting for link up
[   62.447106167,3] PHB#0001[0:1]: LINK: Timeout waiting for link up
[   65.139140552,3] PHB#0001[0:1]: LINK: Timeout waiting for link up
[   67.343171186,3] PHB#0001[0:1]: LINK: Timeout waiting for link up
[   67.343232661,3] PHB#0001[0:1]: Electrical link detected but won't train
[   67.343692522,5] PCI: Probing slots...
```

What additional information do you need to debug?

mikey, 2018-05-11:
Can you try a trace with this?
`nvram -p ibm,skiboot --update-config pci-tracing=true`
Reboot and then send:
`grep PHB#0001 /sys/firmware/opal/msglog`

madscientist159, 2018-05-11:
@mikey If I'm reading this right, it looks like it's trying to train at x16 when the card is much narrower; it only has a x1 edge connector.

```
[   63.130314433,7] PHB#0001:00:00.0 PCI: Registered PHB
[   63.130316302,6] PHB#0001[0:1]: Found /pciex@600c3c0100000 @0x600c3c0100000
[   63.130317860,6] PHB#0001[0:1]:   M32 [0x000600c080000000..0x000600c0ffffffff]
[   63.130319118,6] PHB#0001[0:1]:   M64 [0x0006004000000000..0x0006007fffffffff]
[   63.130320763,7] PHB#0001[0:1]: Base location code not found !
[   63.130322481,7] PHB#0001[0:1]: Core revision 0xa40002
[   63.130323545,7] PHB#0001[0:1]: Found 256 max PEs and 2048 IRQs 
[   63.130324697,6] PHB#0001[0:1]: Max link speed: GEN4
[   63.130325625,7] PHB#0001[0:1]: Override lane equalization settings:
[   63.130326584,7] PHB#0001[0:1]:   0x7777777777777777 0x7777777777777777
[   63.130327721,7] PHB#0001[0:1]:   0x7777777777777777 0x7777777777777777
[   63.130328926,7] PHB#0001[0:1]:   0x7777777777777777 0x7777777777777777
[   63.130434438,7] PHB#0001[0:1]: Initializing PHB4...
[   63.130435538,7] PHB#0001[0:1]: Default system config: 0x4210000040200000
[   63.130436705,7] PHB#0001[0:1]: New system config    : 0x4210000040200000
[   63.130438059,7] PHB#0001[0:1]: Initial PHB CRESET is 0xe000000000000000
[   63.130439524,7] PHB#0001[0:1]: Waiting for DLP PG reset to complete...
[   63.130959693,7] PHB#0001[0:1]: Reset state SRC_ID: 0ff8000000000000
[   63.131163400,7] PHB#0001[0:1]: Initialization complete
[   63.149275339,7] PHB#0001[0:1]: FRESET: Assert skipped
[   63.149278629,7] PHB#0001[0:1]: FRESET: Deassert
[   63.149284181,3] PHB#0001[0:1]: TRACE:0x0000001101000000  0ms          GEN1:x16:detect
[   63.155925383,3] PHB#0001[0:1]: TRACE:0x0000102101000000 12ms presence GEN1:x16:polling
[   63.156515174,3] PHB#0001[0:1]: TRACE:0x0000183101000000 14ms training GEN1:x16:config
[   63.156589677,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 14ms training GEN1:x01:recovery
[   63.168857402,3] PHB#0001[0:1]: TRACE:0x0000001011000000 38ms          GEN1:x01:detect
[   63.168990387,3] PHB#0001[0:1]: TRACE:0x0000001101000000 38ms          GEN1:x16:detect
[   63.175284963,3] PHB#0001[0:1]: TRACE:0x0000102101000000 50ms presence GEN1:x16:polling
[   63.175366862,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 50ms training GEN1:x01:recovery
[   63.187661800,3] PHB#0001[0:1]: TRACE:0x0000001011000000 74ms          GEN1:x01:detect
[   63.187795136,3] PHB#0001[0:1]: TRACE:0x0000001101000000 75ms          GEN1:x16:detect
[   63.194089772,3] PHB#0001[0:1]: TRACE:0x0000102101000000 87ms presence GEN1:x16:polling
[   63.194171702,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 87ms training GEN1:x01:recovery
[   63.206466950,3] PHB#0001[0:1]: TRACE:0x0000001011000000 111ms          GEN1:x01:detect
[   63.206600132,3] PHB#0001[0:1]: TRACE:0x0000001101000000 111ms          GEN1:x16:detect
[   63.212894763,3] PHB#0001[0:1]: TRACE:0x0000102101000000 124ms presence GEN1:x16:polling
[   63.212977331,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 124ms training GEN1:x01:recovery
[   63.225271600,3] PHB#0001[0:1]: TRACE:0x0000001011000000 148ms          GEN1:x01:detect
[   63.225405238,3] PHB#0001[0:1]: TRACE:0x0000001101000000 148ms          GEN1:x16:detect
[   63.231700017,3] PHB#0001[0:1]: TRACE:0x0000102101000000 160ms presence GEN1:x16:polling
[   63.231781853,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 161ms training GEN1:x01:recovery
[   63.244076818,3] PHB#0001[0:1]: TRACE:0x0000001011000000 185ms          GEN1:x01:detect
[   63.244210227,3] PHB#0001[0:1]: TRACE:0x0000001101000000 185ms          GEN1:x16:detect
[   63.250504816,3] PHB#0001[0:1]: TRACE:0x0000102101000000 197ms presence GEN1:x16:polling
[   63.250577429,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 197ms training GEN1:x01:recovery
[   63.262881402,3] PHB#0001[0:1]: TRACE:0x0000001011000000 221ms          GEN1:x01:detect
[   63.263015191,3] PHB#0001[0:1]: TRACE:0x0000001101000000 222ms          GEN1:x16:detect
[   63.269309192,3] PHB#0001[0:1]: TRACE:0x0000102101000000 234ms presence GEN1:x16:polling
[   63.269385319,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 234ms training GEN1:x01:recovery
[   63.281686597,3] PHB#0001[0:1]: TRACE:0x0000001011000000 258ms          GEN1:x01:detect
[   63.281819687,3] PHB#0001[0:1]: TRACE:0x0000001101000000 258ms          GEN1:x16:detect
[   63.288114344,3] PHB#0001[0:1]: TRACE:0x0000102101000000 271ms presence GEN1:x16:polling
[   63.288202783,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 271ms training GEN1:x01:recovery
[   63.300491094,3] PHB#0001[0:1]: TRACE:0x0000001011000000 295ms          GEN1:x01:detect
[   63.300624778,3] PHB#0001[0:1]: TRACE:0x0000001101000000 295ms          GEN1:x16:detect
[   63.306919225,3] PHB#0001[0:1]: TRACE:0x0000102101000000 307ms presence GEN1:x16:polling
[   63.307010490,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 308ms training GEN1:x01:recovery
[   63.319296090,3] PHB#0001[0:1]: TRACE:0x0000001011000000 332ms          GEN1:x01:detect
[   63.319429203,3] PHB#0001[0:1]: TRACE:0x0000001101000000 332ms          GEN1:x16:detect
[   63.325723980,3] PHB#0001[0:1]: TRACE:0x0000102101000000 344ms presence GEN1:x16:polling
[   63.325812501,3] PHB#0001[0:1]: TRACE:0x00001c5811000000 344ms training GEN1:x01:recovery
[   63.338100788,3] PHB#0001[0:1]: TRACE:0x0000001011000000 368ms          GEN1:x01:detect
[   63.338234387,3] PHB#0001[0:1]: TRACE:0x0000001101000000 369ms          GEN1:x16:detect
[... cut ...]
[   84.221970100,7] PHB#0001:00:00.0 Error -6 resetting
[   84.222837386,7] PHB#0001:00:00.0 Link down
[   84.222846966,7] PHB#0001:00:00.0 Scanning (upstream only)...
[   84.222879762,7] PHB#0001:00:00.0 Found VID:1014 DEV:04c1 TYP:4 MF- BR+ EX+
[   84.279369573,5] PHB#0001:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:00..00 
[   86.173842996,7] PHB#0001[0:1]: Purging all IODA tables...
[  338.488829106,7] PHB#0001[0:1]: Purging all IODA tables...
[  340.284070969,7] PHB#0001[0:1]: Purging all IODA tables...
```

madscientist159, 2018-05-11:
@mikey For what it's worth, Romulus has all of the PCIe presence detect pins tied together.  Talos II retains this unusual feature.  How is OPAL supposed to detect the actual electrical width of the card?

mikey, 2018-05-11:
@ricmata can you take a look at the trace?  It's a gen1  x4  card

madscientist159, 2018-05-11:
@mikey It's a little more complex than that because the vendor only brought out one lane of the four on the PI7C9X130.  This is normally supported by the PCIe standards but may not have seen much testing on POWER as it isn't common.

mikey, 2018-05-11:
also try 
> nvram -p ibm,skiboot --update-config pcie-max-link-speed=1

Note: that will limit all PHBs to gen1... but at least it might train

madscientist159, 2018-05-11:
No effect.  Debugging live in other channel ATM.

ricmata, 2018-05-13:
The link speed and link width status fields is a little deceiving because those status bits are only valid for certain state machines in the LTSSM. However, i do see that during polling we see a x16 link width which could be valid if the LTSSM is in Polling.Compliance. The only way that could happen is if we detected a receiver detect on all 16 lanes, entered Polling, and timed out after 24ms and entered Polling.Compliance.
We have observed with the latest p9 pci pll settings, there could be a failure for the P9 RX to detect an exit of Electrical Idle if there are unused lanes showing a receiver detect when there shouldn't be any. So if this is truly a x1 interface, what are the other 15 unused lanes doing? Can someone confirm how many lanes are actually connected. Would help determine if we are seeing this issue we have observed on a different setup I am debugging.

madscientist159, 2018-05-13:
@ricmata I can confirm only 1 lane is electrically connected.  There is no edge connector and no electrical contacts in the rest of the slot; the remainder of the slot is physically empty.

ricmata, 2018-05-14:
Can you check if you have the following configs are enabled in your kernel.

You should see this if everything is good.
$ grep SCOM /boot/config-uname -r
CONFIG_PPC_SCOM=y
CONFIG_SCOM_DEBUGFS=y

madscientist159, 2018-05-14:
@ricmata Yeah, we have access to a debug system with those options enabled.

What would you like to see?

oohal, 2019-04-11:
From memory this was fixed by falling back to training at Gen1 speeds. Can we close this @madscientist159?

madscientist159, 2019-04-11:
@oohal I was the author of the Gen 1 patch :wink:   My understanding is that it only fixed some cards, and that the PCIe bridge that this report is about does not work yet?