open-power/skiboot #179: Console soft/hard lockups during heavy kernel printk

pridhiviraj, 2018-05-29:
```
[  709.189162] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189163] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189165] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189166] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189168] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189169] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189170] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189172] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189173] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189175] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189176] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189177] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189179] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189180] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189182] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189183] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189184] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189186] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189187] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189189] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189190] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189191] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189193] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189194] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189196] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189197] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189198] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189200] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189201] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189203] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189204] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189205] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189207] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189208] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189210] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189211] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189212] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189214] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189215] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189217] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189218] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189219] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189221] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189222] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189224] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189225] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189226] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189228] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189229] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.189230] i40e 0001:05:00.1: ARQ: Unknown event 0x0000 ignored
[  709.362896] Watchdog CPU:26 Hard LOCKUP
[  709.362896] Modules linked in: ip6t_rpfilter ipt_REJECT nf_reject_ipv4 ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_security ip6table_raw iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack iptable_mangle iptable_security iptable_raw ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter ses enclosure at24 i2c_opal sg shpchp opal_prd ipmi_powernv ibmpowernv ipmi_devintf ipmi_msghandler ofpart powernv_flash uio_pdrv_genirq uio mtd ip_tables xfs libcrc32c sd_mod ast i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ttm drm mpt3sas i40e nvme nvme_core aacraid i2c_core ptp raid_class pps_core scsi_transport_sas
[  709.362912]  dm_mirror dm_region_hash dm_log dm_mod
[  709.362914] CPU: 26 PID: 6 Comm: kworker/u256:0 Kdump: loaded Not tainted 4.14.0-71.el7a.ppc64le #1
[  709.362914] Workqueue: poll_mpt3sas0_statu _base_fault_reset_work [mpt3sas]
[  709.362915] task: c000003fe6544500 task.stack: c000003fe658c000
[  709.362916] NIP:  c0000000000b65c0 LR: c0000000000b1ff0 CTR: 000000003003dfbc
[  709.362916] REGS: c000000007e6bd80 TRAP: 0900   Not tainted  (4.14.0-71.el7a.ppc64le)
[  709.362916] MSR:  9000000000009033 <SF,HV,EE,ME,IR,DR,RI,LE>  CR: 42028022  XER: 20040000
[  709.362919] CFAR: c0000000000b2000 SOFTE: 0 
[  709.362919] GPR00: 00000000300051ec c000003fe658f8f0 c000000001488d00 0000000000000000 
[  709.362921] GPR04: 0000000042028022 c0000000000b65c0 9000000000009033 0000000031d90058 
[  709.362922] GPR08: 0000000000000000 0000000031d90058 c0000000000b2004 9000000000001003 
[  709.362923] GPR12: 0000000031d90000 c000000007a31e00 c0000000016c88cc 0000000000000044 
[  709.362925] GPR16: c00000000165de40 c0000000016c88dc c000000000e20c80 000000000000000a 
[  709.362926] GPR20: 0044b82fa09b5a53 000000000000000d 0000000000000000 c00000000165e3a0 
[  709.362927] GPR24: 0000000000000000 c000000001648d78 0000000000000000 0000000000000010 
[  709.362928] GPR28: 0000000000000000 0000000000000034 0000000000000010 c000003fe658f980 
[  709.362930] NIP [c0000000000b65c0] opal_put_chars+0x190/0x2d0
[... 81 lines cut ...]
[  720.180757] GPR12: 0000000031d90000 c000000007a31e00 
[  720.180764] NIP [c00000000000a7c4] replay_interrupt_return+0x0/0x4
[  720.180765] LR [c000000000015b20] arch_local_irq_restore+0xf0/0x170
[  720.180766] Call Trace:
[  720.180767] [c000003fe658f9d0] [0000000000000044] 0x44 (unreliable)
[  720.180769] [c000003fe658f9f0] [c0000000001c6ef0] console_unlock+0x520/0xa70
[  720.180771] [c000003fe658fb00] [c0000000001c7804] vprintk_emit+0x3c4/0x4c0
[  720.180772] [c000003fe658fb80] [c0000000001caf48] vprintk_func+0x78/0x2d0
[  720.180773] [c000003fe658fbd0] [c0000000001ca744] printk+0x40/0x54
[  720.180776] [c000003fe658fbf0] [c0080000116cb048] _base_fault_reset_work+0x168/0x320 [mpt3sas]
[  720.180778] [c000003fe658fc80] [c00000000015ce0c] process_one_work+0x1bc/0x5f0
[  720.180779] [c000003fe658fd20] [c00000000015fdcc] worker_thread+0xac/0x6b0
[  720.180781] [c000003fe658fdc0] [c000000000169458] kthread+0x168/0x1b0
[  720.180783] [c000003fe658fe30] [c00000000000b628] ret_from_kernel_thread+0x5c/0xb4
[  720.180783] Instruction dump:
[  720.180784] 7d8000a6 e9628008 7d200026 618c8000 2c030900 4182e7f8 2c030500 4182f2e0 
[  720.180786] 2c030a00 4182ffa4 2c030e60 4182f090 <4e800020> 7c781b78 480003b9 480003d1 
[  721.189171] i40e 0001:05:00.3: ARQ: Unknown event 0x0000 ignored
[  721.189173] i40e 0001:05:00.3: ARQ: Unknown event 0x0000 ignored
[  721.189174] i40e 0001:05:00.3: ARQ: Unknown event 0x0000 ignored
[  721.189176] i40e 0001:05:00.3: ARQ: Unknown event 0x0000 ignored
[  721.189177] i40e 0001:05:00.3: ARQ: Unknown event 0x0000 ignored
[  721.189179] i40e 0001:05:00.3: ARQ: Unknown event 0x0000 ignored
[  721.189180] i40e 0001:05:00.3: ARQ: Unknown event 0x0000 ignored
```

stewart-ibm, 2018-09-18:
I think we've ended up solving these with some of Nick's patches (Mahesh remembers better than me... I'm just handwaving a bit).... so I'll close this bug for the time being, but if we can reproduce somewhere reliably, let's open a new bug and sort it out.

closed 2018-09-18 by stewart-ibm

bocaigo, 2019-06-19:
I have encountered the same problem. Could you tell me how your patch is implemented.Special thanks!!!