open-power/skiboot #186: GCC 8: no-direct-move is depreciated

shenki, 2018-06-13:
For every c file, I get this warning:
```
powerpc64-linux-gnu-gcc-8: warning: ‘-mno-direct-move’ is deprecated
```
This is with gcc 8.1 in Ubuntu:
```
powerpc64-linux-gnu-gcc-8 (Ubuntu 8.1.0-5ubuntu1) 8.1.0
```

We already try to detect it, but I guess this doesn't work for depreciated flags?
```
        $(call try-cflag,$(CC),-mno-direct-move)
```

shenki, 2018-07-23:
Ubuntu has just updated to make GCC 8 the default, so this is now very annoying:

```
joel@ozrom3:~/src/skiboot$ make
	[CC]  asm/asm-offsets.s
gcc: warning: ‘-mno-direct-move’ is deprecated
	[GN]  include/asm-offsets.h
	[AS]  asm/head.o
	[AS]  asm/misc.o
	[AS]  asm/kernel-wrapper.o
	[AS]  asm/cvc_entry.o
	[AR]  asm/built-in.a
	[CC]  core/relocate.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/console.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/stack.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/init.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/chip.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/mem_region.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/malloc.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/lock.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/cpu.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/utils.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/fdt.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/opal.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/interrupts.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/timebase.o
gcc: warning: ‘-mno-direct-move’ is deprecated
	[CC]  core/opal-msg.o
gcc: warning: ‘-mno-direct-move’ is deprecated
```

stewart-ibm, 2018-07-26:
Joel Stanley <notifications@github.com> writes:
> Ubuntu has just updated to make GCC 8 the default, so this is now very
> annoying:

It's been annoying on Fedora for longer :P

Your patch https://patchwork.ozlabs.org/patch/948167/ should fix it
though.

-- 
Stewart Smith
OPAL Architect, IBM.

closed 2018-07-26 by stewart-ibm