open-power/skiboot #188: pdbg putscom cannot triggers sreset

liyi-ibm, 2018-07-11:
In OpenBMC, we can trigger Host OS kdump using pdbg by sending `sreset` to one core:

```
# pdbg --version
pdbg 1.0 (commit cb2595c)
# pdbg -p0 -c1 -t0 sreset
```
(Note: we have to use an earlier version of pdbg to do sreset, please refer to: https://github.com/open-power/pdbg/issues/37)

We hope to trigger kdump with sreset via SBE fifo, but bellow command does not work (will trigger checkstop)
```
# special wakeup p0 core 1
./sbefifo.py /dev/sbefifo1 "000000060000a20200000000210F010B8000000000000000"
# Stop p0
./sbefifo.py /dev/sbefifo1 "000000030000a7010001fff1"
# sreset p0
./sbefifo.py /dev/sbefifo1 "000000030000a7010001fff3"
# stop p1
./sbefifo.py /dev/sbefifo2 "000000030000a7010001fff1"
# sreset p1
./sbefifo.py /dev/sbefifo2 "000000030000a7010001fff3"
```
(sbififo.py is just a simple wrapper to write data to /dev/sbefifo device).

By enabling debug message in pdbg, the output of `pdbg -p0 -c1 -t0 reset` looks like:
```
root@romulus:/tmp# ./pdbg -p0 -c1 -t0 sreset
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x210f0040, data 0x9c1ea26000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x210f0040, data 0x9c1ea26000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:137 pib_write: addr 0x210f010a, data 0x8000000000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x210f0113, data 0x4555500000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a02, data 0x7070707000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a9b, data 0x0009c00000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010ab3, data 0x00000000555500f0
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:137 pib_write: addr 0x21010a9c, data 0x0100000000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a02, data 0xf070707000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a9b, data 0x0009c00000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010ab3, data 0x00000000555500f0
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a02, data 0xf070707000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a9b, data 0x0009c00000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010ab3, data 0x00000000555500f0
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:137 pib_write: addr 0x21010a9c, data 0x0800000000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a02, data 0x1070707000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010a9b, data 0x8809c00000000000
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:125 pib_read: addr 0x21010ab3, data 0x0000000055550070
MINEDBG: ../../../../../../workspace/sources/pdbg/libpdbg/target.c:137 pib_write: addr 0x210f010a, data 0x0000000000000000
```
However, issue a sequence of pdbg commands as above debug output still cannot trigger Host OS kdump: 
```
./pdbg -p0 getscom 0x210f0040
./pdbg -p0 putscom 0x210f010a 0x8000000000000000 
./pdbg -p0 getscom 0x210f0040
./pdbg -p0 getscom 0x210f0113
./pdbg -p0 getscom 0x21010a02
./pdbg -p0 getscom 0x21010a9b
./pdbg -p0 getscom 0x21010ab3
./pdbg -p0 putscom 0x21010a9c 0x0100000000000000 
./pdbg -p0 getscom 0x21010a02
./pdbg -p0 getscom 0x21010a9b
./pdbg -p0 getscom 0x21010ab3
./pdbg -p0 getscom 0x21010a02
./pdbg -p0 getscom 0x21010a9b
./pdbg -p0 getscom 0x21010ab3
./pdbg -p0 getscom 0x21010a02
./pdbg -p0 putscom 0x21010a9c 0x0800000000000000 
./pdbg -p0 getscom 0x21010a02
./pdbg -p0 getscom 0x21010a9b
./pdbg -p0 getscom 0x21010ab3
./pdbg -p0 putscom 0x210f010a 0x00000000000000
```

We need to figure out why 'pdbg -p0 -c1 -t0 sreset` can trigger Host OS kdump, but sending putscom command via SBE fifo cannot trigger OS kdump. 
 
This issue may be related with OPAL, so open a bug here.

sgupta2m, 2018-07-11:
opal team is doing experiments to see if issue is in OPAL or HW. @maheshsal is working on it.

liuyangl, 2018-07-13:
@maheshsal and @sgupta2m How is the progress on this? There is a deal with Tencent coming this month and now we are gating on this problem. Would you pls help this? Thank you very much.

maheshsal, 2018-10-03:
@liuyangl Do you still see issues with pdbg.

hegdevasant, 2020-11-18:
AFAIK recent P9 BMC based system supports out of band SRESET mechanism. I guess BMC calls SBE chip-op to initiate the SRESET and SBE takes care resetting cores.  I'm not sure whether they use `pdbg` interface or not.

So I guess we can close this issue.

-Vasant

closed 2020-11-18 by hegdevasant