open-power/skiboot #190: fast-reboot got hung in P8 fsp platform

pridhiviraj, 2018-07-25:
```
[  612.100060546,5] OPAL: Reboot request...
[  612.100337385,7] NVRAM: Searched for 'fast-reset' found '1'
[  612.100338969,5] RESET: Initiating fast reboot 1...
[  612.100340074,7] RESET: Resetting from cpu: 0x20 (core 0x4)
[  612.100341939,7] RESET Waking up core 0x4
[  612.100345111,7] RESET Waking up core 0x5
[  612.100347366,7] RESET Waking up core 0x6
[  612.100350753,7] RESET Waking up core 0xc
[  612.100353700,7] RESET Waking up core 0xd
[  612.100356674,7] RESET Waking up core 0xe
[  612.100360115,7] RESET Waking up core 0x4
[  612.100363264,7] RESET Waking up core 0x5
[  612.100366358,7] RESET Waking up core 0x6
[  612.100369849,7] RESET Waking up core 0xc
[  612.100373005,7] RESET Waking up core 0xd
[  612.100376043,7] RESET Waking up core 0xe
[  612.100399042,7] RESET Waking up core 0x4
[  612.100402201,7] RESET Waking up core 0x5
[  612.100405262,7] RESET Waking up core 0x6
[  612.100408759,7] RESET Waking up core 0xc
[  612.100411886,7] RESET Waking up core 0xd
[  612.100414908,7] RESET Waking up core 0xe
[  612.100418418,7] RESET Waking up core 0x4
[  612.100421634,7] RESET Waking up core 0x5
[  612.100423937,7] RESET Waking up core 0x6
[  612.100427507,7] RESET Waking up core 0xc
[  612.100430714,7] RESET Waking up core 0xd
[  612.100433805,7] RESET Waking up core 0xe
[  612.100436861,7] RESET: Stopping the world...
[  612.100678111,7] RESET: Resetting all threads but self...
[  612.100680815,7] RESET: CPU 0x0021 reset in
[  612.100681369,7] RESET: CPU 0x0022 reset in
[  612.100682570,7] RESET: CPU 0x0023 reset in
[  612.100683691,7] RESET: CPU 0x0024 reset in
[  612.100684917,7] RESET: CPU 0x0025 reset in
[  612.100686018,7] RESET: CPU 0x0026 reset in
[  612.100687155,7] RESET: CPU 0x0027 reset in
[  612.100689060,7] RESET: CPU 0x0028 reset in
[  612.100689822,7] RESET: CPU 0x0029 reset in
[  612.100690821,7] RESET: CPU 0x002a reset in
[  612.100691586,7] RESET: CPU 0x002b reset in
[  612.100692614,7] RESET: CPU 0x002c reset in
[  612.100693383,7] RESET: CPU 0x002d reset in
[  612.100694267,7] RESET: CPU 0x002e reset in
[  612.100695111,7] RESET: CPU 0x002f reset in
[  611.479033939,7] RESET: CPU 0x0030 reset in
[  611.479034710,7] RESET: CPU 0x0031 reset in
[  611.479035662,7] RESET: CPU 0x0032 reset in
[  611.479036590,7] RESET: CPU 0x0033 reset in
[  611.479037524,7] RESET: CPU 0x0034 reset in
[  611.479038447,7] RESET: CPU 0x0035 reset in
[  611.479039432,7] RESET: CPU 0x0036 reset in
[  611.479040457,7] RESET: CPU 0x0037 reset in
[  611.479063141,7] RESET: CPU 0x0060 reset in
[  611.479063980,7] RESET: CPU 0x0061 reset in
[  611.479065153,7] RESET: CPU 0x0062 reset in
[  611.479066020,7] RESET: CPU 0x0063 reset in
[  611.479067180,7] RESET: CPU 0x0064 reset in
[  611.479068069,7] RESET: CPU 0x0065 reset in
[  611.479069143,7] RESET: CPU 0x0066 reset in
[  611.479070199,7] RESET: CPU 0x0067 reset in
[  611.481087630,7] RESET: CPU 0x0068 reset in
[  611.481088441,7] RESET: CPU 0x0069 reset in
[  611.481089368,7] RESET: CPU 0x006a reset in
[  611.481090261,7] RESET: CPU 0x006b reset in
[  611.481091267,7] RESET: CPU 0x006c reset in
[  611.481092109,7] RESET: CPU 0x006d reset in
[  611.481092971,7] RESET: CPU 0x006e reset in
[  611.481093837,7] RESET: CPU 0x006f reset in
[... 520 lines cut ...]
[  615.470017109,7] FSP: LID 80f08102 Chunk read -> rc=0x00 off: 00800000 twritten: 0022f108
[  616.367621956,5] Clearing memory... 16/191GB done
[  616.377863690,5] Clearing memory... 32/191GB done
[  616.377864739,5] Clearing memory... 48/191GB done
[  616.377865762,5] Clearing memory... 63/191GB done
[  616.382986965,5] Clearing memory... 79/191GB done
[  616.388108156,5] Clearing memory... 95/191GB done
[  616.388109209,5] Clearing memory... 111/191GB done
[  616.388110287,5] Clearing memory... 127/191GB done
[  616.388111269,5] Clearing memory... 143/191GB done
[  616.388112130,5] Clearing memory... 159/191GB done
[  616.388113125,5] Clearing memory... 175/191GB done
[  616.388114021,5] Clearing memory... 191/191GB done
[  616.388114908,5] Clearing memory... 191/191GB done
[  616.388115910,5] Clearing memory... 191/191GB done
[  616.388116782,5] Clearing memory... 191/191GB done
[  616.388117770,5] Clearing memory... 191/191GB done
[  616.388118665,5] Clearing memory... 191/191GB done
[  616.388119644,5] Clearing memory... 191/191GB done
[  616.388126425,5] INIT: Waiting for kernel...
[  616.388128189,7] INIT: Kernel loaded, size: 3868920 bytes (0 = unknown preload)
[  616.388129480,5] INIT: 64-bit LE kernel discovered
[  616.388130378,7] INIT: 64-bit kernel entry at 0x20010000, size 0x3b08f8
[  616.388132167,5] INIT: Initramfs loaded, size: 10678536 bytes
```

pridhiviraj, 2018-07-25:
@npiggin @stewart-ibm Please have a look at it, looks like it is broken on P8 fsp platform.

oohal, 2019-07-15:
We disabled fast-reboot by default on P8 systems since the SRESET process can't be made 100% reliable there.

closed 2019-07-15 by oohal