open-power/skiboot #194: putscom kernel oops then checkstop

abali999, 2018-09-03:
putscom -c 0 0x1001180b 0x0000000000000000  works
putscom -c 0 0x1001180b 0xf000000000000000  crashes
Any suggestions what I am doing wrong?   The scom register is supposed to turn off some L3 ways, and can be changed dynamically according to docs.
This is on a P9 LC922 box with powerNV kernel and using putscom from the opal-utils package.

 [  218.533737] Unable to handle kernel paging request for data at address 0x00000000
[  218.533821] Faulting instruction address: 0xc000000000353484
[  218.533879] Oops: Kernel access of bad area, sig: 11 [#1]
[  218.533917] LE SMP NR_CPUS=2048 NUMA PowerNV
[  218.533961] Modules linked in: idt_89hpesx joydev input_leds mac_hid vmx_crypto crct10dif_vpmsum at24 ofpart ipmi_powernv cmdlinepart ipmi_devintf powernv_flash ipmi_msghandler opal_prd ibmpowernv uio_pdrv_genirq mtd uio sch_fq_codel ip_tables x_tables autofs4 ses enclosure scsi_transport_sas hid_generic usbhid hid ast ttm drm_kms_helper igb syscopyarea sysfillrect sysimgblt fb_sys_fops crc32c_vpmsum drm i40e i2c_algo_bit aacraid
[  218.534303] CPU: 24 PID: 3409 Comm: sshd Not tainted 4.15.0-33-generic #36-Ubuntu
[  218.534369] NIP:  c000000000353484 LR: c00000000033720c CTR: c00000000032d280
[  218.534436] REGS: c000003fdce376f0 TRAP: 0300   Not tainted  (4.15.0-33-generic)
[  218.534491] MSR:  9000000000009033 <SF,HV,EE,ME,IR,DR,RI,LE>  CR: 24002828  XER: 20040000
[  218.534552] CFAR: c000000000337208 DAR: 0000000000000000 DSISR: 40000000 SOFTE: 1
[  218.534552] GPR00: c00000000033720c c000003fdce37970 c0000000016eb000 c000003fbafa9990
[  218.534552] GPR04: c00000000441e168 00007fc03c5d0000 00007655f99e0000 00007fc03c5d0000
[  218.534552] GPR08: ffffffff00000001 0000000000000000 0000000000000000 c000003fd57dfa98
[  218.534552] GPR12: c00000000032d280 c00000000fa90800 0000000000000d15 0000000000000148
[  218.534552] GPR16: 00000e199e4b86e0 000000000000000b 0000000000008005 0000000000040100
[  218.534552] GPR20: 0000000000000000 00000000418004fc 00000000003c0000 0000000008430000
[  218.534552] GPR24: c00000000188e9f0 0000000000000000 c000003fbafa9990 c000003fdce37a50
[  218.534552] GPR28: c000003fbafa9a08 0000000000000001 c000003fbafabfc0 0000000000000000
[  218.535042] NIP [c000000000353484] unlink_anon_vmas+0x44/0x310
[  218.535089] LR [c00000000033720c] free_pgtables+0x16c/0x240
[  218.535124] Call Trace:
[  218.535145] [c000003fdce37970] [c000000000344e08] __remove_shared_vm_struct+0x68/0x80 (unreliable)
[  218.535208] [c000003fdce379d0] [c00000000033720c] free_pgtables+0x16c/0x240
[  218.535255] [c000003fdce37a30] [c0000000003481a4] exit_mmap+0xc4/0x210
[  218.535301] [c000003fdce37af0] [c000000000109ec8] mmput+0xa8/0x1d0
[  218.535347] [c000003fdce37b20] [c000000000114540] do_exit+0x360/0xc80
[  218.535394] [c000003fdce37be0] [c000000000114f30] do_group_exit+0x60/0x100
[  218.535440] [c000003fdce37c20] [c0000000001264ec] get_signal+0x3cc/0x7a0
[  218.535487] [c000003fdce37d10] [c00000000001d98c] do_signal+0x7c/0x2c0
[  218.535534] [c000003fdce37e00] [c00000000001dd70] do_notify_resume+0xd0/0x100
[  218.535589] [c000003fdce37e30] [c00000000000b8c4] ret_from_except_lite+0x70/0x74
[  218.535642] Instruction dump:
[  218.535670] fb41ffd0 fb61ffd8 fb81ffe0 fba1ffe8 fbc1fff0 fbe1fff8 f8010010 f821ffa1
[  218.535727] 7c7c1b78 7c7a1b78 ebfc0079 7fbcf800 <ebdf0000> 419e02a8 fb010020 fb210028
[  218.535786] ---[ end trace 4456a1917aa7e1d1 ]---
[  218.679869] Watchdog CPU:0 Hard LOCKUP
[  218.679869] Modules linked in: idt_89hpesx joydev input_leds mac_hid vmx_crypto crct10dif_vpmsum at24 ofpart ipmi_powernv cmdlinepart ipmi_devintf powernv_flash ipmi_msghandler opal_prd ibmpowernv uio_pdrv_genirq mtd uio sch_fq_codel ip_tables x_tables autofs4 ses enclosure scsi_transport_sas hid_generic usbhid hid ast ttm drm_kms_helper igb syscopyarea sysfillrect sysimgblt fb_sys_fops crc32c_vpmsum drm i40e i2c_algo_bit aacraid
[  218.679884] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D          4.15.0-33-generic #36-Ubuntu
[  218.679885] NIP:  c00000000017414c LR: c000000000174570 CTR: 0000000000000008
[  218.679886] REGS: c00000003ffffd80 TRAP: 0900   Tainted: G      D           (4.15.0-33-generic)
[  218.679886] MSR:  9000000000009033 <SF,HV,EE,ME,IR,DR,RI,LE>  CR: 24000424  XER: 20040000
[  218.679890] CFAR: c00000000017456c SOFTE: 0
[  218.679890] GPR00: c000000000174570 c0000000016e7e30 c0000000016eb000 c00000000161caa8
[  218.679890] GPR04: c000003feeb96fd8 0000000000000000 0000000000000000 0000000000000000
[  218.679890] GPR08: c0000000016e4080 c0000000016e4000 0000000000000008 c000003feeb9727c
[  218.679890] GPR12: 0000000000000000 c00000000fa80000 0000000000000000 0000000000000000
[  218.679890] GPR16: 0000000000000000 0000000000000000 0000000000000000 0000000000000000
[  218.679890] GPR20: 0000000000000000 0000000000000001 0000000010004d9c 00000000100053ed
[  218.679890] GPR24: 0000000010005411 c000000000000000 c00000000161caa8 0000000000000008
[  218.679890] GPR28: c000003feeb96fd8 c000000001722364 0000000000000001 c000000001721ee0
[  218.679903] NIP [c00000000017414c] call_cpuidle+0x3c/0x90
[  218.679905] LR [c000000000174570] do_idle+0x2b0/0x330
[  218.679905] Call Trace:
[  218.679907] [c0000000016e7e30] [c000000000ac4898] cpuidle_select+0x38/0x50 (unreliable)
[  218.679909] [c0000000016e7e50] [c000000000174570] do_idle+0x2b0/0x330
[  218.679911] [c0000000016e7ea0] [c00000000017482c] cpu_startup_entry+0x3c/0x50
[  218.679912] [c0000000016e7ed0] [c00000000000d4a0] rest_init+0xe0/0x100
[  218.679914] [c0000000016e7f00] [c0000000010b4160] start_kernel+0x5a0/0x5c0
[  218.679916] [c0000000016e7f90] [c00000000000ac7c] start_here_common+0x1c/0x520
[  218.679916] Instruction dump:
[  218.679918] 60000000 7c0802a6 f8010010 f821ffe1 39400008 78290464 39090080 7cc040a8
[  218.679920] 7cc65078 7cc041ad 40c2fff4 7c0004ac <e9290080> 71290004 4082001c 48950761
[  218.681381] Unable to handle kernel paging request for data at address 0xaa530476c81c08f5
[  218.681451] Faulting instruction address: 0xc000000000ac0ce8
[  218.753807] Unable to handle kernel paging request for data at address 0xaa530476c81cd4f5
[... 1602 lines cut ...]
 14.94502|  Callout type             : Procedure Callout
 14.94503|  Procedure                : EPUB_PRC_LVL_SUPP
 14.94503|  Priority                 : SRCI_PRIORITY_MED
 14.94504|------------------------------------------------
 14.94505|  Callout type             : Hardware Callout
 14.94506|  CPU id                   : 0
 14.94507|  Target                   : Physical:/Sys0/Node0/Proc0
 14.94508|  Deconfig State           : NO_DECONFIG
 14.94508|  GARD Error Type          : GARD_NULL
 14.94509|  Priority                 : SRCI_PRIORITY_LOW
 14.94510|------------------------------------------------
 14.94510|
 14.94511|------------------------------------------------
 14.94512|  System checkstop occurred during runtime on previous boot
 14.94513|------------------------------------------------
 14.94513|
 14.94514|------------------------------------------------
 14.94514|  Hostboot Build ID:
 14.94515|================================================
 14.94647|HWAS|FUNCTIONAL> DIMM[03]=A0A0000000000000
 14.94648|HWAS|FUNCTIONAL> Proc[05]=8800000000000000
 14.94649|HWAS|FUNCTIONAL> Core[07]=FFF3CFC3FFFF0000
 15.17252|ISTEP  6.10 - host_revert_sbe_mcs_setup
 15.18312|ISTEP  6.11 - host_start_occ_xstop_handler
 16.27499|ISTEP  6.12 - host_voltage_config

mikey, 2018-09-04:
scoms can do all sorts of nasty things to your computer. Hitting a random scom and it crashing your machine is not a skiboot bug.  Please ask the hardware team what the problem is here.

closed 2018-09-04 by mikey

abalib, 2018-09-04:
It was not a random scom but a documented som register.  I will check with the hardware team.