open-power/skiboot #195: Bug: Soft vs hard minimum for power capping

tpatki, 2018-09-07:
We've been working with Shilpa and Vaidy on an issue that we'd like to see addressed in a future release of OPAL for IBM witherspoon. I'm documenting our issue, Shilpa's fix and some additional notes here. I was hoping to see this change in hw/occ.c in 6.0.8 but looks like it didn't make it in there. 

Minimum power cap in OPAL was based on the "hard" minimum power cap and was assuming that GPUs are in peak power use of 250W, resulting in a minimum node power of 1700 W. Ideally, this minimum should be the "soft" minimum power cap, which is 500 W. This is captured in the figur below. 

There are several scenarios where a workload running on a node may use no or fewer than 4 GPUs, and in these cases, being able to set the power cap to be low enough gives us the ability to redirect extra power to CPUs, or just save power. Thanks!

[image]

Shilpa's fix was as follows:

```
diff --git a/hw/occ.c b/hw/occ.c
index a55bf8e..7e6e02e 100644
--- a/hw/occ.c
+++ b/hw/occ.c
@@ -246,7 +246,8 @@ struct occ_dynamic_data {
        u16 min_pwr_cap;
        u16 max_pwr_cap;
        u16 cur_pwr_cap;
-       u8 pad[112];
+       u16 soft_min_pwr_cap;
+       u8 pad[110];
        struct opal_command_buffer cmd;
        struct occ_response_buffer rsp;
 } __packed;
@@ -1362,7 +1363,7 @@ int occ_get_powercap(u32 handle, u32 *pcap)
 
        switch (powercap_get_attr(handle)) {
        case POWERCAP_OCC_MIN:
-               *pcap = ddata->min_pwr_cap;
+               *pcap = ddata->soft_min_pwr_cap;
                break;
        case POWERCAP_OCC_MAX:
                *pcap = ddata->max_pwr_cap;
@@ -1410,7 +1411,7 @@ int occ_set_powercap(u32 handle, int token, u32 pcap)
                return OPAL_SUCCESS;
 
        if (pcap && (pcap > ddata->max_pwr_cap ||
-           pcap < ddata->min_pwr_cap))
+           pcap < ddata->soft_min_pwr_cap))
                return OPAL_PARAMETER;
 
        pcap_cdata = pcap;
```

stewart-ibm, 2018-09-14:
Thanks for the bug report. This is an area where I've been a bit suspicious there's bugs hiding around the place, and indeed, our `op-test` tests were showing some failures in the powercap tests (at least with op-build master on some platforms).

I'm trying to chase down *why* that is, but https://github.com/open-power/op-test-framework/blob/adeaa20/testcases/OpalSysfsTests.py#L107 is my new and improved test case for it, and this should be enough to at least start to ask questions of the OCC team. I'll try and incorporate pulling some debug data from the OCC on test failure too, as that may help get to the bottom of it.

As for the issue of hard vs soft minimum, hrm... it looks like the device tree bindings and kernel driver completely don't take into account the concept of a soft minimum power cap.

The patch you mention there, which is https://patchwork.ozlabs.org/patch/958105/ is one I don't think I want to take as-is, as then there's no way to work out what the hard minimum is.

I think the correct solution here is going to be twofold:
1. add a soft min powercap to the device tree bindings.
2. add support for this to the kernel.

stewart-ibm, 2018-09-14:
FWIW this is what I'm getting as a current failure on a DD2.1 witherspoon running op-build master and the testcase mentioned above:

```
======================================================================           
FAIL [114.099s]: test_opal_powercap (testcases.OpalSysfsTests.Skiroot)       
----------------------------------------------------------------------                           
Traceback (most recent call last):                                          
  File "/home/stewart/op-test-framework/testcases/OpalSysfsTests.py", line 120, in test_opal_powe
rcap                                                                                             
    self.set_power_cap(min_powercap)
  File "/home/stewart/op-test-framework/testcases/OpalSysfsTests.py", line 98, in set_power_cap
    repr(valid_powercap_values)))
AssertionError: Retrieved powercap was not either the previous one (3050) or the one we're trying
 to set (1700). Got 1550, expected in [3050, 1700]

----------------------------------------------------------------------
Ran 1 test in 114.099s
```

and then, if you run it again without an IPL:
```
======================================================================                           FAIL [58.049s]: test_opal_powercap (testcases.OpalSysfsTests.Skiroot) 
----------------------------------------------------------------------                           
Traceback (most recent call last):                                                              
  File "/home/stewart/op-test-framework/testcases/OpalSysfsTests.py", line 119, in test_opal_powe
rcap                                                                                             
    self.set_power_cap(max_powercap)                                                             
  File "/home/stewart/op-test-framework/testcases/OpalSysfsTests.py", line 105, in set_power_cap
    cur_powercap, value))                                                           
AssertionError: OPAL failed to set power cap value. Got 1550 when trying to set 3050.            
                                                                                                 ----------------------------------------------------------------------                           
Ran 1 test in 58.049s   
```

tpatki, 2018-09-14:
@stewart-ibm: Thanks for looking into this! 

Shilpa's patch is a temporary fix. I agree that adding soft power min to devicetree and adding support for it in the kernel is the right way to go. Both the hard and soft power caps should be exposed through OPAL, along with explanation/documentation about what guarantees to expect in each case. As in, when you set a soft power cap, you may go over the limit temporarily, and more experimentation is needed to understand the time window and maximum range for exceeding the power cap would be.  

Another important knob that's not currently exposed is that of Turbo frequencies/ultrascale, I'll circle around to that once I have a better understanding for it.

stewart-ibm, 2018-09-17:
https://github.com/open-power/op-test-framework/blob/master/testcases/OpTestEM.py has a bunch of tests for turbo / ultraturbo and how that is mapped out in sysfs.

https://open-power.github.io/skiboot/doc/device-tree/ibm,opal/power-mgt.html has some bare bones documentation on the device tree entries too. (and wow, the documentation there really needs expanding upon. Patches would be much appreciated :)

shilpasri, 2019-02-05:
Hi Tapasya,

Sorry for taking a long time to resolve this. I have posted a new version of the skiboot patch to fix the issue which requires minimal changes to the kernel to export the full powercap range. 
https://lists.ozlabs.org/pipermail/skiboot/2019-February/013165.html

This skiboot patch changes the "powercap-min" property to point to soft powercap minimum limit instead of hard powercap minimum. It also adds DT property "powercap-hard-min" to export the hard powercap minimum.

The kernel changes with this DT updates are minimal to adding a new sysfs file to export the hard min powercap limit.

tpatki, 2019-02-07:
Thank you, Shilpa!

shilpasri, 2019-02-27:
The fix for this issue is available in skiboot
https://github.com/open-power/skiboot/commit/a96739c6c1cdf02521cab703733cef8cb54debbf
This patch is queued for stable and should subsequently be available in the next service-pack
The kernel patch to export hard-min powercap is also posted 
https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1943206.html

tpatki, 2019-02-27:
Excellent! Thank you, Shilpa! Looking forward to the next service pack being available.

gautshen, 2019-03-18:
The fix is now in the 6.0.18 stable branch https://github.com/open-power/skiboot/commit/a3c88493956190b5079139f4b1aef0a3cbb2dd42

shilpasri, 2019-06-24:
The skiboot image with the fix  is available in PNOR level OP910_1913D
    IBM-witherspoon-OP9_v2.0.14_1.2
    op-build-v2.0.14-420-gb32daa2
    buildroot-2018.05.1-9-gc99f2ee   
    skiboot-v6.0.18
    hostboot-14fd85c-p80ac538
    occ-394de99
    linux-4.17.12-openpower1-p2f43189
    petitboot-v1.7.5-p6a40ae5
    machine-xml-94a137f
    hostboot-binaries-5c5adec
    capp-ucode-p9-dd2-v4
    sbe-4aa6703

closed 2019-06-24 by hegdevasant

stewart-ibm, 2019-06-24:
shilpasri <notifications@github.com> writes:
> The skiboot image with the fix  is available in PNOR level OP910_1913D
>     IBM-witherspoon-OP9_v2.0.14_1.2

IBM builds aren't too relevant for the public project though, the real
bit that is relevant is the skiboot version:

>     skiboot-v6.0.18

as this op-build version number isn't an upstream commit:

>     op-build-v2.0.14-420-gb32daa2

-- 
Stewart Smith
OPAL Architect, IBM.