open-power/skiboot #196: getting stuck when installing RHEl in power 9 server (8335-GTG)

surendrasss, 2018-10-01:
We facing problems while installing RHEL 7.5 LE in power 9 server.installing using USB getting stuck after starting installation 
Below are the details please help me resolve this issue
The system is going down NOW
sent SIGTERM to all processes
sent SIGKILL to all processes
Kexec_cores:Starting new kernel
OPAL:switch to big-endian OS
OPAL:swtich to little-endian OS
after this it is not proceeding

oohal, 2018-10-01:
Are you using the right RHEL release? There's a separate release stream for Power9 because it requires a newer kernel than the 3.10 based kernel that is generally used with RHEL.

surendrasss, 2018-10-03:
Thanks for the reply sir yes checked in the installation doc supported RHEL for power 9 is RHEL 7.4 LE or later

oohal, 2018-10-05:
[RHEL 7.4 supported architectures](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/7.4_release_notes/chap-red_hat_enterprise_linux-7.4_release_notes-architectures)
[RHEL 7.5 supported architectures](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/7.5_release_notes/chap-red_hat_enterprise_linux-7.5_release_notes-architectures)

The ppc64le release of RHEL 7.4 the only supports POWER8. Use RHEL 7.5 with the alt-kernel for Power9 systems.

surendrasss, 2018-10-10:
Is there any specific option to select alt-kernel because I'm only getting red hat 7.5 LE option that's it

stewart-ibm, 2018-10-10:
This isn't a general support forum.

closed 2018-10-10 by stewart-ibm