open-power/skiboot #199: Fast reboot not blocked on change of garded resources

stewart-ibm, 2018-11-20:
If we clear gard records, we still do a fast-reboot - which may not get the garded hardware back.

oohal, 2019-07-15:
I'm not really sure how we're supposed to fix this since we don't really parse the guard partition at all in skiboot. Suppose we could force a full re-IPL if the host does any flash writes? Seems a bit heavyweight.

stewart-ibm, 2019-07-15:
oohal <notifications@github.com> writes:
> I'm not really sure how we're supposed to fix this since we don't
> really parse the guard partition at all in skiboot. Suppose we could
> force a full re-IPL if the host does any flash writes? Seems a bit
> heavyweight.

Maybe any flash writes to the GUARD partition?

I don't have a good answer unfortunately, probably why it's sat as a
github issue unaddressed for so long :)

-- 
Stewart Smith
OPAL Architect, IBM.