open-power/skiboot #200: opal-gard clear clears in a way that Hostboot thinks is ECC errors

stewart-ibm, 2018-11-20:
`opal-gard clear all`

followed by `ipmitool power off ; ipmitool power on`

```
--== Welcome to Hostboot hostboot-40a34c9-p36fbe89/hbicore.bin ==--

  7.68954|secure|SecureROM valid - enabling functionality
 11.75460|Booting from SBE side 0 on master proc=00050000
 11.80687|ISTEP  6. 5 - host_init_fsi
 12.26229|ISTEP  6. 6 - host_set_ipl_parms
 12.47868|ECC error in PNOR flash in section offset 0x0002F600

 12.47868|Clearing section GUARD due to ECC error

 12.59438|Done

 12.59446|System shutting down with error status 0x60F

--== Welcome to SBE - CommitId[0xcf61dc39] ==--
```

oohal, 2018-11-20:
What system type was this on? opal-guard uses the same ECC programming code that pflash uses so this is strange...

Do you have a copy of the GUARD partition that it failed on?

stewart-ibm, 2018-11-20:
oohal <notifications@github.com> writes:
> What system type was this on? opal-guard uses the same ECC programming code that pflash uses so this is strange...
>
> Do you have a copy of the GUARD partition that it failed on?

I should be able to get a copy of one that it will fail on, it's my
all-CI-runs-here box and it's solidly broken at the moment :/

-- 
Stewart Smith
OPAL Architect, IBM.

oohal, 2019-07-15:
Should be fixed by 27d1ef2e

closed 2019-07-15 by oohal