open-power/skiboot #201: Userspace access to OPAL's powercap

tpatki, 2018-11-27:
At present, root privileges are required for someone to set a power cap (basically the following):

``echo [power-cap-value] > /sys/firmware/opal/powercap/system-powercap/powercap-current``

How can I set up OPAL permissions so non-privileged users or a certain group can have access to the powercap/psr files (not other OPAL files)? 

Is simply changing file permissions here from 664 to 666 right thing to do? I've tried this on a test node, and it works, but can I assume that it is safe at all time?

shilpasri, 2019-03-11:
Hi Tapasya,

You can do the following to enable write access to powercap/psr files only to users belonging  to a certain group. The following command enables users belonging to group 'powercap'
`# /bin/chown root:powercap /sys/firmware/opal/powercap/system-powercap/powercap-current /sys/firmware/opal/psr/cpu_to_gpu_0 /sys/firmware/opal/psr/cpu_to_gpu_8`

And changing the file permissions will also work and is safe. OPAL returns -EBUSY for concurrent OCC powercap/psr commands and OPAL also checks for correctness of the input value. 

But I would still recommend using a custom group allowing access only to the members of the group as it is more restrictive and can refrain unwanted writes from non-members.

hegdevasant, 2020-11-18:
I guess issue is resolved and safe to close this issue.

Feel free to REOPEN if you have any further queries on this.

-Vasant

closed 2020-11-18 by hegdevasant