open-power/skiboot #206: GCOV build fails with gcc 8.2

shenki, 2018-12-12:
```
powerpc64-linux-gnu-ld: section .bss VMA [0000000000300000,000000000035f3f7] overlaps section .data VMA [00000000002cf878,000000000030a7af]
powerpc64-linux-gnu-ld: section .dtors.65435 VMA [000000000030a7b0,000000000030af5f] overlaps section .bss VMA [0000000000300000,000000000035f3f7]
```

powerpc64-linux-gnu-gcc (Ubuntu 8.2.0-7ubuntu1) 8.2.0

shenki, 2018-12-12:
When this is fixed, consider moving ubuntu-latest out of the expected failures list in travis.

stewart-ibm, 2018-12-12:
Joel Stanley <notifications@github.com> writes:
> ```
> powerpc64-linux-gnu-ld: section .bss VMA [0000000000300000,000000000035f3f7] overlaps section .data VMA [00000000002cf878,000000000030a7af]
> powerpc64-linux-gnu-ld: section .dtors.65435 VMA [...]

I swear I had a patch for this somewhere.... it's a relatively simple
fix at least. I'll try and whip something up.

-- 
Stewart Smith
OPAL Architect, IBM.

oohal, 2019-07-15:
@stewart-ibm still a problem?

stewart-ibm, 2019-07-15:
oohal <notifications@github.com> writes:
> @stewart-ibm still a problem?

Nah, I managed to fix it at some point. Not sure when though.

-- 
Stewart Smith
OPAL Architect, IBM.

closed 2019-07-23 by stewart-ibm