open-power/skiboot #218: build failure v6.2 fedora with new gcc9

michelmno, 2019-02-13:
referenced bug https://bugzilla.redhat.com/show_bug.cgi?id=1675573 and reported warnings/errors
```
gcc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mcpu=power8 -mtune=power8 -fasynchronous-unwind-tables -fstack-clash-protection -I. -I../../include -I../../  -c libflash/libffs.c -o libflash-libffs.o
In file included from /usr/include/string.h:494,
                 from libflash/libffs.c:21:
In function 'strncpy',
    inlined from 'ffs_part_info' at libflash/libffs.c:525:3:
/usr/include/bits/string_fortified.h:106:10: warning: '__builtin_strncpy' output may be truncated copying 15 bytes from a string of length 15 [-Wstringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
gcc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mcpu=power8 -mtune=power8 -fasynchronous-unwind-tables -fstack-clash-protection -I. -I../../include -I../../  -c libflash/libffs.c -o libflash-libffs.o
In file included from /usr/include/string.h:494,
                 from libflash/libffs.c:21:
In function 'strncpy',
    inlined from 'ffs_part_info' at libflash/libffs.c:525:3:
/usr/include/bits/string_fortified.h:106:10: warning: '__builtin_strncpy' output may be truncated copying 15 bytes from a string of length 15 [-Wstringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
gcc -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mcpu=power8 -mtune=power8 -fasynchronous-unwind-tables -fstack-clash-protection -O2 -Wall -Werror -Wno-stringop-truncation -I. -c libflash/ecc.c -o libflash-ecc.o
libflash/ecc.c: In function 'memcpy_to_ecc_unaligned':
libflash/ecc.c:419:24: error: taking address of packed member of 'struct ecc64' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  419 |   memcpy(inc_uint64_by(&ecc_word.data, alignment), src, bytes_wanted);
      |                        ^~~~~~~~~~~~~~
libflash/ecc.c:448:24: error: taking address of packed member of 'struct ecc64' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  448 |   memcpy(inc_uint64_by(&ecc_word.data, len), inc_ecc64_by(dst, len),
      |                        ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

hegdevasant, 2019-02-15:
https://github.com/open-power/skiboot/pull/212 should fix first part of the warning.

-Vasant

hegdevasant, 2019-02-18:
And https://lists.ozlabs.org/pipermail/skiboot/2019-February/013297.html this should remaining issue.

FYI https://koji.fedoraproject.org/koji/taskinfo?taskID=32879051
 --> Sample koji build with above two fixes.

-Vasant

closed 2019-02-19 by stewart-ibm