open-power/skiboot #236: P8: creating multiple gard entries fails

kachini, 2019-04-12:
I'm trying to create multiple gard entries on P8 and it fails:

[root@localhost gard]# ./gard create /Sys0/Node0/DIMM15 
Unit /Sys0/Node0/DIMM15 is already GARDed by record 0x000001
[root@localhost gard]# ./gard create /Sys0/Node0/DIMM16
[root@localhost gard]# ./gard create /Sys0/Node0/DIMM17
ECC: uncorrectable error: d2ffffff00000002 23
The data at the GUARD partition does not appear to be valid gard data
Clear the entire GUARD partition? [y/N]

oohal, 2019-07-15:
This is probably fixed by 27d1ef2ebabb, can you try it with a newer version of opal-gard?

hegdevasant, 2020-11-18:
I'm going to close this one.. Feel free to REOPEN if you hit issue with upstream code.

-Vasant

closed 2020-11-18 by hegdevasant