open-power/skiboot #237: On the P8 platform, the devicetree contains duplicate unit addresses

maxpoliak, 2019-04-16:
dtc utility on the linux host:

<stdout>: Warning (unique_unit_address): /memory@0: duplicate unit-address
(also used in node /interrupt-controller@0)

Nodes:

interrupt-controller@0 {
        #address-cells = < 0x00 >;
        device_type = "PowerPC-Interrupt-Source-Controller";
        compatible = "IBM,ppc-xics\0IBM,opal-xics";
        #interrupt-cells = < 0x02 >;
        reg = < 0x00 0x00 0x00 0x00 >;
        phandle = < 0x10000454 >;
        interrupt-controller;
};

memory@0 {
        ibm,chip-id = < 0x00 >;
        device_type = "memory";
        reg = < 0x00 0x00 0x40 0x00 >;
        phandle = < 0x100000cd >;
        ibm,associativity = < 0x04 0x00 0x00 0x00 0x00 >;
};

oohal, 2019-04-17:
It's been like this since before the inital open source release of skiboot. I'm not sure what we're supposed to be using as the unit address though. we could change it to 0xff...ff without problems though.

maxpoliak, 2019-04-17:
I'm not sure about the memory@0, but I didn`t find a code in the Linux source, where the "reg" property or unit addr is used for "interrupt-controller" node.

If I understand correctly, you are proposing to make the following changes:
```
interrupt-controller@0xff..ff {
        #address-cells = < 0x00 >;
        device_type = "PowerPC-Interrupt-Source-Controller";
        compatible = "IBM,ppc-xics\0IBM,opal-xics";
        #interrupt-cells = < 0x02 >;
        reg = < 0xffffffff 0xffffffff 0x0 0x0 >;
        phandle = < 0x10000454 >;
        interrupt-controller;
};
```
?
Maybe I will change the reg and addr values for this node and add a new patch to the upstream, if you don't mind?

maxpoliak, 2019-04-22:
Please see this patch at https://patchwork.ozlabs.org/patch/1088767/
Thanks

maxpoliak, 2019-05-14:
https://patchwork.ozlabs.org/patch/1089094/

closed 2019-05-14 by maxpoliak