open-power/skiboot #240: 512b read of last flash sector == stuck?

bdragon28, 2019-04-29:
FreeBSD's new opal_flash driver seems to be hanging on the new Talos II beta firmware (post skiboot v6.3-rc2)

I instrumented the driver and the access patterns it is attempting are as follows:
opalflash_read offset 0 count 512
opalflash_read offset 512 count 512
opalflash_read offset 0 count 512
opalflash_read offset 67108352 count 512
(hangs at this point inside opal)

So it enters opal and never returns when attempting to do a 512b read of the end of the last block. (the 512b reads are because FreeBSD is treating it as a disk, so it's running the generic partition probing reads that it does for all disks.)

I have a suspicion that this might be a corner case in 7f291166283f0f02e75787023326279036feebe1 -- perhaps it's attempting a read past the end of the device?

Manually-constructed backtrace:
 .bt_add_ipmi_msg_head
 .time_wait_nopoll
 .ipmi_queue_msg_sync
 .hiomap_queue_msg_sync
 .hiomap_window_move
 .ipmi_hiomap_cmd_cb
 .blocklevel_raw_read
 .opal_flash_op
 opal_entry

regs:
```
p0 c0 t0
NIA   : 0x0000000030019460
CFAR  : 0x0000000030019480
MSR   : 0x9000000002803002
LR    : 0x00000000300193c8
CTR   : 0x0000000030059ec8
TAR   : 0x0000000000000000
CR    : 0x40000202
XER   : 0x00000000
GPRS  :
 0x000000003001dd30 0x0000000031c037e0 0x000000003010b800 0x0000002acc767541
 0x00000000d00100e4 0x0000000000000004 0x00000000303142e0 0x0000000031c00060
 0x0000000031c00060 0x0000000000000000 0x00000000302fb800 0x0000000000000000
 0x0000000040000204 0x0000000031c00000 0x0000000000000000 0x000000000000000f
 0x0000000003fffe00 0xc00000000b8b2800 0x0000000000000016 0xe00000009ed94758
 0x0000000000000005 0x00000000300ee0ac 0x00000000300ee10b 0x000000003049b590
 0x0000000031c03b40 0x0000000000000001 0x0000000000000004 0x0000000000000003
 0x000000003049b590 0x00000000303141c0 0x0000000000001400 0x0000002acc939203
LPCR  : 0x000000000000700a
PTCR  : 0x0000000000010004
LPIDR : 0x0000000000000000
PIDR  : 0x0000000000000004
HFSCR : 0x000000000000059f
HDSISR: 0x00000000
HDAR  : 0x0000000000000000
HEIR : 0x0000000000000000
HID0 : 0x0080000000000000
HSRR0 : 0xc00000000312759c
HSRR1 : 0x9000000000009032
HDEC  : 0xffffffd97c942b63
HSPRG0: 0xc00000000235f780
HSPRG1: 0xc0000019ff7ff480
FSCR  : 0x0000000000000000
DSISR : 0x02000000
DAR   : 0xe0000000baf08ff8
SRR0  : 0xc00000000312e558
SRR1  : 0x9000000000009032
DEC   : 0x000000009197eb83
TB    : 0x0000002c85379be7
SPRG0 : 0xc00000000400d300
SPRG1 : 0xe00000009ed94620
SPRG2 : 0x0000000020000408
SPRG3 : 0x00000000000007a0
PPR   : 0x0004000000000000
[... 126 lines cut ...]
   19458:       7c 21 0b 78     mr      r1,r1
   1945c:       7f ff fb 78     mr      r31,r31
-> 19460:       7c 6c 42 a6     mftb    r3
   19464:       7f bf 18 40     cmpld   cr7,r31,r3
   19468:       41 9e 00 0c     beq     cr7,19474 <.cpu_idle_delay+0xb4>
   1946c:       7d 23 f8 51     subf.   r9,r3,r31
   19470:       40 81 00 28     ble     19498 <.cpu_idle_delay+0xd8>
   19474:       3d 42 00 1f     addis   r10,r2,31
   19478:       89 2a 7e b0     lbz     r9,32432(r10)
   1947c:       2f 89 00 00     cmpwi   cr7,r9,0
   19480:       41 9e ff e0     beq     cr7,19460 <.cpu_idle_delay+0xa0>
   19484:       7c 63 f8 50     subf    r3,r3,r31
   19488:       7f be 18 40     cmpld   cr7,r30,r3
   1948c:       40 9c ff d4     bge     cr7,19460 <.cpu_idle_delay+0xa0>
   19490:       7c 42 13 78     mr      r2,r2
   19494:       4b ff ff 74     b       19408 <.cpu_idle_delay+0x48>
   19498:       7c 42 13 78     mr      r2,r2
   1949c:       38 21 00 90     addi    r1,r1,144
   194a0:       48 0a 74 68     b       c0908 <_restgpr0_29>
   194a4:       00 00 00 00     .long 0x0
   194a8:       00 00 00 01     .long 0x1
   194ac:       80 03 00 00     lwz     r0,0(r3)
```

I'm not entirely certain that FreeBSD has the timebase fired up at this point or not. I guess I should do a check if this is a general problem with moving the window around...

oohal, 2019-04-29:
I have a vague memory of this being due a bug in mboxd on the BMC. @amboar any ideas?

hegdevasant, 2019-04-29:
> Manually-constructed backtrace:
> .bt_add_ipmi_msg_head
> .time_wait_nopoll
> .ipmi_queue_msg_sync
> .hiomap_queue_msg_sync
> .hiomap_window_move
> .ipmi_hiomap_cmd_cb
> .blocklevel_raw_read
> .opal_flash_op
> opal_entry

Thanks for detailed report. Specially manual backtrace. its easy to debug now.

> I have a suspicion that this might be a corner case in 7f29116 -- perhaps it's attempting a read past the > end of the device?

I don't think its creating issue. 

Recently we had similar issue reported on qemu environment as well.  I think below patch should solve your problem. Please try below patch and let me know how it goes.

[1] https://lists.ozlabs.org/pipermail/skiboot/2019-April/013897.html

-Vasant

bdragon28, 2019-04-29:
Yes, that patch DOES seem to do the trick! Nice call!

```
opalflash_getattr()
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_strategy()
opalflash_read offset 512 count 512
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_strategy()
opalflash_read offset 67108352 count 512
opalflash_getattr()
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash close()
opalflash open()
opalflash_strategy()
opalflash_read offset 67108352 count 512
opalflash_strategy()
opalflash_read offset 65536 count 8192
opalflash_strategy()
opalflash_read offset 8192 count 8192
opalflash_strategy()
opalflash_read offset 0 count 8192
opalflash_strategy()
opalflash_read offset 262144 count 8192
opalflash_strategy()
opalflash_read offset 65536 count 8192
opalflash_strategy()
opalflash_read offset 8192 count 8192
opalflash_strategy()
opalflash_read offset 0 count 8192
opalflash_strategy()
opalflash_read offset 262144 count 8192
opalflash_strategy()
opalflash_read offset 32768 count 512
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_strategy()
opalflash_read offset 1024 count 512
opalflash_strategy()
opalflash_read offset 8192 count 512
opalflash_strategy()
opalflash_read offset 65536 count 512
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_getattr()
opalflash close()
opalflash open()
opalflash_getattr()
opalflash close()
```

amboar, 2019-04-30:
Glad to hear the patch @hegdevasant pointed out fixes your issue @bdragon28. @stewart-ibm / @legoater it would be nice to merge that patch given it's breaking non-emulated accesses as well. How strongly are we feeling about the `TODO`? How confident are we w.r.t. the locking given taking the nopoll() path is what's preventing progress? I think I traced through it and convinced myself there wasn't a problem prior to going on leave, but that's pretty fuzzy now.

FWIW the patchwork equivalent is here: http://patchwork.ozlabs.org/patch/1077344/

stewart-ibm, 2019-04-30:
Yeah, I remember thinking "i need to sit down and convince myself everything is kosher there" when seeing that patch.

stewart-ibm, 2019-04-30:
As an aside, is FreeBSD calling `OPAL_POLL_EVENTS` or similar every `/ibm,opal/ibm,heartbeat-ms` milliseconds?

That mechanism is kind of a catch-all for these kinds of bugs, as well as in situations (such as bring-up and various simulators) where we don't have any way to generate an interrupt to ourselves to do periodic tasks.

bdragon28, 2019-04-30:
FreeBSD doesn't even have SMP or timers up at the point in startup that this is happening.

bdragon28, 2019-04-30:
FreeBSD has an async management layer that does polling on tokens when something needs to act as if it were a synchronous API.
See https://svnweb.freebsd.org/base/head/sys/powerpc/powernv/opal_async.c?revision=345974&view=markup#l101

It was expected that this was sufficient to pump the state machine. If this is not the case early in the process, I guess we'll need to adjust our code.

bdragon28, 2019-04-30:
But of course the problem is that opal has control of the only thread that is up, and doesn't return control. None of the other threads have started, so there's no way for FreeBSD to make progress unless a timer has already been set up, I guess.

So yeah, before SMP and timers, I guess there's no actual way to solve this from the FreeBSD side.

chmeeedalf, 2019-04-30:
Hi, I wrote the OPAL flash driver for FreeBSD, and the expectation of the driver was that any asynchronous requirements would be met by the OPAL API contract.  We do expect that if there's any asynchronous actions required that it return appropriately, and our asynchronous token management subsystem will handle both the heartbeat completing the token and single-threaded local completions in cases that the heartbeat isn't running.

Since the new HIOMAP backend does expect asynchronous behaviors, it needs to either inform the OS of such, through returning from OPAL_FLASH_READ, or handle the timers on its own.  Cédric's patch looks correct to solving this issue under the covers, but as the TODO in the patch mentions, it probably should include a timeout to push the problem back up to the OS, instead of getting stuck.

There's new firmware ready to publish for the Talos II, and I would like to see this issue fixed for that release, so I hope this gets merged in soon so FreeBSD isn't stuck.  I can add a workaround in FreeBSD if needed, but would prefer not to since the patchwork patch looks ready to land.

madscientist159, 2019-04-30:
@chmeeedalf The patch is being cherry picked into v2.00 of the Talos II firmware (and by extension the upcoming Blackbird firmware).  A third and (hopefully final) beta version should be posted later today for testing.

stewart-ibm, 2019-05-02:
https://patchwork.ozlabs.org/patch/1093619/ as an alternative to the original https://patchwork.ozlabs.org/patch/1077344/ patch is likely what I'd look at, although cherry-picking https://patchwork.ozlabs.org/patch/1077344/ **should** be fine from a practical point of view (I haven't gone through and audited all the code paths for locking, so take it with a maintainer-sized-grain-of-salt :)

stewart-ibm, 2019-05-02:
> FreeBSD has an async management layer that does polling on tokens when something needs to act as if it were a synchronous API.
> See [...]

Interestingly enough, you're the only user (ever possibly) of the OPAL_CHECK_ASYNC_COMPLETION call. Linux doesn't use it, instead using the generic OPAL_POLL_EVENTS and looking for  OPAL_MSG_ASYNC_COMP message.

Looking back into the deep-history of the Linux OPAL code, and indeed the pre-open source history of skiboot itself, it looks like the OPAL_CHECK_ASYNC_COMPLETION call was added around November 2013 (POWER8 OPAL systems first GA'd in June 2014 IIRC) but never actually called in the kernel.

The good news is that the call appears to work (although I'm sorry to say it's never been tested), the bad news is that OPAL_CHECK_ASYNC_COMPLETION isn't enough to crank any state machine, it strictly checks the status of the async op. So anything spinning on waiting for that will end up relying on something else to crank it.

.. and not even looking at but just remembering the state of our OPAL API docs, all I can do is apologize for them.

stewart-ibm, 2019-05-02:
> Hi, I wrote the OPAL flash driver for FreeBSD, and the expectation of the driver was that any asynchronous requirements would be met by the OPAL API contract. We do expect that if there's any [...]

As discussed on the list, timeout should be okay, but I have gotten someone to look (properly) at doing it fully async.

The reason it isn't today was that getting hiomap all in, done, and working, was a decently high priority back when due to it being needed to address CVE-2019-6260 (https://www.flamingspork.com/blog/2019/01/23/cve-2019-6260-gaining-control-of-bmc-from-the-host-processor/ if interested in details) and we figured we could live with the probably-only-slight-amount-of-long-time spent in OPAL for the near future (especially considering that OPAL_FLASH_READ could block for a lot longer if given the right parameters).

> 
> There's new firmware ready to publish for the Talos II, and I would like to see this issue fixed for that release, so I hope this gets merged in soon so FreeBSD isn't stuck. I can add a workaround in FreeBSD if needed, but would prefer not to since the patchwork patch looks ready to land.

stewart-ibm, 2019-05-02:
@bdragon28 does https://patchwork.ozlabs.org/patch/1093619/ also fix the problem?

(at some point it'd be great to add FreeBSD into various bits of testing we have, another OPAL API consumer is certainly exciting)

bdragon28, 2019-05-04:
I'll check this over the weekend, thanks for following up.

bdragon28, 2019-05-05:
Yes, the reworked patch from https://patchwork.ozlabs.org/patch/1093619/ appears to work as well.
```opalflash_getattr()
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_strategy()
opalflash_read offset 512 count 512
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_strategy()
opalflash_read offset 67108352 count 512
opalflash_getattr()
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash close()
opalflash open()
opalflash_strategy()
opalflash_read offset 67108352 count 512
opalflash_strategy()
opalflash_read offset 65536 count 8192
opalflash_strategy()
opalflash_read offset 8192 count 8192
opalflash_strategy()
opalflash_read offset 0 count 8192
opalflash_strategy()
opalflash_read offset 262144 count 8192
opalflash_strategy()
opalflash_read offset 65536 count 8192
opalflash_strategy()
opalflash_read offset 8192 count 8192
opalflash_strategy()
opalflash_read offset 0 count 8192
opalflash_strategy()
opalflash_read offset 262144 count 8192
opalflash_strategy()
opalflash_read offset 32768 count 512
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_strategy()
opalflash_read offset 1024 count 512
opalflash_strategy()
opalflash_read offset 8192 count 512
opalflash_strategy()
opalflash_read offset 65536 count 512
opalflash_strategy()
opalflash_read offset 0 count 512
opalflash_getattr()
opalflash close()
opalflash open()
opalflash_getattr()
opalflash close()
```

hegdevasant, 2019-05-13:
@oohal merged https://patchwork.ozlabs.org/patch/1093619/ to master last week.

This issue is fixed in master and v.6.3.1 release.

-Vasant

closed 2019-05-13 by hegdevasant