open-power/skiboot #245: make install fails for external/gard

sharkcz, 2019-07-16:
make install fails for external/gard with
````
+ make -C external/gard install DESTDIR=/home/dan/rpmbuild/BUILDROOT/opal-prd-6.4-1.fc31.ppc64le prefix=/usr
make: Entering directory '/home/dan/projects/fedora/ppc/opal-prd/skiboot-6.4/external/gard'
cc -O2 -Wall -Werror -Wno-stringop-truncation -I. -c version.c -o version.o
cc  -O2 -Wall -Werror -Wno-stringop-truncation -I. version.o gard.o units.o libflash-libflash.o libflash-libffs.o libflash-ecc.o libflash-blocklevel.o libflash-file.o ccan-list-list.o common-arch_flash.o -o opal-gard
install -D gard /home/dan/rpmbuild/BUILDROOT/opal-prd-6.4-1.fc31.ppc64le/usr/sbin/opal-gard
install: cannot stat 'gard': No such file or directory
make: *** [rules.mk:47: install] Error 1
````
Fix available in https://lists.ozlabs.org/pipermail/skiboot/2019-July/014950.html

hegdevasant, 2019-07-17:
Thanks for the fix. Looks good to me.

-Vasant

oohal, 2019-07-30:
Fixed in 53baedfd19436e8cc33151a0c17cb57863f7303a

closed 2019-07-30 by oohal