open-power/skiboot #246: v 6.2.2 Fails to build with gcc9-9.1.1+r273003

hramrach, 2019-07-23:
[   57s] hw/sbe-p9.c: In function 'p9_sbe_msg_send':
[   57s] hw/sbe-p9.c:270:9: error: taking address of packed member of 'struct p9_sbe_msg' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   57s]   270 |  data = &msg->reg[0];
[   57s]       |         ^~~~~~~~~~~~
[   57s] hw/sbe-p9.c: In function 'p9_sbe_msg_receive':
[   57s] hw/sbe-p9.c:303:9: error: taking address of packed member of 'struct p9_sbe_msg' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   57s]   303 |  data = &resp->reg[0];
[   57s]       |         ^~~~~~~~~~~~~
[   57s] hw/sbe-p9.c: In function 'p9_sbe_msg_complete':
[   57s] hw/sbe-p9.c:326:11: error: taking address of packed member of 'struct p9_sbe_msg' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   57s]   326 |  list_del(&msg->link);
[   57s]       |           ^~~~~~~~~~
[   57s] hw/sbe-p9.c: In function 'p9_sbe_queue_msg':
[   57s] hw/sbe-p9.c:423:32: error: taking address of packed member of 'struct p9_sbe_msg' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   57s]   423 |  list_add_tail(&sbe->msg_list, &msg->link);
[   57s]       |                                ^~~~~~~~~~
[   57s] hw/sbe-p9.c: In function 'p9_sbe_cancelmsg':
[   57s] hw/sbe-p9.c:480:11: error: taking address of packed member of 'struct p9_sbe_msg' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   57s]   480 |  list_del(&msg->link);
[   57s]       |           ^~~~~~~~~~

oohal, 2019-07-23:
@hegdevasant can you have a look

hramrach, 2019-07-23:
Already fixed in master.

hegdevasant, 2020-11-18:
v6.2.x is unmaintained .. Current stable version is v6.6.x and soon it will be v6.7.x .. Hence closing this one.

Let me know , if you still need this issue to be addressed in v6.2.x branch.

-Vasant

closed 2020-11-18 by hegdevasant