open-power/skiboot #259: Big-Core support broke fast-reboot

klauskiwi, 2020-10-06:
See some context here - https://github.com/open-power/op-test/pull/605

But essentially, looks like the Big-Core patches broke fast-reboot (and perhaps other things) in a significant way

bisecting shows that fast-reboot works until commit 98f5834253c7e940f961ad68a02ff13f817b06fe but I suspect that the earlier Big-Core patches may have changed some ad-hoc assumption about threads or how they organize in the fast-reboot path.

reproduce is easy by building the aforementioned commit and using op-test to flash/perform the fast-reboot test:

`./op-test --flash-skiboot skiboot.lid.xz.stb --run testcases.OpTestFastReboot.OpTestFastReboot`

oohal, 2020-10-06:
Patches welcome.

hegdevasant, 2020-11-30:
Below commit fixes this issue:

closed 2020-11-30 by hegdevasant