open-power/skiboot #260: Checking something: Endianness of scom addresses

hanetzer, 2020-10-15:
https://github.com/open-power/skiboot/blob/f901fcafae14d38e29f1cc11440086ee678785d0/hw/xscom.c#L108
So I know the p9 can run in either big or little endian mode, but this bit of code leads me to believe that
scom addresses are always big endian, regardless of whether the cpu is running in big or little mode.

Just a bit of clarification would be nice, thank you.

oohal, 2020-10-15:
SCOM addresses are just offsets into the XSCOM MMIO area. The SCOMs themselves are big endian MMIO registers so we use the in_be64()  and out_be64() when accessing them.

hanetzer, 2020-10-15:
Oh, so like, recv_status_reg is at `0x603fc00000000 + recv_status_reg` on p9 systems?
And I assume that's true for any other scom reg? Neat.

oohal, 2020-10-16:
More or less. Historically SCOMs were accessed from outside the chip via a address/data register pair and the "scom address" needs to be translated from a register number to an offset into the XSCOM MMIO area. For P9 all that requires is shifting the address up by three bits, but it was a little more complex on P8 which is why we use `xscom_addr()` to do the translation.

The other thing is that each chip has its own xscom MMIO range so `0x603fc00000000 + (recv_status_reg << 3)` is the address for chip 0 and `0x623fc00000000 + recv_status_reg` is the same scom on the 2nd chip. `xscom_addr()` also handles that.