open-power/skiboot #264: HDAT documentation

macpijan, 2021-05-20:
Hi,

We are working on coreboot support for POWER9.
Recently we have achieved the milestone where we could initialize the
memory on the Talos II platform and start the coreboot ramstage phase:
https://github.com/3mdeb/openpower-coreboot-docs/blob/main/releases.md

What we would like to achieve next is to start skiboot from coreboot.

We can see that there is this HDAT interface between Hostboot and skiboot
(https://open-power.github.io/skiboot/doc/overview.html#booting) which we probably
would need to reimplement.

Sadly, the documentation (https://open-power.github.io/skiboot/doc/overview.html#booting)
says that "The HDAT specification is currently not public.".
Any idea why is that? Can we  get some more information about
the HDAT somewhere (apart from reading the Hostboot code)?

Thanks,

macpijan, 2021-05-20:
This question was also asked here: https://lists.ozlabs.org/pipermail/openpower-firmware/2021-May/000641.html
And it looks like using the devicetree might be an option.

oohal, 2021-05-22:
I'll 2nd the suggestion Stewart made in that thread and say you should generate a DTB and pass that to skiboot.

The HDAT spec is less of a spec and more of a description of what the IBM proprietary firmware stack happens to be doing for each system generation. It's probably never going to become public so I think you're better off just ignoring it entirely. I don't think anyone thought having skiboot consume HDAT on OpenPower systems was a really good idea. It was just convenient at the time.

macpijan, 2021-05-26:
@oohal Thank you! Are there any pointers to how this DTB should look like? How is the DTB presented by the earlier firmware to `skiboot` different from the one that `skiboot` produces (https://github.com/open-power/skiboot/blob/master/doc/device-tree.rst) ?

hegdevasant, 2021-06-01:
> @oohal Thank you! Are there any pointers to how this DTB should look like?

Look into `/usr/src/devtree` in hostboot/master-p8 branch. That will give you an idea.

> How is the DTB presented by the earlier firmware to `skiboot` different from the one that `skiboot` produces (https://github.com/open-power/skiboot/blob/master/doc/device-tree.rst) ?

Hosbtoot provided DT contains only hardware details. We expand that in skiboot and add other nodes like `/ibm,opal` , `/firmware` etc before passing it to host. 

-Vasant