open-power/skiboot #265: GCC11: problems to compile skiboot with gcc11

jziviani, 2021-07-02:
Hello!

Our build system is currently using GCC11 but we're facing problems to build skiboot with that compiler version. Basically, the errors are:

```
# make
        [CC]  asm/asm-offsets.s
        [GN]  include/asm-offsets.h
        [AS]  asm/head.o
        [AS]  asm/misc.o
        [AS]  asm/kernel-wrapper.o
        [AS]  asm/cvc_entry.o
        [AR]  asm/built-in.a
        [CC]  core/relocate.o
        [CC]  core/console.o
In file included from /root/skiboot/include/skiboot.h:10,
                 from core/console.c:10:
core/console.c: In function ‘clear_console’:
/root/skiboot/libc/include/string.h:34:16: error: ‘__builtin_memset’ offset [0, 1048575] is out of the bounds [0, 0] [-Werror=array-bounds]
   34 | #define memset __builtin_memset
core/console.c:82:9: note: in expansion of macro ‘memset’
   82 |         memset(con_buf, 0, INMEM_CON_LEN);
      |         ^~~~~~
cc1: all warnings being treated as errors
make: *** [/root/skiboot/Makefile.rules:59: core/console.o] Error 1
```

And even if I disable "**warning as errors**", a new problem appears:

```
# make CFLAGS=" -Wno-error"
        [CC]  core/console.o
        [CC]  core/stack.o
        [CC]  core/init.o
In file included from /root/skiboot/include/cpu.h:7,
                 from core/init.c:12:
/root/skiboot/include/processor.h: In function ‘mtmsrd’:
/root/skiboot/include/processor.h:306:9: warning: ‘asm’ operand 1 probably does not match constraints
  306 |         asm volatile("mtmsrd %0,%1" : : "r"(val), "i"(l) : "memory");
      |         ^~~
/root/skiboot/include/processor.h:306:9: error: impossible constraint in ‘asm’
In file included from /root/skiboot/include/skiboot.h:10,
                 from core/init.c:8:
core/init.c: In function ‘copy_exception_vectors’:
/root/skiboot/libc/include/string.h:36:16: warning: ‘__builtin_memcpy’ writing 12032 bytes into a region of size 0 overflows the destination [-Wstringop-overflow=]
   36 | #define memcpy __builtin_memcpy
core/init.c:870:9: note: in expansion of macro ‘memcpy’
  870 |         memcpy((void *)0x100, (void *)(SKIBOOT_BASE + 0x100),
      |         ^~~~~~
make: *** [/root/skiboot/Makefile.rules:59: core/init.o] Error 1
```

Thank you

hegdevasant, 2021-09-13:
commit 8246de86360666d81e71785c1c8be03a72563d17
Author: Joel Stanley <joel@jms.id.au>
Date:   Thu Jul 29 17:24:38 2021 +0930

    Makefile: Avoid errors with GCC 11

Above commit should resolve this issue.. 

But yes. we should still go through all warnings and fix them.

-Vasant

jziviani, 2021-09-13:
Thank you @hegdevasant ! I think you can close this bug then. :+1:

closed 2021-09-28 by hegdevasant