open-power/skiboot #270: Test with GitHub Actions

ruscur, 2021-11-03:
I used to run the snowpatch tests for skiboot that showed up on https://patchwork.ozlabs.org/project/skiboot/list/ - these were triggered through a Jenkins server on openpower.xyz that no longer exists (the domain and the server are both gone).   I rewrote snowpatch to use GitHub Actions instead, and that's what's being used for linuxppc-dev and pdbg currently.

We should write some workflows for GitHub Actions that can be run on every commit to this repo, can be used by developers if they fork on GitHub during development, and consumed by snowpatch to be run on every series on the list & published to Patchwork.  If this is wanted, I would think the best way to go would be for me to start using them in my skiboot repo for snowpatch use and then send the workflows upstream once they're stable.

So what would we want to test?

- `make` / `make check` on a few different distros with different toolchains
- qemu powernv boots with P8/P9/P10? models
- mambo boots (dunno how well this would go for P8/P9 yet because the distro packages here http://public.dhe.ibm.com/software/server/powerfuncsim/ are pretty old)

I'm assuming https://github.com/open-power/op-build/releases/ is the best thing to use for petitboot/kernel.

extra stuff:

- basic tests for stuff in `external/` that only trigger when there's commits touching them
- static analysis of some sort?

Anything else we might want?

--
Some references:

- snowpatch https://github.com/ruscur/snowpatch/tree/rewrite
- simplest possible workflow (pdbg) https://github.com/ruscur/pdbg/blob/master/.github/workflows/build.yml
- linuxppc-dev with smart sparse diff + code annotations https://patchwork.ozlabs.org/project/linuxppc-dev/patch/20211028101822.14278-1-yao.jing2@zte.com.cn/
- previous skiboot job on Jenkins that just did `make check` with qemu https://github.com/ajdlinux/openpower.xyz-snowpatch/blob/master/skiboot-build.sh
- linuxppc documentation on simple GitHub Actions usage for developers https://github.com/linuxppc/wiki/wiki/Testing-with-GitHub-Actions

legoater, 2021-11-03:
> used to run the snowpatch tests for skiboot that showed up on https://patchwork.ozlabs.org/project/skiboot/list/ - these were triggered through a Jenkins server on openpower.xyz that no longer [...]

What was really useful was the tests being done automatically on patches received on the list. At least the one checking that the patch applied on upstream. 

> We should write some workflows for GitHub Actions that can be run on every commit to this repo, can be used by developers if they fork on GitHub during development, and consumed by snowpatch to be [...]

@fbarrat has given this a  try here : 
> So what would we want to test?
>
>     * `make` / `make check` on a few different distros with different toolchains
>
>     * qemu powernv boots with P8/P9/P10? models
>
>     * mambo boots (dunno how well [...]

legoater, 2021-11-03:
> I used to run the snowpatch tests for skiboot that showed up on https://patchwork.ozlabs.org/project/skiboot/list/ - these were triggered through a Jenkins server on openpower.xyz that no longer [...]

The tests being run on patches reaching the list were very useful. Specially the one checking that the patch applied. Would that be still possible ? 

> We should write some workflows for GitHub Actions that can be run on every commit to this repo, can be used by developers if they fork on GitHub during development, and consumed by snowpatch to be [...]

@fbarrat has given GitHub Actions a try :

  https://github.com/fbarrat/skiboot/actions/runs/1399852239 

Looks good.

> So what would we want to test?
> 
>     * `make` / `make check` on a few different distros with different toolchains

yes. Latest compilers should be checked. Is there a minimum version requirement ? 

>     * qemu powernv boots with P8/P9/P10? models

QEMU has a PowerNV machine acceptance test we could use :

  https://git.qemu.org/?p=qemu.git;a=blob;f=tests/acceptance/boot_linux_console.py;h=06fc967f6c4e68b15ca32416a0cf589c1277d3d3;hb=HEAD#l1179

It's upstream (P10 is not but there are other QEMU trees) 

We could add a few other tests with a more complex topology may be ? @fbarrat ? 

>     * mambo boots (dunno how well this would go for P8/P9 yet because the distro packages here http://public.dhe.ibm.com/software/server/powerfuncsim/ are pretty old)

P8 is old. P9 is from 2019 and P10 from 2021. They should be fine.

> I'm assuming https://github.com/open-power/op-build/releases/ is the best thing to use for petitboot/kernel.
> 
> extra stuff:
> 
>     * basic tests for stuff in `external/` that only trigger when there's commits touching them

Could we compile some of the tools for ARM ? 

>     * static analysis of some sort?

Coverity.

> Anything else we might want?

I think this is good enough for a start.

ruscur, 2021-11-05:
Up and running from @fbarrat: 89156c0583b40167cc67fca835cbf2c4e3eb15d2

snowpatch is now running with what's in the master branch: https://patchwork.ozlabs.org/project/skiboot/patch/20210908183901.3773783-2-grimm@linux.ibm.com/

Nothing else recent on the list still applies as-is but it should be useful for anything new coming in.