open-power/skiboot #271: Failing to build against OpenSSL 3.0 (EVP_PKEY_get1_EC_KEY deprecation)

cpaelzer, 2021-11-26:
Hi,
this was found as part of the Ubuntu openssl 3 transition and repoorted there as
https://bugs.launchpad.net/ubuntu/+source/qemu/+bug/1952448

# Error example

It causes the following error:

```
...
	[ HOSTCC ] libstb/create-container.c
libstb/create-container.c: In function ‘getPublicKeyRaw’:
libstb/create-container.c:72:17: error: ‘EVP_PKEY_get1_EC_KEY’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
   72 |                 key = EVP_PKEY_get1_EC_KEY(pkey);
      |                 ^~~
In file included from libstb/create-container.c:13:
/usr/include/openssl/evp.h:1374:19: note: declared here
 1374 | struct ec_key_st *EVP_PKEY_get1_EC_KEY(EVP_PKEY *pkey);
      |                   ^~~~~~~~~~~~~~~~~~~~
libstb/create-container.c:76:17: error: ‘EC_KEY_get0_group’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
   76 |                 ecgrp = EC_KEY_get0_group(key);
      |                 ^~~~~
In file included from libstb/create-container.c:11:
/usr/include/openssl/ec.h:1032:39: note: declared here
 1032 | OSSL_DEPRECATEDIN_3_0 const EC_GROUP *EC_KEY_get0_group(const EC_KEY *key);
      |                                       ^~~~~~~~~~~~~~~~~
libstb/create-container.c:80:17: error: ‘EC_KEY_get0_public_key’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
   80 |                 ecpoint = EC_KEY_get0_public_key(key);
      |                 ^~~~~~~
In file included from libstb/create-container.c:11:
/usr/include/openssl/ec.h:1060:39: note: declared here
 1060 | OSSL_DEPRECATEDIN_3_0 const EC_POINT *EC_KEY_get0_public_key(const EC_KEY *key);
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
libstb/create-container.c:84:17: error: ‘EC_POINT_point2bn’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
   84 |                 pubkeyBN = EC_POINT_point2bn(ecgrp, ecpoint, POINT_CONVERSION_UNCOMPRESSED,
      |                 ^~~~~~~~
In file included from libstb/create-container.c:11:
/usr/include/openssl/ec.h:768:31: note: declared here
  768 | OSSL_DEPRECATEDIN_3_0 BIGNUM *EC_POINT_point2bn(const EC_GROUP *,
      |                               ^~~~~~~~~~~~~~~~~
libstb/create-container.c:89:17: error: ‘EC_KEY_free’ is deprecated: Since OpenSSL 3.0 [-Werror=deprecated-declarations]
   89 |                 EC_KEY_free(key);
      |                 ^~~~~~~~~~~
In file included from libstb/create-container.c:11:
/usr/include/openssl/ec.h:1001:28: note: declared here
 1001 | OSSL_DEPRECATEDIN_3_0 void EC_KEY_free(EC_KEY *key);
      |                            ^~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [/root/skiboot/libstb/Makefile.inc:25: libstb/create-container] Error 1
```

# Repro steps including good/bad case steps

```
# Get a system with openssl 3, like jammy-proposed right now
https://cloud-images.ubuntu.com/jammy/current/

$ apt-get install gcc-powerpc64le-linux-gnu gcc valgrind \
        expect libssl-dev device-tree-compiler make \
        xz-utils libmbedtls-dev
$ git clone https://github.com/open-power/skiboot.git
$ cd skiboot
$ CROSS=powerpc64le-linux-gnu- make
```

# Working so far, switch to openssl 3 in proposed

```
$ echo "deb http://archive.ubuntu.com/ubuntu jammy-proposed main restricted universe multiverse" >> /etc/apt/sources.list
$ apt update
$ apt upgrade
$ CROSS=powerpc64le-linux-gnu- make clean
$ CROSS=powerpc64le-linux-gnu- make
```

Currently the two ssl versions you can switch between in Ubuntu Jammy are
```
apt-cache policy libssl-dev
libssl-dev:
  Installed: 3.0.0-1ubuntu1
  Candidate: 3.0.0-1ubuntu1
  Version table:
 *** 3.0.0-1ubuntu1 500
        500 http://archive.ubuntu.com/ubuntu jammy-proposed/main amd64 Packages
        100 /var/lib/dpkg/status
     1.1.1l-1ubuntu1 500
        500 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages
```

And with the 3.0 version this time the build breaks due to a deprecation as reported above.

# References

Upstream openssl outlines the steps for that deprecation at
https://www.openssl.org/docs/man3.0/man3/EVP_PKEY_get1_EC_KEY.html

erichte-ibm, 2022-01-19:
I've sent [a patch to the mailing list](https://lists.ozlabs.org/pipermail/skiboot/2022-January/018145.html) that should address this problem. I haven't been able to test it fully integrated into op-build and on actual hardware yet to confirm the usage of the new API is exactly compatible with the old deprecated functions, but from my limited inspection of building inside and outside of an Ubuntu Jammy (w/ OpenSSL 3.0) docker container, the resulting data appears to be identical.

rarbab, 2023-07-12:
We can close this. The patch above was applied in commit 9a1f95f87004.

closed 2023-07-12 by rarbab