open-power/skiboot #273: Rainier:PHB still fenced !!

lili-lilili, 2023-01-10:
I am testing OPAL on the rainier，and skiboot often reports error: "CRESET: PHB still fenced !!".
There's an nvme card plugged into the slot behind this PHB.
Important: this problem only occurs when there is no nvme disk insert on the card.

Here is the log.
See attachment for complete log.
> [  182.241939621,3] PHB#0000[0:0]: LINK: Timeout waiting for link up
> [  182.247955720,3] PHB#0000[0:0]: CRESET: PHB still fenced !!
> [  182.248162118,3] PHB#0000[0:0]:             PCI FIR=1000000000000000
> [  182.248200851,3] PHB#0000[0:0]:         PCI FIR WOF=1000000000000000
> [  182.248236708,3] PHB#0000[0:0]:            NEST FIR=0000800000000000
> [  182.248269153,3] PHB#0000[0:0]:        NEST FIR WOF=0000002000000000
> [  182.248305109,3] PHB#0000[0:0]:            ERR RPT0=0000000000000001
> [  182.248337394,3] PHB#0000[0:0]:            ERR RPT1=0000000000000000
> [  182.248376417,3] PHB#0000[0:0]:             AIB ERR=00ce100000000000
> [  183.481572548,5] PHB#0000:00:00.0 [ROOT] 1014 0652 R:00 C:060400 B:01..1a 
> [  183.481819812,5] PHB#0000:01:00.0 [SWUP] 11f8 4052 R:00 C:060400 B:02..1a LOC_CODE=U78DB.ND0.WZS008L-P0
> [  183.482175973,5] PHB#0000:02:00.0 [SWDN] 11f8 4052 R:00 C:060400 B:03..03 
> [  183.482333011,5] PHB#0000:03:00.0 [EP  ] 144d a80a R:00 C:010802 (  mass-storage) LOC_CODE=U78DB.ND0.WZS008L-P0
> [  183.482924386,5] PHB#0000:02:01.0 [SWDN] 11f8 4052 R:00 C:060400 B:04..04 
> [  183.483086742,5] PHB#0000:04:00.0 [EP  ] 144d a80a R:00 C:010802 (  mass-storage) LOC_CODE=U78DB.ND0.WZS008L-P0
> [  183.483810920,5] PHB#0000:02:02.0 [SWDN] 11f8 4052 R:00 C:060400 B:05..05 
> [  183.484414564,5] PHB#0000:05:00.0 [EP  ] 144d a80a R:00 C:010802 (  mass-storage) LOC_CODE=U78DB.ND0.WZS008L-P0
> [  183.484745988,5] PHB#0000:02:03.0 [SWDN] 11f8 4052 R:00 C:060400 B:06..06 
> [  183.484846897,5] PHB#0000:06:00.0 [EP  ] 144d a80a R:00 C:010802 (  mass-storage) LOC_CODE=U78DB.ND0.WZS008L-P0
> [  183.485317556,5] PHB#0000:02:04.0 [SWDN] 11f8 4052 R:00 C:060400 B:07..0b 
> [  183.485549585,5] PHB#0000:02:05.0 [SWDN] 11f8 4052 R:00 C:060400 B:0c..10 
> [  183.485779716,5] PHB#0000:02:06.0 [SWDN] 11f8 4052 R:00 C:060400 B:11..15 
> [  183.486178295,5] PHB#0000:02:07.0 [SWDN] 11f8 4052 R:00 C:060400 B:16..1a 
> [  183.486451637,5] PHB#0000:01:00.1 [EP  ] 11f8 4052 R:00 C:058000 (memory-controller) LOC_CODE=U78DB.ND0.WZS008L-P0

It seems that link training failed when freset, then creset is executed, and then this error is reported.
I enable link training trace by cmd: nvram -p ibm,skiboot --update-config pci-tracing=true.
And found that sometimes the training state machine is not correct: 
the first state is "polling", and the next state is detect, then training time out. 
I'm not sure why training started from "polling", and I'm not sure if this is the reason for triggering PHB fenced?

Here is the log.
See attachment for complete log
> [  162.468834658,7] PHB#0000:00:00.0 PCI: Registered PHB
> [  162.468836359,6] PHB#0000[0:0]: Found /pciex@600c3c0000000 @0x600c3c0000000
> [  162.468837412,6] PHB#0000[0:0]:   M32 [0x000600c000000000..0x000600c07fffffff]
> [  162.468838359,6] PHB#0000[0:0]:   M64 [0x0006000000000000..0x0006003fffffffff]
> [  162.468840454,7] PHB#0000[0:0]: Core revision 0xa50002
> [  162.468841475,7] PHB#0000[0:0]: Found 512 max PEs and 4096 IRQs 
> [  162.468842642,6] PHB#0000[0:0]: Max link speed: GEN5, max link width 8
> [  162.468843601,7] PHB#0000[0:0]: Override lane equalization settings:
> [  162.468844133,7] PHB#0000[0:0]:   0x4444444444444444 0x4444444444444444
> [  162.468845010,7] PHB#0000[0:0]:   0x4444444444444444 0x4444444444444444
> [  162.468845858,7] PHB#0000[0:0]:   0x4444444444444444 0x4444444444444444
> [  162.468846563,7] PHB#0000[0:0]:   0x9999999999999999 0x9999999999999999
> [  162.468933137,7] PHB#0000[0:0]: Initializing PHB...
> [  162.468933990,7] PHB#0000[0:0]: Default system config: 0x4410000050400000
> [  162.468935000,7] PHB#0000[0:0]: New system config    : 0x4410000050400000
> [  162.468935798,7] PHB#0000[0:0]: Initial PHB CRESET is 0xe000000000000000
> [  162.468937077,7] PHB#0000[0:0]: LINK: Disabling scaled flow control
> [  162.468939201,7] PHB#0000[0:0]: Waiting for DLP PG reset to complete...
> [  162.469454260,7] PHB#0000[0:0]: store EOI is enabled
> [  162.469455377,7] PHB#0000[0:0]: Using ABT mode. ESB: 0x0006051fc0000000
> [  162.469456243,7] PHB#0000[0:0]: Reset state SRC_ID: 0ff8000000000000
> [  162.469781810,7] PHB#0000[0:0]: Initialization complete
> [  162.491298081,7] PHB#0000[0:0]: FRESET: Starts
> [  162.491299715,7] PHB#0000[0:0]: FRESET: Prepare for link down
> [  162.491306050,7] PHB#0000[0:0]: FRESET: Assert skipped
> [  162.491429890,7] PHB#0000[0:0]: FRESET: Deassert
> [  162.491687954,5] PHB#0000[0:0]: TRACE:0x0000102101000000  0ms presence GEN1:x16:polling
> [  162.493721058,5] PHB#0000[0:0]: TRACE:0x0000001101000000  3ms          GEN1:x16:detect
> [  165.491685542,5] PHB#0000[0:0]: TRACE: Timed out after 3000ms
> [  165.491686627,7] PHB#0000[0:0]: LINK: Start polling
> [  166.492391250,7] PHB#0000[0:0]: LINK: No in-band presence
> [  167.002877890,7] PHB#0000:00:00.0 Link down
> [  167.002878755,7] PHB#0000:00:00.0 Scanning (upstream only)...
> [  167.003010880,7] PHB#0000:00:00.0 Found VID:1014 DEV:0652 TYP:4 MF- BR+ EX+
> [  167.234048338,5] PHB#0000:00:00.0 [ROOT] 1014 0652 R:00 C:060400 B:00..00 
> [  167.323719699,7] irq feffe name: phb#0000-inf [level]
> [  167.323720630,7] irq fefff name: phb#0000-err [level]
> [  172.138602839,7] PHB#0000[0:0]: Purging all IODA tables...

[msglog-pci_tracing-timeout_err.txt](https://github.com/open-power/skiboot/files/10380078/msglog-pci_tracing-timeout_err.txt)
[obmc-console-log-PHB-still-fenced.txt](https://github.com/open-power/skiboot/files/10380079/obmc-console-log-PHB-still-fenced.txt)

fbarrat, 2023-01-10:
I'm wondering if that's not 2 separate issues:
1. in the 'msglog' file, I don't see the "PHB still fenced" problem, because skiboot didn't see the presence bit in the DLP Training Control  (or more mysteriously, the bit was briefly set, as shown by the extra pci tracing, but it didn't stick). I agree that the link state transitions reported in the extra pci tracing look suspicious.

2. with the log from the console, skiboot must have detected the presence of the card, but training fails. And the following CRESET is not going well as the PHB is still fenced.

Is that hardware configuration always the same? Do you get the 2 scenarios randomly ?

Before pulling a hardware developer to look at it, there's something I'd like to try, hoping it could alter what we see on the first scenario. Could you try with the following patch:

diff --git a/hw/phb4.c b/hw/phb4.c
index f329e130..d640ad80 100644
--- a/hw/phb4.c
+++ b/hw/phb4.c
@@ -6000,7 +6000,7 @@ static void phb4_create(struct dt_node *np)
         * Note: In absence of property (ie, FSP-less), we stick to the old
         * behaviour and set skip_perst to true
         */
-       p->skip_perst = true; /* Default */
+       p->skip_perst = false; /* Default */
 
        iplp = dt_find_by_path(dt_root, "ipl-params/ipl-params");
        if (iplp) {
        

We normally skip setting PERST on the link on a cold boot, but looking at phb4_freset(), I think it's worth trying to check whether the workaround mentioned is also needed in that case.

lili-lilili, 2023-01-11:
> 

Hi fbarrat:

The hardware configuration is the same.
If we don't enable "pci-tracing",  we will encounter the first scenario.
If we enable "pci-tracing", we will encounter the second scenario.
They all happen by probability(about 5%).

We have tried not skip assert PERST, and it doesn't work.
Also, we tried to directly do creset, and it doesn't  work too.

fbarrat, 2023-01-13:
I've checked the hw errata and I didn't find anything you wouldn't already have (the workaround for HW551382 looked interesting, but it's already part of the code you have and the extra tests you've done should have ruled it out).

Do you have ways to contact the hw team to see how to debug it further?

lili-lilili, 2023-01-16:
Unfortunately, I don't have a direct way to contact the hardware team.
Can you help to report this problem to hw team? 
In addition, can you confirm which hardware caused the problem? CPU？ NVMe card?

oohal, 2023-01-16:
> Important: this problem only occurs when there is no nvme disk insert on the card.

...

> [ 182.241939621,3] PHB#0000[0:0]: LINK: Timeout waiting for link up
[ 182.247955720,3] PHB#0000[0:0]: CRESET: PHB still fenced !!
[ 182.248162118,3] PHB#0000[0:0]: PCI FIR=1000000000000000
[ 182.248200851,3] PHB#0000[0:0]: PCI FIR WOF=1000000000000000
[ 182.248236708,3] PHB#0000[0:0]: NEST FIR=0000800000000000
[ 182.248269153,3] PHB#0000[0:0]: NEST FIR WOF=0000002000000000
[ 182.248305109,3] PHB#0000[0:0]: ERR RPT0=0000000000000001
[ 182.248337394,3] PHB#0000[0:0]: ERR RPT1=0000000000000000
[ 182.248376417,3] PHB#0000[0:0]: AIB ERR=00ce100000000000
[ 183.481572548,5] PHB#0000:00:00.0 [ROOT] 1014 0652 R:00 C:060400 B:01..1a
[ 183.481819812,5] PHB#0000:01:00.0 [SWUP] 11f8 4052 R:00 C:060400 B:02..1a LOC_CODE=U78DB.ND0.WZS008L-P0
[ 183.482175973,5] PHB#0000:02:00.0 [SWDN] 11f8 4052 R:00 C:060400 B:03..03
[ 183.482333011,5] PHB#0000:03:00.0 [EP ] 144d a80a R:00 C:010802 ( mass-storage) LOC_CODE=U78DB.ND0.WZS008L-P0

If there's no NVMe disks under the card why are there mass-storage devices being discovered when enumerating PCI devices? Are they synthetic devices being generated by the switch? If so you might just need to increase the link-up wait time. Considering we seem to be able to scan the device after the subsequent CRESET I think the switch might just be taking its sweet time to become ready after PERST is lifted.

Oh and I had some scripts back in the day to parse the meaning of those PHB register dumps. I'm not sure who has them nowdays, but they might shed some light on what's going on there.

lili-lilili, 2023-01-16:
Hi oohal & fbarrat:

I think i sent the wrong message.
I checked the log，and the “mass storage” is nvme disk.
During the test, if nvme disks are connected, the probability of "PHB fenced" is very low.

And，it this the script you're talking about?
https://github.com/open-power/skiboot/blob/master/external/pci-scripts/phberr.py

lili-lilili, 2023-01-16:
Hi oohal & fbarrat:

I think i sent the wrong conclusion.
I checked the log，and the “mass storage” is nvme disk.
It appears that PHB fenced also occurs when the nvme disk is inserted。

And，it this the script you're talking about?
https://github.com/open-power/skiboot/blob/master/external/pci-scripts/phberr.py

fbarrat, 2023-01-16:
Could we recap so we're sure to be on the same page? Here is what I understand:

1. If there's no nvme disk on the card, you still expect the link to train (showing the intermediate bridge with vendor/device=11f8 4052), right?

2. when enabling the extra pci trace via the nvram option, we see the presence bit (bit 19 of the "DLP Training Control Register") the first time we poll the register, but it disappears on all the following polling, for at least 3 seconds (duration of the pci-trace). Then another 1 second (normal polling in the PHB4_SLOT_LINK_WAIT_ELECTRICAL state). So skiboot gives up after 4 seconds and report "LINK: No in-band presence". So the issue here is why the presence bit doesn't stick and the extra hardware tracing in bits 24:27 shows that no progress is being made for 3 seconds.

3. without the pci trace, we can reach the PHB4_SLOT_LINK_WAIT state, presumably because the presence bit 19 showed up once on the first polling, but we never get the link active (bit 23 of the "DLP Training Control Register"). So skiboot timeouts after 2 seconds with "LINK: Timeout waiting for link up". So skiboot issues a complete reset (CRESET), which reports "CRESET: PHB still fenced !!". But that message is only a warning and we keep going. And the link somehow trains in less than 1.5 seconds after the CRESET.

If the above is correct, then the 2nd bullet shows that there's a problem with the reporting with the presence bit (19). We can test if waiting longer helps, but we're already waiting for a total of 4 seconds.
Considering the 3rd bullet, I'm inclined to believe that the CRESET makes a difference with that adapter, since we trained a lot faster (less than 1.5 second). We would still have to explain the "PHB still fenced !!" warning message, but it could be a side-effect of the previously aborted training sequence (not that it makes it right).

Could you try the following:
1. increase the timeouts, as Oliver suggested? 
2. if the increased timeout doesn't help, could we start the whole sequence with a CRESET, to confirm that it indeed makes a difference with that adapter? And enable the extra pci-trace in that scenario also, to monitor the state of the various bits of the DLP training control register and confirm that they are indeed coherent after the PERST.

> And，it this the script you're talking about?
> https://github.com/open-power/skiboot/blob/master/external/pci-scripts/phberr.py

I think so, I'm not aware of anything else. The script (or checking the PHB doc) just says "ETU reset error". The PBAIB Error Report Register seems to have more information, but that register is just listing individual error status bits which are summarized in the FIR register and the doc is not giving more description.

lili-lilili, 2023-01-17:
fbarrat
Your list of 3 is correct.
One more thing that I think is unusual is that the training state machine starts with polling.

“Increase the timeout“  -- has no help.

“Start the whole sequence with a CRESET” -- has no help.
If we start with CRESET, many other PHB will report “PHB still fenced” too.
Here is the log for PHB0(connect nvme card):
> [  251.076901026,7] PHB#0000[0:0]: Initializing PHB...
> [  251.076901890,7] PHB#0000[0:0]: Default system config: 0x4410000050400000
> [  251.076902963,7] PHB#0000[0:0]: New system config    : 0x4410000050400000
> [  251.076903811,7] PHB#0000[0:0]: Initial PHB CRESET is 0xe000000000000000
> [  251.076905126,7] PHB#0000[0:0]: LINK: Disabling scaled flow control
> [  251.076907044,7] PHB#0000[0:0]: Waiting for DLP PG reset to complete...
> [  251.077421938,7] PHB#0000[0:0]: store EOI is enabled
> [  251.077423014,7] PHB#0000[0:0]: Using ABT mode. ESB: 0x0006051fc0000000
> [  251.077423954,7] PHB#0000[0:0]: Reset state SRC_ID: 0ff8000000000000
> [  251.077741922,7] PHB#0000[0:0]: Initialization complete
> [  251.099225844,7] PHB#0000[0:0]: CRESET: Starts
> [  251.099227943,6] PHB#0000[0:0]: CRESET: Workaround for HW551382
> [  251.104393362,7] PHB#0000[0:0]: CRESET: No pending transactions
> [  251.104395477,3] PHB#0000[0:0]: CRESET: PHB still fenced !!
> [  251.104469426,3] PHB#0000[0:0]:             PCI FIR=1000000000000000
> [  251.104818657,3] PHB#0000[0:0]:         PCI FIR WOF=1000000000000000
> [  251.104857794,3] PHB#0000[0:0]:            NEST FIR=0000800000000000
> [  251.104894929,3] PHB#0000[0:0]:        NEST FIR WOF=0000002000000000
> [  251.104925400,3] PHB#0000[0:0]:            ERR RPT0=0000000000000001
> [  251.104959488,3] PHB#0000[0:0]:            ERR RPT1=0000000000000000
> [  251.104994307,3] PHB#0000[0:0]:             AIB ERR=00ce100000000000
> [  251.110150706,7] PHB#0000[0:0]: CRESET: Reinitialization
> [  251.110151574,7] PHB#0000[0:0]: Initializing PHB...
> [  251.110152373,7] PHB#0000[0:0]: Default system config: 0x4410000050400000
> [  251.110153189,7] PHB#0000[0:0]: New system config    : 0x4410000050400000
> [  251.110154020,7] PHB#0000[0:0]: Initial PHB CRESET is 0xe000000000000000
> [  251.110154897,7] PHB#0000[0:0]: LINK: Disabling scaled flow control
> [  251.110155684,7] PHB#0000[0:0]: Waiting for DLP PG reset to complete...
> [  251.110670499,7] PHB#0000[0:0]: store EOI is enabled
> [  251.110671670,7] PHB#0000[0:0]: Using ABT mode. ESB: 0x0006051fc0000000
> [  251.110674417,7] PHB#0000[0:0]: Reset state SRC_ID: 0ff8000000000000
> [  251.110805795,7] PHB#0000[0:0]: Initialization complete
> [  251.110806981,7] PHB#0000[0:0]: CRESET: wait_time = 228
> [  251.227545489,7] PHB#0000[0:0]: FRESET: Starts
> [  251.227546192,7] PHB#0000[0:0]: FRESET: Prepare for link down
> [  251.227546948,7] PHB#0000[0:0]: FRESET: Assert skipped
> [  251.227547728,7] PHB#0000[0:0]: FRESET: Deassert
> [  251.227548630,7] PHB#0000[0:0]: LINK: Start polling
> [  251.228062977,7] PHB#0000[0:0]: LINK: Electrical link detected
> [  251.279787138,7] PHB#0000[0:0]: LINK: Link is up
> [  252.279789944,7] PHB#0000[0:0]: LINK: Link is stable
> [  252.279830529,7] PHB#0000[0:0]: LINK: Card [11f8:4052] Optimal Retry:disabled
> [  252.279831697,7] PHB#0000[0:0]: LINK: Speed Train:GEN4 PHB:GEN5 DEV:GEN4
> [  252.279832401,7] PHB#0000[0:0]: LINK: Width Train:x08 PHB:x08 DEV:x16
> [  252.279833267,7] PHB#0000[0:0]: LINK: RX Errors Now:0 Max:8 Lane:0x0000
> [  253.089383154,7] PHB#0000:00:00.0 Link up at x8 width
> [  253.089384980,7] PHB#0000:00:00.0 Scanning (upstream+downsteam)...
> [  253.089510439,7] PHB#0000:00:00.0 Found VID:1014 DEV:0652 TYP:4 MF- BR+ EX+
> [  253.089832133,7] PHB#0000:00:00.0 Bus 01..ff scanning...
> [  253.089835875,7] PHB#0000:00:00.0 waiting for link... 
> [  253.141039590,7] PHB#0000:00:00.0 link is up
> [  253.141159666,7] PHB#0000:01:00.0 Found VID:11f8 DEV:4052 TYP:5 MF+ BR+ EX+
> [  253.141510885,7] PHB#0000:01:00.1 Found VID:11f8 DEV:4052 TYP:0 MF+ BR- EX+
[msglog-start-with-creset.txt](https://github.com/open-power/skiboot/files/10430349/msglog-start-with-creset.txt)

fbarrat, 2023-01-17:
I've been talking with the hardware team. They are still looking into it.
I was expecting to replicate the "PHB still fenced !!" warning locally by forcing a CRESET on all the slots. But I can't, everything runs smoothly here. Other than the adapter you have on PHB#0000, how different is your hw setup compared to other IBM's rainier systems? May you share your skiboot tree so that I can try with that directly?

lili-lilili, 2023-01-18:
skiboot is v7.0, and we added the following patch.
[skiboot.zip](https://github.com/open-power/skiboot/files/10440970/skiboot.zip)

lili-lilili, 2023-01-18:
The hardware configuration of our rainier machine is very simple: CPU*2 + nvme card*2 + nvme disk*4 + network card*1;

Some test info:
We found that this issue was related to the nvme card and its location.
We number the nvme cards no.1 and no.2.
If connect them to a different PHB, the test results are different.
Only PHB#0000 + nvme card no.1 will report "PHB  still fenced".
PHB#0000 + nvme card no.2, PHB#0003 + nvme card no.1, PHB#0003 + nvme card no.2, do not report "PHB still fenced".

Because my colleague only checked the log for the "PHB still fenced" keyword, I'm not sure if there are any other errors in the log.
We will do the test again and give an accurate result.
Because we are about to have the Chinese New Year holiday, the test result will ready about 10 days later.

fbarrat, 2023-01-31:
I can reproduce the "PHB still fenced" problem locally now. As you noticed, it only shows up on some slots. I started seeing it when moving to a different system. It's under investigation.

lili-lilili, 2023-02-02:
This is really good news

fbarrat, 2023-02-06:
The "PHB still fenced" issue is now understood. The error is caused by another component, known as the OCC, doing a xscom access on the ETU while it is in reset (the OCC is doing some performance monitoring by accessing perf counters throughout the system). The error can be seen on any slot doing the CRESET sequence, but there's a race condition with the OCC, so it's not always observed. In any case, while I agree the message is worrying, it was treated like a warning by the current code and should not create any problem. I believe it's the case here, the link ended up training ok and the host could access the device.
In any case, I'll send a patch of the skiboot mailing list shortly.

lili-lilili, 2023-02-07:
Thanks for your update.
Regarding OCC access ETUs, can you give details of the relevant OOC code? Which file? Which function?
I would like to know more details, thanks!

fbarrat, 2023-02-07:
> Regarding OCC access ETUs, can you give details of the relevant OOC code? Which file? Which function?

It starts here:
https://github.com/open-power/occ/blob/release-fw1020/src/occ_gpe1/gpe1_24x7.h#L182
The 24x7 service does performance monitoring and the file above has a list of registers to configure/read the counters. If you want more details about the OCC, you'll need to get in touch with that team.

fbarrat, 2023-02-07:
I've just sent a patch on the skiboot mailing list which addresses the "PHB still fenced" warning message.

Regarding the other problem, i.e. "state machine starts with polling", I believe what happens here is that we somehow reach the "polling" state very quickly with that adapter, so that's the first state we see when polling the register. Then we seem to be hitting a problem (no more forward progress and more importantly the presence detect signal disappears), which hints that something bad is happening on the adapter. I'm told that going back from "polling" to "detect" is expected if no progress is made after 24 ms (that's also visible in the state transition diagram found in section "Link Training and Status State Machine (LTSSM) Descriptions" of the PCI spec). That specific adapter apparently needs to go through a full reset sequence to train correctly. Which is what happens by default if you ignore the "PHB still fenced" message from the previous issue.

lili-lilili, 2023-02-08:
Let me try to understand the whole process of the problem;

Is the following procedure correct？
```
freset -->LINK: Timeout waiting for link up-->Creset-->PHB still fenced
                                                ↑
                                 OCC read PHB PMCR scom register here
```

There is another question, why does the freset fail.
If I replace this nvme card with a network card, it seems everything OK.
In fact, we want to confirm that there is no problem on the CPU side.

lili-lilili, 2023-02-08:
@fbarrat 
Can you share your experience in analyzing this problem?
I have no idea how you located this problem to “OCC access PHB scom register”.

fbarrat, 2023-02-08:
> @fbarrat Can you share your experience in analyzing this problem? I have no idea how you located this problem to “OCC access PHB scom register”.

First, I had to rule out that skiboot was not doing a PHB scom access. At that stage, we're still booting, linux is not even running, so skiboot is the only thing running on the host. All the scom accesses you see in CRESET are targeting the PEC and not the PHB (you can think of the PEC as one layer on top of the PHB). I had also instrumented the scom layer to make sure nothing else was running behind my back. Everything was clean. I had also noticed that for the cases where we were not seeing the "PHB still fenced" message, it was racy. If we poll several times on the FIR registers while the ETU is in reset, we always end up seeing PFIR bit 3 set. That, plus the fact that the hw team was telling me PFIR bit 3 could only be a ETU scom access while in reset, means there was something polling on the system. And it was not skiboot.
I then thought it could maybe come from the BMC. But that didn't go anywhere.
Then I thought about the OCC. grep'ing through the source code, I saw some PHB scom addresses for perf counters. Talking with an OCC developer, I could disable the 24x7 service polling perf registers and the PFIR bit 3 never showed up, even if polling for a while while the ETU is in reset.

That's actually not new behavior. I could also reproduce the same thing on P9.

fbarrat, 2023-02-08:
.> Is the following procedure correct？
> 
> ```
> freset -->LINK: Timeout waiting for link up-->Creset-->PHB still fenced
>                                                 ↑
>                                  OCC read PHB PMCR scom register here
> ```

Yes, that diagram is correct.

> There is another question, why does the freset fail. If I replace this nvme card with a network card, it seems everything OK. 

So one adapter works fine and not the other. Bad adapter? The troubling part is the the presence detect bit is not stable (bit 19 of the "DLP training control register"). We can see that in the log where you enabled the extra PCI tracing. That signal is coming straight from the physical layer. Also, that same adapter shows no problem when we go through a CRESET sequence, which first asserts the PERST signal to reset the adapter. That is all pointing to something fishy happening on the adapter, right?

lili-lilili, 2023-02-27:
> I've just sent a patch on the skiboot mailing list which addresses the "PHB still fenced" warning message.
>
> Regarding the other problem, i.e. "state machine starts with polling", I believe what [...]

Hi fbarrat:
When will this commit be merged?

closed 2023-02-27 by rarbab