open-power/skiboot #275: P10 small core cpu checkstop

lili-lilili, 2023-05-18:
When we test a Power10 small core CPU(48 core * 2) on the rainier, when skiroot kernel start CPU will checkstop due to "L3 Dir read UE".
If we use a Big core CPU, there is no problem.
If we reduce the core to 48(by "Field core override"),  then we can boot skiroot kernel sucess.
Is there anything to need special attention when using small core cpu? Like some attribute settings, or some config?

Here is the event log:
```
{
"Private Header": {
"Section Version": "1",
"Sub-section type": "0",
"Created by": "0xE500",
"Created at": "05/17/2023 05:48:45",
"Committed at": "05/17/2023 05:48:45",
"Creator Subsystem": "BMC",
"CSSVER": "",
"Platform Log Id": "0x50000011",
"Entry Id": "0x50000011",
"BMC Event Log Id": "243"
},
"User Header": {
"Section Version": "1",
"Sub-section type": "0",
"Log Committed by": "0x2000",
"Subsystem": "Processor Unit (CPU)",
"Event Scope": "Entire Platform",
"Event Severity": "Unrecoverable Error",
"Event Type": "Not Applicable",
"Action Flags": [
"Service Action Required",
"Report Externally",
"HMC Call Home"
],
"Host Transmission": "Not Sent",
"HMC Transmission": "Not Sent"
},
"Primary SRC": {
"Section Version": "1",
"Sub-section type": "1",
"Created by": "0xE500",
"SRC Version": "0x02",
"SRC Format": "0x55",
"Virtual Progress SRC": "False",
"I5/OS Service Event Bit": "False",
"Hypervisor Dump Initiated":"False",
"Backplane CCIN": "2E2F",
"Terminate FW Error": "False",
"Deconfigured": "False",
"Guarded": "True",
"Error Details": {
"Message": "Error Signature: 0x20DA0020 0x00020001 0x5074140E"
},
"Valid Word Count": "0x09",
"Reference Code": "BD13E510",
"Hex Word 2": "00000055",
"Hex Word 3": "2E2F0010",
"Hex Word 4": "00000000",
"Hex Word 5": "01000000",
"Hex Word 6": "20DA0020",
"Hex Word 7": "00020001",
"Hex Word 8": "5074140E",
"Hex Word 9": "00000000",
"Callout Section": {
"Callout Count": "1",
"Callouts": [{
"FRU Type": "Normal Hardware FRU",
"Priority": "Medium Priority",
"Location Code": "U78DB.ND0.WZS008L-P0-C24",
"Part Number": "F210110",
"CCIN": "AB42",
[... 999 lines cut ...]
[  302.007332343,3] STB: EV_SEPARATOR (pcr7) NOT MEASURED. No TPM registered/enabled
[  302.007382166,3] tss_get_random_number: failed, rc 000b0008
[  302.007416675,3] TSS_RC_NO_CONNECTION - Failure connecting to lower layer
[  302.007461384,3] Failed to generate the auth password: failed, rc 000b0008
[  302.007504184,3] TSS_RC_NO_CONNECTION - Failure connecting to lower layer
[  302.062581859,5] INIT: Starting kernel at 0x20011000, fdt at 0x30a21098 723361 bytes
[  302.075743204,5] DUMP: Architected register dest addr : 0x39fa0000, size : 0x200000

zImage starting: loaded at 0x0000000020011000 (sp: 0x000000002108aed8)

Allocating 0x1b3db20 bytes for kernel...

Decompressing (0x0000000000000000 <- 0x0000000020023000:0x0000000021088464)...

Done! Decompressed 0x1974c88 bytes

Linux/PowerPC load: 

Finalizing device tree... flat tree at 0x2108c0c0

[  303.485727285,5] XIVE: [ IC 00  ] Resetting one xive...
[  304.013917348,5] XIVE: [ IC 02  ] Resetting one xive...
[  304.053624245,5] XIVE: [ IC 04  ] Resetting one xive...
[  304.094036000,5] XIVE: [ IC 06  ] Resetting one xive...
```

npiggin, 2023-05-19:
I don't know of any problem or attribute. I'll have to check if we have an OPAL test system with this many cores.

Is the Linux boot quiet? It could be useful if we could narrow down a bit more where we are in the kernel when it checkstops.

lili-lilili, 2023-05-30:
Update log:
1. Override wof table(48 small core)
2. Set Hostboot config CONSOLE_OUTPUT_TRACE
3. Enable skiboot and linux kernel log

Console log:
https://github.com/lili-lilili/share_files/blob/main/obmc-console-wof-match-HBTrace.log

L3 Dir Read UE PEL:
https://github.com/lili-lilili/share_files/blob/main/event_logs_2023-05-30_20-07-29-L3-Dir-read-UE.txt

edwin-wang, 2023-05-31:
@npiggin @mikey I would like to express my gratitude for your previous assistance. With overriding WOF in hostboot, WOF mismatch has gone. But the L3 Dir checkstop is still there. It happened after OPAL IPL/in petitboot. System will reboot when hitting the checkstop. With @dcrowell77 's suggestion, the verification with disabling power management in hostboot is on-going. Will share you the result soon.
@lili-lilili shared logs above, do you have any insight?

lili-lilili, 2023-05-31:
@npiggin @mikey @edwin-wang @dcrowell77 
“disabling power management in hostboot” encounter some problem。
It looks like OPAL do not support Hostboot unset START_OCC_DURING_BOOT, if we do that, the skiboot can't start properly.

Here is the error log:
```
[  142.407815526,6] MFSI 4:1: Initialized
[  142.407836051,6] MFSI 6:0: Initialized
[  142.407857445,6] MFSI 6:2: Initialized
[  142.407878277,6] MFSI 6:1: Initialized
[  142.408180145,6] LPC: LPC[000]: Initialized
[  142.408213764,7] LPC: access via MMIO @0x6030000000000
[  142.408269210,7] LPC: Default bus on chip 0x0
[  142.408348115,7] CPU: New max PIR set to 0x677
[  142.408506123,6] MEM: parsing reserved memory from node /ibm,hostboot/reserved-memory
[  142.408565491,3] ***********************************************
[  142.408601155,3] < assert failed at core/mem_region.c:516 >
[  142.408633582,3]     .
[  142.408648442,3]      .
[  142.408662098,3]       .
[  142.408676167,3]         OO__)
[  142.408693122,3]        <"__/
[  142.408709506,3]         ^ ^
[  142.408746774,3] Fatal TRAP at 000000003002508c   .mem_free+0xec  MSR 9000000000021002
[  142.408797821,3] CFAR : 0000000030025044 MSR  : 9000000000021002
[  142.408832323,3] SRR0 : 000000003002508c SRR1 : 9000000000021002
[  142.408868125,3] HSRR0: 0000000030020024 HSRR1: 9000000000001000
[  142.408904605,3] DSISR: 00000000         DAR  : 0000000000000000
[  142.408938927,3] LR   : 000000003002502c CTR  : 0000000000000000
[  142.408971865,3] CR   : 22000224         XER  : 00000000
[  142.409002521,3] GPR00: 0000000030026c10 GPR16: 0000000034db0000
[  142.409036308,3] GPR01: 0000000034db37c0 GPR17: 00000000300031bc
[  142.409075381,3] GPR02: 0000000030192e00 GPR18: 0000000000000000
[  142.409110629,3] GPR03: 0000000000000001 GPR19: 0000000000000000
[  142.409143521,3] GPR04: 0000000030261380 GPR20: 0000000000000000
[  142.409177070,3] GPR05: 0000000030121f54 GPR21: 00000000306ca478
[  142.409209744,3] GPR06: 0000000000400000 GPR22: 0000000030121fd8
[  142.409243154,3] GPR07: 0000000000000000 GPR23: 0000000030121fb1
[  142.409279068,3] GPR08: 0000000000000000 GPR24: 0000000030122016
[  142.409314095,3] GPR09: 0000000030600000 GPR25: 0000000030121ff6
[  142.409349180,3] GPR10: 0000000030600010 GPR26: 0000000030121fef
[  142.409383930,3] GPR11: 0000000000000000 GPR27: 0000000030120709
[  142.409417545,3] GPR12: 0000000042000222 GPR28: 0000000030121f54
[  142.409450792,3] GPR13: 0000000003b78908 GPR29: 0000000030121f54
[  142.409484560,3] GPR14: 0000000000000000 GPR30: 0000000030261298
[  142.409520337,3] GPR15: 0000000000000000 GPR31: 0000000030261380
CPU 0634 Backtrace:
 S: 0000000034db3ad0 R: 0000000030023dd0   .new_region+0x30
 S: 0000000034db3b60 R: 0000000030026c10   .__free+0x40
 S: 0000000034db3bf0 R: 000000003002457c   .add_region+0x1ac
 S: 0000000034db3c80 R: 0000000030024884   .mem_region_parse_reserved_nodes+0x168
 S: 0000000034db3d50 R: 0000000030025cf0   .mem_region_init+0x3c4
 S: 0000000034db3e20 R: 0000000030022f38   .main_cpu_entry+0x504
 S: 0000000034db3f00 R: 000000003000321c   go_primary+0x10c
 --- OPAL boot ---
```
Complete log:
https://github.com/lili-lilili/share_files/blob/main/obmc-console-stop-pm.log

mikey, 2023-06-01:
This section in the OPAL messages suggest the reserved memory regions passed via HDAT are broken. I think the later crash is likely due to us train wrecking on this earlier issue. 

```[  131.261716061,7] MEM: Reserve 'hb-rsv-mem' e0000000-efffffff (type/inst=0x00000000)
[  131.261772266,7] MEM: Reserve 'homer-image' 0-3fffff (type/inst=0x04000000)
[  131.261818208,7] MEM: Reserve 'homer-image' 0-3fffff (type/inst=0x04000002)
[  131.261861943,3] DT: dt_attach_root failed, duplicate ibm,homer-image@0
[  131.261901921,3] Unable to create node for ibm,homer-image@0
[  131.261933759,3] unable to add node?
[  131.261954306,7] MEM: Reserve 'homer-image' 0-3fffff (type/inst=0x04000004)
[  131.261994015,3] DT: dt_attach_root failed, duplicate ibm,homer-image@0
[  131.262033005,3] Unable to create node for ibm,homer-image@0
[  131.262066913,3] unable to add node?
[  131.262087139,7] MEM: Reserve 'homer-image' 0-3fffff (type/inst=0x04000006)
[  131.262131610,3] DT: dt_attach_root failed, duplicate ibm,homer-image@0
[  131.262173339,3] Unable to create node for ibm,homer-image@0
[  131.262206536,3] unable to add node?
[  131.262228035,7] MEM: Reserve 'arch-reg-data' c00ffce00000-c00ffd3fffff (type/inst=0x01000000)
[  131.262277522,7] MEM: Reserve 'hbrt-data' c00ffccc0000-c00ffcdfffff (type/inst=0x01000000)
[  131.262329603,7] MEM: Reserve 'hbrt-code-image' c00ffc7c0000-c00ffccbffff (type/inst=0x01000000)
[  131.262386866,7] MEM: Reserve 'sbe-comm' c00ffc7b0000-c00ffc7bffff (type/inst=0x01000000)
[  131.262438920,7] MEM: Reserve 'sbe-ffdc' c00ffc7a0000-c00ffc7affff (type/inst=0x01000000)
[  131.262487652,7] MEM: Reserve 'sbe-comm' c00ffc790000-c00ffc79ffff (type/inst=0x01000002)
[  131.262536250,7] MEM: Reserve 'sbe-ffdc' c00ffc780000-c00ffc78ffff (type/inst=0x01000002)
[  131.262587669,7] MEM: Reserve 'sbe-comm' c00ffc770000-c00ffc77ffff (type/inst=0x01000004)
[  131.262640270,7] MEM: Reserve 'sbe-ffdc' c00ffc760000-c00ffc76ffff (type/inst=0x01000004)
[  131.262693654,7] MEM: Reserve 'sbe-comm' c00ffc750000-c00ffc75ffff (type/inst=0x01000006)
[  131.262744806,7] MEM: Reserve 'sbe-ffdc' c00ffc740000-c00ffc74ffff (type/inst=0x01000006)
[  131.262794420,7] MEM: Reserve 'sbe-psu' c00ffc73f000-c00ffc73ffff (type/inst=0x01000000)
[  131.262844186,7] MEM: Reserve 'secure-crypt-algo-code' c00ffc72f000-c00ffc73efff (type/inst=0x03000000)
[  131.262907583,7] MEM: Ignoring zero length range
<snip>
[  131.264566958,7] MEM: Ignoring zero length range
[  131.264591962,7] MEM: Ignoring zero length range
```
We shouldn't be reserving memory at zero. That seems like a HDAT bug to try to do that. 

You could try this patch as a workaround for this. Untested  
```
diff --git a/hdata/memory.c b/hdata/memory.c
index efdb502b1c8d..77fefedf78ec 100755
--- a/hdata/memory.c
+++ b/hdata/memory.c
@@ -797,6 +797,12 @@ static void get_hb_reserved_mem(struct HDIF_common_hdr *ms_vpd)
 			continue;
 		}
 
+		/* Check if region starts at zero */
+		if (start_addr == 0) {
+			prlog(PR_DEBUG, "MEM: Ignoring region starting at zero\n");
+			continue;
+		}
+
 		/* remove the HRMOR bypass bit */
 		start_addr &= ~HRMOR_BIT;
 		end_addr &= ~HRMOR_BIT;
```

lili-lilili, 2023-06-01:
It seems that homer-image is not correct when START_OCC_DURING_BOOT unset.
If skip homer-image parsing, OPAL  can success boot up，and no checkstop happened.

@dcrowell77 @mikey Have you had any ideas?

```
diff --git a/hdata/memory.c b/hdata/memory.c
index 212aa05..f559a09 100755
--- a/hdata/memory.c
+++ b/hdata/memory.c
@@ -13,6 +13,7 @@
 
 #include "spira.h"
 #include "hdata.h"
+#include "string.h"
 
 struct HDIF_ram_area_id {
 	__be16 id;
@@ -782,6 +783,11 @@ static void get_hb_reserved_mem(struct HDIF_common_hdr *ms_vpd)
 		start_addr = be64_to_cpu(hb_resv_mem->start_addr);
 		end_addr = be64_to_cpu(hb_resv_mem->end_addr);
 
+        if (0 == memcmp(hb_resv_mem->label, "homer-image", strlen("homer-image"))) {
+            prlog(PR_ERR, "MEM: skip homer-image\n");
+			continue;
+        }
+
 		/* Zero length regions are a normal, but should be ignored */
 		if (start_addr - end_addr == 0) {
 			prlog(PR_DEBUG, "MEM: Ignoring zero length range\n");
@@ -797,6 +803,11 @@ static void get_hb_reserved_mem(struct HDIF_common_hdr *ms_vpd)
 			continue;
 		}
 
+		if (start_addr == 0) {
+			prlog(PR_ERR, "MEM: Ignoring region starting at zero\n");
+			continue;
+		}
+
 		/* remove the HRMOR bypass bit */
 		start_addr &= ~HRMOR_BIT;
 		end_addr &= ~HRMOR_BIT;
```

dcrowell77, 2023-06-01:
The HB code that does the reserved memory stuff is in src/usr/runtime/populate_hbruntime.C .  You might need to move the CONFIG_START_OCC_DURING_BOOT check up a little bit to skip more code.

The truth is that we haven't supported OPAL boots in P10 so this entire path is untested.  I also can't remember the last time anyone ran without CONFIG_START_OCC_DURING_BOOT set even in P9.  You'll have to dig into the details as issues like this come up.

dcrowell77, 2023-06-01:
> OPAL can success boot up，and no checkstop happened.

Focusing on the main problem, this means that you've isolated the issue to something in the power management layer.  You'll need to work through the appropriate channels for more help on smallcore support for that.

lili-lilili, 2023-06-02:
> > OPAL can success boot up，and no checkstop happened.
> 
> Focusing on the main problem, this means that you've isolated the issue to something in the power management layer. You'll need to work through the appropriate channels for more help on smallcore support for that.

Thinks very much.
Communication with IBM is underway.
Power manage refer to OCC or MVDP data or something else ?

dcrowell77, 2023-06-02:
"Power Management" refers to the entire complex that handles thermal/voltage/current monitoring and related actions.  The specific portion here would be the logic related to handling STOP states on the cores.  The most likely explanation is some missing function within the HCODE that manages the transitions in and out of STOP states.

svaidy, 2023-06-05:
_Some observations on this issue:_

1. "L3 Dir Read UE" is less likely to be caused by software or boot up sequence.  We most likely have a real bad cache line directory
2. The error is not showing up when booted in big-core mode may just mean we have not stressed this path or cache state scenario.
3. In order to prove that this is really hardware issue, we can deactivate only the failing core (and its quad just for test) then the system should boot up fine similar to what has been proved by reducing number of available cores
4. Disabling power management (OCC) makes the problem go away because then system will run at lower frequency and this hardware issue could be masked.
5. This issue has nothing to do with core idle states (stop states) because we have only stop0 enabled and stop2,3 disabled. No stop11 support anyway.  So the only enabled state: stop0 is completely hardware stop/start logic and no hcode or software is involved in core wakeup.
6. Corrupt homer region will become an issue when we run stop2,3, but that is not the case here

**Summary:**  The issue signature points to a hardware error, disabling the failing core should prove the hypothesis. 
Disabling power management -- OCC will drop core frequency and can block the hardware issue

_General notes:_

- Power management in general involves three sub-systems/frameworks as follows:
- Thermal management, fan control, power cap, ... generally manage the system to operate in safe limits, throttle cpu and memory
- Cpu core/chip frequency management: OCC complex and related areas, WoF and enhanced frequency range management
- Cpu core idle state management: stop states, stop0,2,3,11

lili-lilili, 2023-06-05:
I did a test few days ago:
1. deconfig core1-core11 on each proc，machine can boot to os success，and do stream test for each active core，no check stop happen。
2. deconfig core12-core23 on each proc，machine can boot to os success，and do stream test for each active core，no check stop happen。
So it doesn't look like some error with specified cores.

dcrowell77, 2023-06-05:
@lili-lilili - I didn't follow your comment about deconfiguring things.  It passed in smallcore mode with none of the workarounds?  Are those cores you disabled the ones where the checkstop fired?

Is the checkstop always on the same core?  If you guard that core does it move to a new one?

I'm not as confident as @svaidy that it is bad hardware since that is an order of magnitude more rare than a code issue on a configuration that nobody is testing.  Even for stop0 there is likely some level of uniqueness in the configuration between smallcore and bigcore.

lili-lilili, 2023-06-06:
> @lili-lilili - I didn't follow your comment about deconfiguring things. It passed in smallcore mode with none of the workarounds? 

Yes，without any workarounds.
The only different is that force deconfig some cores when hostboot discovery target.

>Are those cores you disabled the ones where the checkstop fired?

No.
The problem happened on 48 small core CPU.
I tried to deconfig core1-core11 or core12-core23 on each proc.

> 
> Is the checkstop always on the same core? If you guard that core does it move to a new one?

Not the same core.

lili-lilili, 2023-06-06:
We also test 24core and 48core small core CPU，no checkstop happened.

svaidy, 2023-06-16:
Thanks for the detailed tests.  The nature of the error is L3 dir UE.  One point Dan made is can this be a setup issue, like a bad cache line is masked correctly in big-core mode but missed in small core mode, in this case we should be hitting this on specific core.

My suspicion is this is related to frequency/voltage/margin/guard band related power-delivery area.  I think so because, every test that passes has lower power demand.  Either running at lower frequency or lower core count masks the issue.

We hit the problem at boot, when we run small core, there is more boot-code or logic running in the system because every core's t0 is engaged.  While in big core mode the boot-up load is less as we have less cores and their t0 engaged.
If we stress the cores hard enough post boot we should possibly hit this problem even in big core mode.

Some suggestions:
1. Can we take help from power management WoF team to adjust the table (V/F) points or cut some PStates to verify that this is indeed the case of running the chip too close to guard-band or lower voltage margin?
2. If adjusting WOF table is difficult, then we can workaround in s/w by making OPAL set max-pstate to a lower value when passing to Linux.  But this means we have to boot at least till end of OPAL and not check stop before that.  So this workaround is less likely to work.
3. Is this fail reproducible in more one than one system and/or more than one type of chip (core count)?

In our lab (Linux development) we are able to run OPAL_PowerNV Linux on rainier system in small core and big core modes without any check stop.  We routinely run tests in small core mode. We have seen stop2/3 and OCC related issues but not boot-up checkstop like this L2 dir UE or something deep in the hardware.

lili-lilili, 2023-06-16:
> Can we take help from power management WoF team to adjust the table (V/F) points or cut some PStates to verify that this is indeed the case of running the chip too close to guard-band or lower voltage margin?

@svaidy Thanks.
We are already receiving support from the CPU team.
As you suspect, this is likely a voltage related issue. 
We are still working to solve this problem.

lili-lilili, 2023-06-16:
> We routinely run tests in small core mode. We have seen stop2/3 and OCC related issues but not boot-up checkstop like this L2 dir UE or something deep in the hardware.

Can you share the details of this issue?
We have also encountered some issues that may be related to OCC, such as' CPU can only run to Wof base frequency '

closed 2024-03-05 by lili-lilili